AravisProject/aravis #353: ** (lt-arv-viewer:13151): WARNING **: bayer2rgb0: size 3187128 is not a multiple of unit size 3186816

chogi89, 2020-05-07:
**Describe the bug**
A clear and concise description of what the bug is.
I'm using DALSA Genie Nano C2020.
When I display the image using art-viewer in viewer folder, it display the comment which is "** (lt-arv-viewer:13151): WARNING **: bayer2rgb0: size 3187128 is not a multiple of unit size 3186816
" in terminal

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Camera description:**
 - Manufacturer: DALSA
 - Model: Genie Nano C2020
 - Interface: GigE

**Platform description:**
 - Aravis version: 0.6.4
 - OS: Ubuntu 16.04 
 - Hardware: x86_64 and I suffer in the nvidia TX2 

**Additional context**
Add any other context about the problem here.

EmmanuelP, 2020-05-10:
Please attach the console output with log enabled: `arv-viewer-0.6 -d all`.

chogi89, 2020-05-11:
Thanks, EmmanuelP,

This is result for arv-viewer -d all.

./arv-viewer -d all
Found 0 USB3Vision device (among 5 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 169.254.177.153
[GvDiscoverSocket::new] Add interface 192.168.0.112
[GvDiscoverSocket::new] Add interface 172.17.0.1
[GvInterface::discovery] Device 'Teledyne DALSA-Nano-C2020-S1222415' found (interface 169.254.177.153) user_id 'S1222415' - MAC '00:01:0d:c4:d0:a3'
Found 0 USB3Vision device (among 5 USB devices)
[GvDevice::new] Interface address = 169.254.177.153
[GvDevice::new] Device address = 169.254.9.215
[GvDevice::load_genicam] xml url = 'Local:TeledyneDALSA_Nano-IMX264-265_Bayer_3.2M-5.1M_ccef7ad8_ACA18.0016.zip;b000;11d6a' at 0x200
[GvDevice::load_genicam] Xml address = 0xb000 - size = 0x11d6a - TeledyneDALSA_Nano-IMX264-265_Bayer_3.2M-5.1M_ccef7ad8_ACA18.0016.zip
[GvDevice::load_genicam] Zipped xml data
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (pAlias)
[Genicam::create_element] Unknown tag (Representation)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (ImposedAccessMode)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (ImposedAccessMode)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (ImposedAccessMode)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (ImposedAccessMode)
[GcFeature::set_attribute] Unknown attribute 'Comment'
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (pSelected)
[Genicam::create_element] Unknown tag (pSelected)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (pError)
[Genicam::create_element] Unknown tag (pValueCopy)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (pError)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (ImposedAccessMode)
[Genicam::create_element] Unknown tag (Visibility)
[... 1176 lines cut ...]
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (pSelected)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
[GcFeature::set_attribute] Unknown attribute 'Comment'
[ArvDevice::set_string_feature_value] Node 'GevGVSPExtendedIDMode' is not a string
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
[PixelFormat::to_gst_caps_string] 0x01100011 not found
[GvDevice::create_stream] Number of stream channels = 1
[GvStream::stream_new] Packet size = 1500 byte(s)
[GvStream::stream_new] Destination stream port = 47444
[GvStream::stream_new] Source stream port = 46572
[GvStream::stream_thread] Packet timeout = 20 ms
[GvStream::stream_thread] Frame retention = 100 ms
Failed to connect make realtime: org.freedesktop.DBus.Error.AccessDenied
Nice level successfully changed to -10
[GvStream::loop] Standard socket method

** (lt-arv-viewer:6061): WARNING **: bayer2rgb0: size 3187128 is not a multiple of unit size 3186816

-Gangik Cho-

EmmanuelP, 2020-05-11:
What is the output of `arv-tool-0.6 control Width Height PayloadSize` ?

chogi89, 2020-05-11:
It is output of ./arv-viewer control Width Height PayloadSize

(lt-arv-viewer:13298): GLib-GIO-CRITICAL **: This application can not open files.

=================================================================
==13298==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 16384 byte(s) in 1 object(s) allocated from:
    #0 0x7f2a8ba2d662 in malloc (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x98662)
    #1 0x7f2a895197b8 in g_malloc (/lib/x86_64-linux-gnu/libglib-2.0.so.0+0x4f7b8)

SUMMARY: AddressSanitizer: 16384 byte(s) leaked in 1 allocation(s).

chogi89, 2020-05-11:
It is output of ./arv-tool-0.6 control Width Height PayloadSize
Teledyne DALSA-Nano-C2020-S1222415 (169.254.9.215)
Width = 2064
Height = 1544
PayloadSize = 3187128

chogi89, 2020-05-14:
I tried install and test on Ubuntu 18.04 and it outputs same result.

EmmanuelP, 2020-05-14:
Could you try to change the pixel format to Mono8 and see if you still have the issue ?

chogi89, 2020-05-14:
I changed firmware to display with Mono8 or YUV422_8, and it works well.
Thank you.

EmmanuelP, 2020-05-14:
Good to know. But you still have issue with bayer, right ?

chogi89, 2020-05-14:
Yes, right. I cannot solved that problem yet.

kohtala, 2020-05-18:
2064 * 1544 = 3186816. Wonder why PayloadSize = 3187128, 312 bytes too large for Bayer, but not for Mono8.

chogi89, 2020-05-18:
I succeed in display of the image using the ROS package where "https://github.com/shone-io/camera_aravis" using Bayer format. But the ROS package doesn't work with YUV format.
Aravis doesn't work with Bayer format yet.