AravisProject/aravis #354: arv-fake-gv-camera: support Bayer pixel formats

lovesegfault, 2020-05-08:
**Is your feature request related to a problem? Please describe.**
I have some code that expects to stream a camera, using Aravis, and to receive bayered frames that it then de-bayers and outputs BGR. I wanted to write some tests for that piece of code using `arv-fake-gv-camera` but I found that I cannot set the pixel format to BayerRG8 as I desired.

**Describe the solution you'd like**
I'd like for `arv-fake-gv-camera` to support Bayer pixel formats for testing code that expects to have to demosaic images.

EmmanuelP, 2020-05-08:
Hi Bernardo,

That would be a nice to have feature.

If you know how to generate a bayer test image, it should be trivial to add to arvfakecamera.c.

lovesegfault, 2020-05-08:
@EmmanuelP Would I just need to hook the bayer pattern generator in here?
https://github.com/AravisProject/aravis/blob/b537efc2e86b4d2adf134bc135ebff5560a4a444/src/arvfakecamera.c#L530

EmmanuelP, 2020-05-09:
Yes, and add an entry in arv-fake-camera.xml here:

https://github.com/AravisProject/aravis/blob/b537efc2e86b4d2adf134bc135ebff5560a4a444/src/arv-fake-camera.xml#L220

lovesegfault, 2020-05-09:
How are those values determined?

EmmanuelP, 2020-05-09:
https://www.emva.org/wp-content/uploads/GenICamPixelFormatValues.pdf
https://www.emva.org/wp-content/uploads/GenICam_PFNC_2_3.pdf

lovesegfault, 2020-05-10:
I have a prototype working, will make a PR soon:
[image]

closed 2020-05-10 by EmmanuelP

Rz-Rz, 2025-03-04:
I wonder if this is possible to change the pixel depth to 10 for this use case ? i.e. using bayerrg10 ? Would love to implement it if feasible

PR #355 (2020-05-10): arvfakecamera: implement Bayer pixel formats
Adds support for all Bayer formats to the `arv-fake-gv-camera` utility.

Closes #354
commit 2e19e33a (2020-05-10): fakecamera: implement Bayer pixel formats (#355)

Adds support for all Bayer formats to the arv-fake-gv-camera utility. Closes #354.