AravisProject/aravis #365: Can't seem to start arv-fake-gv-camera on macOS.

lloydc99, 2020-05-18:
**Describe the bug**
When I try to start arv-fake-gv-camera-0.6 I get:
Failed to start camera

**To Reproduce**
Steps to reproduce the behavior:
arv-fake-gv-camera-0.6 
Failed to start camera

arv-fake-gv-camera-0.6 -i en0
Failed to start camera

**Expected behavior**
I thought it would run.

**Camera description:**
Fake camera

**Platform description:**
 - Aravis version: brew 0.6, also compiled via meson from git (0.6.4, 0.7.3, 0.7.4)
 - OS: macOS Cataline 10.15.4
 - Hardware: iMac

**Additional context**
The fake camera seems to work just fine in the viewer and I've been able to see actual cameras show up in the list from arv-tool.

EmmanuelP, 2020-05-18:
Hi Chuck,

That is something I've seen while trying to enable travis ci on OSX. The fake camera instatiation fails during the creation of the listening interface.

Unfortunately, that is something I can not work on, as I don't have an OSX platform at hand.

EmmanuelP, 2020-05-18:
> The fake camera seems to work just fine in the viewer

It works here because this fake camera is a bit different that the one created by arv-fake-gv-camera, as it does not use sockets for data transfer.

lloydc99, 2020-05-18:
Okay, thanks. I"ll look at the code there.

zyp, 2020-10-08:
As far as I can tell, you're simply not allowed to bind `255.255.255.255` in macOS.

`Failed to bind socket: Error binding to address 255.255.255.255:3956: Can't assign requested address`

 Binding `0.0.0.0` instead (and disabling the subnet discovery socket to avoid a conflict) appears to work.

EmmanuelP, 2021-12-13:
fake camera is working on macOS now.