AravisProject/aravis #371: Commercial licensing option?

robotwhispering, 2020-06-02:
I am reaching out to inquire if there are any possibilities for a commercial closed-source licensing option for Aravis.

EmmanuelP, 2020-06-02:
Hi,

No there is no commercial closed-source option for using Aravis.

That being said, Aravis is LGPL licensed, which means you are free to link it with a proprietary software. Is really the obilgation to publish your modifications to aravis an issue ?

robotwhispering, 2020-06-02:
I have no intentions on modifying Aravis. But to close up security holes
that are introduced when you dynamically link libraries.

Warm regards,

Scott Uneberg
(786) 757-7190

EmmanuelP, 2020-06-02:
> But to close up security holes
that are introduced when you dynamically link libraries.

I don't see why it would prevent the use of LGPL based aravis. Security fixes should definitely be open sourced.

Cheers,

    Emmanuel;

closed 2020-06-02 by EmmanuelP