AravisProject/aravis #373: GStreamer cannot understand pixel format

kasia21, 2020-06-02:
I'm trying to use Aravis 0.7.2 to read out images from a gigE-compliant Teledyne Dalsa Shadobox 6k detector. The arv-tool utility can find the detector, but when I run arv-viewer, I get the following error: "GStreamer cannot understand the camera pixel format: 0x1100025!" and I do not get any images to show in the viewer window. I'm attaching the full output from the arv-viewer with debug mode turned on, as well as the xml file for the detector. This is being run in a Docker container that is using Ubuntu 18.04. Please let me know what further information you may need to help diagnose. I should note that I have also run this test with a Basler Ace gigE camera, and it works without issue. Thanks in advance.
[Dalsa_output.txt](https://github.com/AravisProject/aravis/files/4719456/Dalsa_output.txt)
[Teledyne_ShadoBox.txt](https://github.com/AravisProject/aravis/files/4719460/Teledyne_ShadoBox.txt)

EmmanuelP, 2020-06-03:
Hi,

0x1100025 is Mono14 unpacked. This is not a pixel format for which aravis has a mapping to an equivalent GStreamer format.

https://github.com/AravisProject/aravis/commit/a665af1fdeb8d8d4af3e49c991638481b7bf2c23 should fix this issue.

kasia21, 2020-06-03:
Thanks so much, that did take care of the camera pixel format error. I am still not getting any images to show in the viewer, and the image counter does not increment. I'm attaching the new debug file. Please let me know what else to provide for debugging. 
[output.txt](https://github.com/AravisProject/aravis/files/4725802/output.txt)

EmmanuelP, 2020-06-04:
In the log I see the packet socket is used.

Could you attach the output of `./tests/arv-camera-test` and `./tests/arv-camera-test --on-packe-socket` ?

kasia21, 2020-06-04:
I ran the arv-camera-test with and without the packet socket flag. I also ran the heartbeat test and am attaching its output. For the heartbeat test, I get the following output in the terminal: 

> ** (arv-heartbeat-test:2875): CRITICAL **: 15:31:31.952: arv_gc_feature_node_get_value_as_string: assertion 'ARV_IS_GC_FEATURE_NODE (self)' failed
Setting Gain from (null) to 10
> ** (arv-heartbeat-test:2875): CRITICAL **: 15:31:31.952: arv_gc_feature_node_set_value_from_string: assertion 'ARV_IS_GC_FEATURE_NODE (self)' failed

[camera_test_nopacketsocket.txt](https://github.com/AravisProject/aravis/files/4731109/camera_test_nopacketsocket.txt)
[camera_test.txt](https://github.com/AravisProject/aravis/files/4731110/camera_test.txt)
[heartbeat.txt](https://github.com/AravisProject/aravis/files/4731098/heartbeat.txt)

kasia21, 2020-06-08:
I ran some more tests and found that arv-viewer outputs images when the pixel format is set to "mono12packed". I am attaching the debug output of running arv-viewer using both pixel formats. The main distinction between the two debug files seems to be the error "Maximum number of packet requests reached at dt = [], n_requests = 1073/9256". I am also attaching the output of arv-camera-test with debug mode turned on and with pixel format set to "mono14". I contacted the camera manufacturers and they stated that the detector is sending out extra data (I think 8 bytes) which does not affect the image data in any way; it is in the gvsp trailer packet. I tried editing the "block_size" variable in arvgvstream.c to accommodate the 8 extra bytes, which reduces the "n_size_mismatch_errors" to 0, but I still have nonzero "n_missing_packets" and "n_error_packets" when running arv-camera-test. Thanks so much. 
[arv_cam_test_mono14.txt](https://github.com/AravisProject/aravis/files/4747923/arv_cam_test_mono14.txt)
[viewer_mono12.txt](https://github.com/AravisProject/aravis/files/4747924/viewer_mono12.txt)
[viewer_mono14.txt](https://github.com/AravisProject/aravis/files/4747925/viewer_mono14.txt)

EmmanuelP, 2020-06-09:
It would be nice if you could capture the traffic using wireshark.

In order to limit the amount of data, set the image size to something small (100x100) and the framerate to 1Hz.