AravisProject/aravis #374: Does errno matter?

AndyCash1, 2020-06-10:
I've been using aravis for a while with GigEVision cameras using Go, starting with https://github.com/thinkski/go-aravis as the cgo wrapper, and its working great.

I'm now starting to use a USB3 camera, and have built aravis with USB3 support.  Since this, regardless of if I use a USB3 camera or not, `arv_camera_new` always gives errno(11), resource temporarily unavailable.  This is coming from cgo, which will return the errno for any C function call.

It doesn't look like anything in aravis is setting the errno.  And when I simply ignore the errno, it seems to work fine.  I wanted to ask if the errno is something that should be looked at, or can be safely ignored.  Thanks!

EmmanuelP, 2020-06-10:
Hi Andy,

I did not care about errno until then. Since it only happens since you have compiled usb support, and even if you don't use a USB device, I guess errno is set somewhere in arvuvinterface.

You may for now ignore errno. But it would be a good idea to try to locate which function call is setting errno to 11.

deadprogram, 2020-09-29:
Not sure if related to this specific issue, but I also get errno(11) "resource temporarily unavailable" sometimes 

I am also using the go-aravis wrapper to call Aravis from Go using CGo, however I am not using USB3 support.

This happens on many of the calls to `arv_stream_try_pop_buffer()` by way of the go-aravis wrapper function `TryPopBuffer()` https://github.com/thinkski/go-aravis/blob/master/stream.go#L29 probably when the buffer in question is not yet ready. Not sure if this is the expected behaviour here or not.

AndyCash1, 2020-09-29:
Yeah that was causing my errors as well. I ended up ignoring errno and doing a nil check, using IS_ARV_BUFFER, and also arv_buffer_get_status to validate the buffer. From looking at many of the C examples, arv_buffer_get_status is primarily used to error check, so that seemed reasonable to me.

closed 2025-02-16 by EmmanuelP