AravisProject/aravis #382: There is no way to tell if ArvGvDevice has control access over the camera

casperroo, 2020-07-08:
Currently the 0.6.4 API does not provide any function that tells whether the user has a control access to the camera (GV).
The ArvGvDeviceIOData has this information as the initialization process attempts to gain control over the camera by means of arv_gv_device_take_control function. If it fails the function yields debug warning but this is not reported to the user otherwise.
Subsequent GeniCam writes fail obviously but report timeout as the reason.
 
I propose to add a function that simply returns the private field gv_device->priv->io_data->is_controller.
Please find a patch that does exactly that attached. I am new to this project so I do not know if this is right. This method may require mutex as arv_gv_device_heartbeat_thread may update this field.
If the camera was initialized without control access, the arv_gv_device_heartbeat_thread will never emit control_lost_signal.

[0001-gvdevice-add-funtion-returning-info-on-control-acces.patch.txt](https://github.com/AravisProject/aravis/files/4890350/0001-gvdevice-add-funtion-returning-info-on-control-acces.patch.txt)

sh0, 2020-07-23:
Nice idea. Two quick comments:

- `Since: 0.6.5` should probably be `Since: 0.8.0`.
- Add `arv_gv_device_get_is_controller` to the appropriate place in `docs/reference/aravis/aravis-sections.txt` for the documentation to be visible.

I don't know about mutex issues, but this patch will probably progress better if you have time to do a pull request.

casperroo, 2020-07-24:
Thanks for the feedback. I have added pull request #393 
I have written since 0.6.5 as initially I have introduced it on the stable (0.6.4) branch. I have now ported it to 0.8.0 API

casperroo, 2020-07-30:
Resolved with #393

closed 2020-07-30 by casperroo