AravisProject/aravis #385: Aravis not recognizing USB3Vision camera

coha2902, 2020-07-10:
My camera is recognized as a USB3Vision camera on my tx2, but whenever I try to run any uv aravis command is get an error like:
arv_camera_uv_set_bandwidth: assertion 'arv_camera_is_uv_device (camera)' failed

I am using aravis 0.6. Without running any of the uv commands I am able to initialize the camera, but I am not able to pop a buffer.

EmmanuelP, 2020-07-11:
Could you attach the console output of `arv-tool-0.6 --debug all:3` ?

Did you try with aravis 0.7.5 ?

coha2902, 2020-07-13:
`Found 1 USB3Vision device (among 3 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 160.205.78.142
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = The Imaging Source Europe GmbH
name         = 
model        = DMK 33GP031
address      = 160.205.78.186
00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 80 00 00 01  ................
00000010 00 00 00 07 48 81 b3 f3 00 00 00 07 00 00 00 06  ....H...........
00000020 00 00 00 00 00 00 00 00 00 00 00 00 a0 cd 4e ba  ..............N.
00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00 a0 cd 4e fe  ..............N.
00000050 54 68 65 20 49 6d 61 67 69 6e 67 20 53 6f 75 72  The Imaging Sour
00000060 63 65 20 45 75 72 6f 70 65 20 47 6d 62 48 00 00  ce Europe GmbH..
00000070 44 4d 4b 20 33 33 47 50 30 33 31 00 00 00 00 00  DMK 33GP031.....
00000080 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000090 52 65 76 20 32 33 35 32 2c 20 46 50 47 41 20 47  Rev 2352, FPGA G
000000a0 69 67 45 33 3a 31 30 33 52 65 76 20 32 33 35 32  igE3:103Rev 2352
000000b0 2c 20 46 50 47 41 20 47 69 67 45 33 3a 31 30 33  , FPGA GigE3:103
000000c0 00 33 47 40 4d 6f 64 65 6c 3d 4d 54 39 50 30 33  .3G@Model=MT9P03
000000d0 31 5f 4d 40 00 00 00 00 00 00 00 00 00 00 00 00  1_M@............
000000e0 32 39 39 31 30 32 31 31 00 00 00 00 00 00 00 00  29910211........
000000f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

[GvInterface::discovery] Device 'TIS-29910211' found (interface 160.205.78.142) user_name '' - MAC_name 'The Imaging Source Europe GmbH-29910211'
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = The Imaging Source Europe GmbH
name         = 
model        = DMK 33GP031
address      = 160.205.78.158
00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 80 00 00 01  ................
00000010 00 00 00 07 48 81 b4 02 00 00 00 07 00 00 00 06  ....H...........
00000020 00 00 00 00 00 00 00 00 00 00 00 00 a0 cd 4e 9e  ..............N.
00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00 a0 cd 4e fe  ..............N.
00000050 54 68 65 20 49 6d 61 67 69 6e 67 20 53 6f 75 72  The Imaging Sour
00000060 63 65 20 45 75 72 6f 70 65 20 47 6d 62 48 00 00  ce Europe GmbH..
00000070 44 4d 4b 20 33 33 47 50 30 33 31 00 00 00 00 00  DMK 33GP031.....
00000080 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000090 52 65 76 20 32 33 35 32 2c 20 46 50 47 41 20 47  Rev 2352, FPGA G
000000a0 69 67 45 33 3a 31 30 33 00 00 00 00 00 00 00 00  igE3:103........
000000b0 40 54 79 70 65 3d 34 40 48 6f 75 73 69 6e 67 3d  @Type=4@Housing=
000000c0 33 33 47 40 4d 6f 64 65 6c 3d 4d 54 39 50 30 33  33G@Model=MT9P03
000000d0 31 5f 4d 40 00 00 00 00 00 00 00 00 00 00 00 00  1_M@............
000000e0 32 39 39 31 30 32 32 36 00 00 00 00 00 00 00 00  29910226........
000000f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

[GvInterface::discovery] Device 'TIS-29910226' found (interface 160.205.78.142) user_name '' - MAC_name 'The Imaging Source Europe GmbH-29910226'
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = The Imaging Source Europe GmbH
name         = 
model        = DMK 33GP031
address      = 160.205.78.147
00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 80 00 00 01  ................
00000010 00 00 00 07 48 81 b3 fc 00 00 00 07 00 00 00 06  ....H...........
00000020 00 00 00 00 00 00 00 00 00 00 00 00 a0 cd 4e 93  ..............N.
00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00 a0 cd 4e fe  ..............N.
00000050 54 68 65 20 49 6d 61 67 69 6e 67 20 53 6f 75 72  The Imaging Sour
00000060 63 65 20 45 75 72 6f 70 65 20 47 6d 62 48 00 00  ce Europe GmbH..
[... 18 lines cut ...]
address      = 160.205.78.162
00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 80 00 00 01  ................
00000010 00 00 00 07 48 81 b3 f7 00 00 00 07 00 00 00 06  ....H...........
00000020 00 00 00 00 00 00 00 00 00 00 00 00 a0 cd 4e a2  ..............N.
00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00 a0 cd 4e fe  ..............N.
00000050 54 68 65 20 49 6d 61 67 69 6e 67 20 53 6f 75 72  The Imaging Sour
00000060 63 65 20 45 75 72 6f 70 65 20 47 6d 62 48 00 00  ce Europe GmbH..
00000070 44 4d 4b 20 33 33 47 50 30 33 31 00 00 00 00 00  DMK 33GP031.....
00000080 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000090 52 65 76 20 32 33 35 32 2c 20 46 50 47 41 20 47  Rev 2352, FPGA G
000000a0 69 67 45 33 3a 31 30 33 00 00 00 00 00 00 00 00  igE3:103........
000000b0 40 54 79 70 65 3d 34 40 48 6f 75 73 69 6e 67 3d  @Type=4@Housing=
000000c0 33 33 47 40 4d 6f 64 65 6c 3d 4d 54 39 50 30 33  33G@Model=MT9P03
000000d0 31 5f 4d 40 00 00 00 00 00 00 00 00 00 00 00 00  1_M@............
000000e0 32 39 39 31 30 32 31 35 00 00 00 00 00 00 00 00  29910215........
000000f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

[GvInterface::discovery] Device 'TIS-29910215' found (interface 160.205.78.142) user_name '' - MAC_name 'The Imaging Source Europe GmbH-29910215'
PixeLINK-2F62C5ED (USB3)
TIS-29910211 (160.205.78.186)
TIS-29910215 (160.205.78.162)
TIS-29910220 (160.205.78.147)
TIS-29910226 (160.205.78.158)
`

coha2902, 2020-07-23:
Any other suggestions to try with aravis 0.6? I will be trying with aravis 0.7.5 in a few days.

EmmanuelP, 2020-07-24:
Hi,

Unfortunately, the console output was not helpful.

May be this one will help: `arv-tool-0.6 --debug all control Width`.

coha2902, 2020-07-24:
```
Found 1 USB3Vision device (among 3 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 160.205.78.142
[GvInterface::discovery] Device 'TIS-29910211' found (interface 160.205.78.142) user_name '' - MAC_name 'The Imaging Source Europe GmbH-29910211'
[GvInterface::discovery] Device 'TIS-29910220' found (interface 160.205.78.142) user_name '' - MAC_name 'The Imaging Source Europe GmbH-29910220'
[GvInterface::discovery] Device 'TIS-29910215' found (interface 160.205.78.142) user_name '' - MAC_name 'The Imaging Source Europe GmbH-29910215'
[GvInterface::discovery] Device 'TIS-29910226' found (interface 160.205.78.142) user_name '' - MAC_name 'The Imaging Source Europe GmbH-29910226'
PixeLINK-2F62C5ED (USB3)
[UvDevice::new] Vendor  = PixeLINK
[UvDevice::new] Product = D3010
[UvDevice::new] S/N     = 2F62C5ED
[UvDevice::new] Using control endpoint 1, interface 0
[UvDevice::new] Using data endpoint 2, interface 1
Get genicam
MANUFACTURER_NAME =        'PixeLINK'
MAX_DEVICE_RESPONSE_TIME = 0x00000258
DEVICE_CAPABILITY        = 0x0000000000000f09
SRBM_ADDRESS =             0x0000000000010000
MANIFEST_TABLE_ADDRESS =   0x0000000000030000
U3VCP_CAPABILITY =         0x00000001
MAX_CMD_TRANSFER =         0x00000400
MAX_ACK_TRANSFER =         0x00000400
SIRM_OFFSET =              0x0000000000020000
SI_INFO =                  0x02000000
SI_CONTROL =               0x00000000
SI_REQ_PAYLOAD_SIZE =      0x00000000004c8000
SI_REQ_LEADER_SIZE =       0x00000034
SI_REQ_TRAILER_SIZE =      0x00000020
SI_MAX_LEADER_SIZE =       0x00000034
SI_PAYLOAD_SIZE =          0x00100000
SI_PAYLOAD_COUNT =         0x00000004
SI_TRANSFER1_SIZE =        0x000c8000
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_TRAILER_SIZE =      0x00000020
MANIFEST_N_ENTRIES =       0x0000000000000001
MANIFEST ENTRY
00000000 00 00 00 01 00 04 01 01 48 00 03 00 00 00 00 00  ........H.......
00000010 8b 50 00 00 00 00 00 00 8b a5 ff ce 5b 17 19 3f  .P..........[..?
00000020 a5 9b ca da 91 29 d7 8d cd f9 59 3e 00 00 00 00  .....)....Y>....
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

genicam address =          0x0000000000030048
genicam size    =          0x000000000000508b
zip file = PixeLINK_PL_D_v2_3_18.xml
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[Genicam::create_element] Unknow tag (String)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Visibility)
[... 1234 lines cut ...]
[Genicam::create_element] Unknow tag (DisplayPrecision)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Representation)
[Genicam::create_element] Unknow tag (DisplayNotation)
[Genicam::create_element] Unknow tag (DisplayPrecision)
[Genicam::create_element] Unknow tag (Extension)
[Genicam::create_element] Unknow tag (ImposedAccessMode)
[Genicam::create_element] Unknow tag (Representation)
[Genicam::create_element] Unknow tag (DisplayNotation)
[Genicam::create_element] Unknow tag (DisplayPrecision)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Slope)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[Genicam::create_element] Unknow tag (Visibility)
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
Width = 2592 (min:96;max:2592)
```

coha2902, 2020-07-24:
@EmmanuelP Here is that ouput:
[output.txt](https://github.com/AravisProject/aravis/files/4972801/output.txt)

coha2902, 2020-07-30:
@EmmanuelP With aravis 0.7.6 the camera is being recognized as a USB camera, but I am having trouble getting the buffer to work. This block of code is in a function to initialize the camera:
```
  // Create a new buffer to store the images
    // NOTE: Check the status to ensure it was actually created
   // ArvBuffer* image_buffer = arv_buffer_new(camera_payload, nullptr);
   ArvBuffer* image_buffer = arv_buffer_new_allocate(camera_payload);

    // This should never be null, unless we run out of RAM in which case we're hosed anyway
    assert(image_buffer);

    // Pushes the image buffer to the video stream thread
    arv_stream_push_buffer(camera_stream_.get(), image_buffer);
    log_.debug("IMAGE BUFFER PUSHED TO STREAM");
```
Then later in a take photo function:
```
 // Pop the image buffer out of the video stream thread
    ArvBuffer* popped_image_buffer = arv_stream_try_pop_buffer(camera_stream_.get());
    if (!popped_image_buffer) {
        log_.debug("NO BUFFER AVAILABLE FROM VIDEO STREAM");
        return false;
    }
    else {
        log_.debug("POPPED THE IMAGE BUFFER OUT OF THE VIDEO STREAM");
    }
```
The function returns false and is never able to pop a buffer. Also when I run the check below, I always get a bad buffer status:
```
 ArvBufferStatus buffer_status2 = arv_buffer_get_status(arv_buffer_new(arv_camera_get_payload(camera_.get(),&error),NULL));
        log_.info("get_payload%s%s",error != NULL ? ": " : "",error != NULL ? error->message : "");
            g_clear_error(&error);
        if (buffer_status2 == ARV_BUFFER_STATUS_SUCCESS){
        	log_.info("Buffer status good");
        }
        else{
        	log_.info("Buffer status bad");
        }
```

EmmanuelP, 2020-07-31:
What gives `./tests/arv-camera-test` ?

coha2902, 2020-08-03:
I can't run arv-camera-test currently because it is not flashed on my tx2. I have tracked my issue down to the fact that when I try to run ```arv_camera_set_triggers```, ```arv_camera_set_trigger_source```, or ```arv_camera_clear_triggers``` I get ```[ArvDevice::write_memory] Timeout``` from the GError placeholder of the function. I am therefore unable to change my trigger source to software as necessary and it defaults to Line0.

antortjim, 2021-05-10:
Dear aravis developers

My a2A2590-60umBAS USB3 camera from Basler is not getting recognized by aravis, even though it works as expected when using Pylon Viewer

### OS & Environment
```
Ubuntu 20.04
Python 3.7.4
conda 4.10.1
gcc 9.3.0
g++ 9.3.0
meson 0.58.0
ninja 1.10.0.git.kitware.jobserver-1
```

### How I installed aravis

```
git clone https://github.com/AravisProject/aravis/
pip install meson
pip install ninja
apt install libglib2.0-dev
apt install libxml2-dev
```

### How I launch aravis

From the `build` folder created during install:

```
LD_LIBRARY_PATH=./src/:$LIBRARY_PATH /usr/local/bin/arv-camera-test-0.8
# Looking for the first available camera
# No camera found: No supported device found
```

I also tried passing the name of the camera by adding ` --name     a2A2590-60umBAS` but I get this
```
# Looking for camera 'a2A2590-60umBAS'
# No camera found: Device 'a2A2590-60umBAS' not found
```

Running in debug mode

```
LD_LIBRARY_PATH=./src/:$LIBRARY_PATH /usr/local/bin/arv-tool-0.8 --debug all:3
# [18:56:31.052] 🅸 interface> [GvDiscoverSocket::new] Add interface 127.0.0.1
# [18:56:31.053] 🅸 interface> [GvDiscoverSocket::new] Add interface 10.34.178.0
# No device found
```

This does not change if I run either of the commands as `root`

The camera is listed in the output of `lsusb -t` as a `Miscellaneous Device`
```
lsusb -t

/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 4, If 0, Class=Miscellaneous Device, Driver=, 5000M
    |__ Port 2: Dev 4, If 1, Class=Miscellaneous Device, Driver=, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 1: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
    |__ Port 3: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/8p, 480M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=ehci-pci/2p, 480M
    |__ Port 1: Dev 7, If 0, Class=Hub, Driver=hub/6p, 480M

```

PS I don't know how to link the `libaravis-0.8.so.0` file available in `src`, but the `LD_LIBRARY_PATH` trick does the job (not ideal, I know).

Could you please let me know how to fix this issue? Thank you very much!
Antonio

EmmanuelP, 2021-05-10:
The development package for libusb is not installed (libusb-1.0-0-dev).

antortjim, 2021-05-10:
https://github.com/AravisProject/aravis/issues/507