AravisProject/aravis #395: Use Aravis for multi-camera acquisition

plumpmango, 2020-07-28:
Hello,
As part of my internship I have to use GigE cameras for online video acquisition. I'm starting from a work based on Aravis 0.5 to do online acquisition with the first available camera. For my part I modified the code so that I can select a camera among several and launch the acquisition. This works correctly.

The goal is to have an application that can launch at least two network cameras at the same time and make continuous acquisition. Currently when I try to acquire with both cameras the first camera launched works well. On the other hand, the second one recovers empty images. I still generate two videos: the first one is noisy when the double acquisition is launched and the second one is corrupted with a message "impossible to multiplex".

To manage the signals I use the slots in QT. 

Looking at the camera documentation I first thought of a problem with the delay between packages, so I modified it as suggested but to no avail.

As I couldn't find any example of multi-camera GigE acquisition with Aravis I don't know what I missed. Can you help me?

Thanks

Clémence

plumpmango, 2020-07-30:
I finnally found a solution : it necessary to set up the frequency of acquisition and the inter packet delay for cameras i use (Smartek). It is explain in Smartek camera documentation. For the moment I can't run the two camera acquisition with frequency > 10 Hz. But it is a beginning.

EmmanuelP, 2020-07-30:
Hi Clémence,

I don't understand what you mean with aravis 0.5. Is this really the version you meant ?

Anyway I'm glad you found a solution.

In the future, please ask your question on the aravis discourse instance, which is better suited for subject like this.

https://aravis-project.discourse.group/

Cheers.

closed 2020-07-30 by EmmanuelP