AravisProject/aravis #398: How to access "Set Exposure Time Mode"

lionfish0, 2020-07-30:
The [camera](https://en.hikrobotics.com/en/Hikrobotics/Machine%20Vision/01%20Product/%E5%B7%A5%E4%B8%9A%E9%9D%A2%E9%98%B5%E7%9B%B8%E6%9C%BA/CA/spec/MV-CA032-10GMGC_datasheet_20191127.pdf) I'm using has a feature to go down to 1μs exposure using something called the "UltraShort exposure mode: 1 μs to 14 μs".

If I set self.aravis_camera.set_exposure_time(1) the exposure seems to stick at 15μs.

The [manual](https://en.hikrobotics.com/en/Hikrobotics/Machine%20Vision/01%20Product/%E5%B7%A5%E4%B8%9A%E9%9D%A2%E9%98%B5%E7%9B%B8%E6%9C%BA/CA/%E7%94%A8%E6%88%B7%E6%89%8B%E5%86%8C/UD19137B_GigE%20Area%20Scan%20Camera%20User%20Manual_V3.4.9_20200710.pdf) [p59] has a brief mention of this: They set something called the "Exposure time mode" to "ultrashort" instead of "standard".

I can't see any mention of either of these key words in the aravis docs. Presumably I can edit or access some register somewhere, but how do I know which register I need to alter?

Thanks for the great module.

EmmanuelP, 2020-07-31:
Hi,

Thanks for your interrest in Aravis.

This issue tracker is not a support forum. Please ask your question on aravis discource instance: https://aravis-project.discourse.group/

Cheers,

    Emmanuel.

lionfish0, 2020-08-05:
Just to answer my own question for anyone who might find this issue. [This page](https://docs.baslerweb.com/exposure-time) had some useful information. 

My code for aravis (with python) is now:
```
        aravis_device = self.aravis_camera.get_device();
        aravis_device.set_string_feature_value("ExposureTimeMode","UltraShort")   
        self.aravis_camera.set_exposure_time(7) #us
        self.aravis_camera.set_gain(0)
        self.aravis_camera.set_pixel_format (Aravis.PIXEL_FORMAT_MONO_8)
        self.aravis_camera.set_trigger("Line1")     
        
        #to delay exposure
        aravis_device.set_string_feature_value("TriggerSelector","FrameStart")   
        aravis_device.set_float_feature_value("TriggerDelay",2.0) #us
```

closed 2020-08-05 by lionfish0