AravisProject/aravis #399: FLIR Blackfly S BFS-PGE-122S6M-C not working

abrock, 2020-08-12:
**Describe the bug**

Aravis is not getting any images / genicam data from the camera.

**To Reproduce**
Steps to reproduce the behavior:

1. Connect the camera via GigE to a network interface of the computer (no switch / router involved). 
2. Power it up and wait until it gets and IP address via DHCP. Make sure it respond to `arping <IP>`
3. Run arv-viewer.

Result:
```
arv-viewer-0.8 -d all
Found 0 USB3Vision device (among 13 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 10.3.99.139
[GvDiscoverSocket::new] Add interface 192.168.250.1
[GvDiscoverSocket::new] Add interface 10.0.3.1
[GvDiscoverSocket::new] Add interface 192.168.0.2
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = FLIR
name         = 
model        = Blackfly S BFS-PGE-122S6M
address      = 192.168.0.193
00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 80 00 00 01  ....
00000010 00 00 2c dd a3 31 87 d7 00 00 00 07 00 00 00 06  ..,.
00000020 00 00 00 00 00 00 00 00 00 00 00 00 c0 a8 00 c1  ....
00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff ff 00  ....
00000040 00 00 00 00 00 00 00 00 00 00 00 00 c0 a8 00 02  ....
00000050 46 4c 49 52 00 00 00 00 00 00 00 00 00 00 00 00  FLIR
00000060 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ....
00000070 42 6c 61 63 6b 66 6c 79 20 53 20 42 46 53 2d 50  Blac
00000080 47 45 2d 31 32 32 53 36 4d 00 00 00 00 00 00 00  GE-1
00000090 31 37 30 37 2e 33 2e 31 2e 30 00 00 00 00 00 00  1707
000000a0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ....
000000b0 4f 63 74 20 31 31 20 32 30 31 37 20 31 31 3a 31  Oct 
000000c0 34 3a 31 32 00 00 00 00 00 00 00 00 00 00 00 00  4:12
000000d0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ....
000000e0 32 30 30 32 33 32 35 35 00 00 00 00 00 00 00 00  2002
000000f0 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ....

[GvInterface::discovery] Device 'FLIR-Blackfly S BFS-PGE-122S6.168.0.2) user_id '' - MAC '2c:dd:a3:31:87:d7'
```

4. Click on the fake camera. Immediately pixel format etc. is available and video acquisition can be started.
5. Click on the BFS camera. Wireshark does not show any packets, neither send nor receive. The gui freezes for ~6 seconds and after that the line is selected but all the controls are grey and video acquisition can not be started. arv-viewer-0.8 show this output:

```
[GcRegisterNode::_set_integer_value] address = 0x124, value = 0x0
[GcCommand::execute] AcquisitionStop (0x0)
Found 0 USB3Vision device (among 13 USB devices)
[GvDevice::new] Interface address = 192.168.0.2
[GvDevice::new] Device address = 192.168.0.193
packet_type  = cmd
packet_flags = ack-required
command      = read-memory-cmd
size         = 8
id           = 65301
address      =        512 (0x00000200)
size         =        512 (0x00000200)
00000000 42 01 00 84 00 08 ff 15 00 00 02 00 00 00 02 00  B...............

[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
packet_type  = cmd
packet_flags = ack-required
[... 351 lines cut ...]
id           = 65303
address      =       2560 (0x00000a00)
value        =          0 (0x00000000)
00000000 42 01 00 82 00 08 ff 17 00 00 0a 00 00 00 00 00  B...............

[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
Failed to open device 'FLIR-Blackfly S BFS-PGE-122S6M-20023255': Invalid Genicam data
```

Wireshark only shows two packets ever, first discovery then the answer from the camera [discovery.zip](https://github.com/AravisProject/aravis/files/5063098/discovery.zip)

**Expected behavior**

arv-viewer showing video acqusition, more packets sent by arv-tool to get genicam data etc.

**Camera description:**
 - Manufacturer: FLIR
 - Model: BFS-PGE-122S6M-C
 - Interface: GigE

**Platform description:**
 - Aravis version latest git 18d0eb8e1dc91ecfd2641dc1929408e70ff703e1
 - OS: Ubuntu 20.04
 - Hardware x86_64

EmmanuelP, 2020-08-12:
Hi,

Thanks for the detailed bug report. I'm puzzled by the wireshark capture. We should at least the packet sent by aravis for the read requests. Do you have an active firewall service ?

abrock, 2020-08-12:
Hi,

Thank you for the fast answer.
I found that Wireshark captures more packets when selecting interface "any" instead of the specific interface which is super weird. Now I see 6x READMEM_CMD followed by 5x WRITEREG_CMD with 0.5s between each packet but I never see any answers:
[readwrite.zip](https://github.com/AravisProject/aravis/files/5064966/readwrite.zip)

I also looked into the running services and shut some of them down, that made no difference. I also ran arv-tool as root, also no difference.

```
 [ + ]  acpid           - power management
 [ + ]  apparmor        - access control? I stopped it
 [ + ]  apport          - crash reports
 [ + ]  atd             - periodic command scheduler
 [ + ]  avahi-daemon
 [ + ]  binfmt-support
 [ + ]  bluetooth
 [ + ]  cron
 [ + ]  cups
 [ + ]  cups-browsed
 [ + ]  dbus
 [ + ]  dnsmasq
 [ + ]  grub-common
 [ + ]  haveged         - entropy daemon
 [ + ]  irqbalance
 [ + ]  kerneloops      - collect kernel crash info
 [ + ]  kmod            - userspace toold for kernel management
 [ + ]  lxcfs           - userspace filesystem
 [ + ]  mono-xsp4       - mono webserver, stopped it
 [ + ]  multipath-tools - partition named
 [ + ]  networking      - disabled it
 [ + ]  procps          - process status utilities
 [ + ]  rsyslog 
 [ + ]  sddm            - simple desktop display manager
 [ + ]  ssh
 [ + ]  sysstat
 [ + ]  udev
 [ + ]  unattended-upgrades
 [ + ]  virtualbox
 [ + ]  whoopsie
```

abrock, 2020-08-13:
I found the problem: My routing setup was wrong, the setup script only set the route for the (fixed IP) camera I usually use with that network interface. Once I set the route correctly I got images without any problems using arv-viewer-0.8. I'm sorry for the noise.

closed 2020-08-13 by abrock

EmmanuelP, 2020-08-13:
Thanks for the follow up.

Cheers.