AravisProject/aravis #400: There is no way to relinquish control access without finalizing the whole camera

Ekleog, 2020-08-18:
**Is your feature request related to a problem? Please describe.**

A way to know whether we currently have control access has been made available with https://github.com/AravisProject/aravis/issues/382. Unfortunately, there is currently no good way to relinquish control access without finalizing the whole camera.

**Describe the solution you'd like**

Would it make sense to just make `arv_gv_device_take_control` and `arv_gv_device_leave_control` non-`static`?

**Describe alternatives you've considered**

Not doing this, but then it would mean the only way to relinquish control access on the camera is to actually finalize the whole camera, which is especially unpleasant in affine-typed languages (like Rust) where we can't have uninitialized variables hanging around.

**Additional context**

Well… as usual, thank you for all the work put on aravis! (and yes, I'll eventually find the time to come back to #396, sorry for the delay…)

EmmanuelP, 2020-08-19:
Hi,

> Would it make sense to just make arv_gv_device_take_control and arv_gv_device_leave_control non-static?

It is not as simple as that. If we want to expose this control API, we have to take care to handle correctly the heartbeat thread.

Also, it would be the opportunity to implement a lazy control access mode. ArvGvdevice would gain device control access only for write or command actions, and would release the control access after a timeout, or when the device is finalized. This default mode could exist in addition to an explicit control access.

lovesegfault, 2020-09-15:
@EmmanuelP Any idea when this can/will be landed?

Thanks!