AravisProject/aravis #402: Problem installing latest 0.6.4 on ubuntu 18.04

JTray123, 2020-08-19:
Hi all,

I downloaded 0.6.4 from http://ftp.acc.umu.se/pub/GNOME/sources/aravis/0.6/ and installed it used the following commands

./configure --enable-viewer
sudo make
sudo make install

Problem seems to be with sudo make install where it says it can create a particular directory or file as it already exists

Here is a snapshot of the error. I had a look in the file that does exist and it looks like it has information to do with paho/ mqtt that I use for another project. Any idea on how best to proceed?

make[2]: Entering directory '/home/user/Downloads/LATEST-IS-0.6/aravis-0.6.4/man'
make[2]: Nothing to be done for 'install-exec-am'.
 /bin/mkdir -p '/usr/local/share/man/man1'
/bin/mkdir: cannot create directory ‘/usr/local/share/man/man1’: File exists
Makefile:424: recipe for target 'install-man1' failed
make[2]: *** [install-man1] Error 1
make[2]: Leaving directory '/home/user/Downloads/LATEST-IS-0.6/aravis-0.6.4/man'
Makefile:520: recipe for target 'install-am' failed
make[1]: *** [install-am] Error 2
make[1]: Leaving directory '/home/user/Downloads/LATEST-IS-0.6/aravis-0.6.4/man'
Makefile:582: recipe for target 'install-recursive' failed
make: *** [install-recursive] Error 1

EmmanuelP, 2020-08-19:
Hi,

Any reason you are not using the latest stable release (0.8.0) ?

Anyway, this should not happen. If I recall correctly, the installation of the man page should be pretty standard stuff. I'll have a look in order to understand this error. Meanwhile, any help would be appreciated !

johntraynor, 2020-08-19:
Many thanks for the reply. I went looking for 0.8 but couldn't find it. Could you provide a working link and I'll try and build with meson. If we can get 0.8 working then i'm not too worried about getting this resolved

EmmanuelP, 2020-08-19:
> I went looking for 0.8 but couldn't find it.

That is a good reason. I forgot to upload the tarball to the release area. It should be available soon.

johntraynor, 2020-08-19:
Ah - I thought i was loosing it -. You might reply to this post when it's there. Thanks

EmmanuelP, 2020-08-20:
The tarball is avaliable here: https://ftp.gnome.org/pub/GNOME/sources/aravis/0.8/

johntraynor, 2020-08-20:
Many thanks - No problems getting the viewer working using the latest build with meson

EmmanuelP, 2020-08-20:
I have checked how man pages are installed using autotools. It looks like pretty standard stuff, so there is nothing I can do in aravis.

Thanks for the report.

closed 2020-08-20 by EmmanuelP