AravisProject/aravis #403: JAI Camera node 'AcquisitionFrameRateAbs' not found

lclayton2002, 2020-08-20:
Hi Emmanuel,
                       As requested, attached
[JAI Features.txt](https://github.com/AravisProject/aravis/files/5101106/JAI.Features.txt)

 is the output from arv-tool-0.8 features.
Regards
Les

EmmanuelP, 2020-08-20:
They have a funky way to set the acquisition frame rate.

```
    Category    : 'JAIAcquisitionAndTriggerControl'
        Enumeration : 'JAIAcquisitionFrameRate'
            EnumEntry   : 'fps1_8'
            EnumEntry   : 'fps1_4'
            EnumEntry   : 'fps1_2'
            EnumEntry   : 'fpsNormal'
```

There is 2 changes we can do in aravis:

* Implement a special case for the frame rate setting. Could you attach the output of: `arv-tool-0.8 values` and also the first line of `arv-tool-0.8 genicam` ?

* In the gstreamer plugin, we should not make the plugin initialization fail if feature like framerate can not be set, but just show a warning. That way, even if a feature is not handled correctly in the ArvCamera API, it should still be possible to set the right feature values using the `features` plugin property.

lclayton2002, 2020-08-20:
Hi Emmanuel,
                      In my use case for Aravis showing a warning would suffice. Though I am an experience C# / Java coder, my experience programming in C is Zero. Though taking a look at your arvcamera.c file, the level of comments and structure is pretty easy to follow.   Data attached.
Thanks Les
[arv-tool-genicam.txt](https://github.com/AravisProject/aravis/files/5101429/arv-tool-genicam.txt)
[arv-tool-values.txt](https://github.com/AravisProject/aravis/files/5101430/arv-tool-values.txt)