AravisProject/aravis #407: Why arv_buffer_get_data returns const pointer?

casperroo, 2020-09-21:
arv_buffer_get_data returns const void * meaning that the caller cannot modify the contents of the data. This creates some problems:
1. If used with opencv there is no way (without casting the const away) to use it as a data for cv::Mat as they have no overload that takes const data. BTW: I know - the goal of this project is not to solve openscv's problems :) 
2. Generally if one needs to make some modification of the data (say simple in-place threshold) one must create a copy of the data.

So generally speaking when using aravis one should always create copy of the buffer data before touching it. In some scenarios this is not optimal.

Is there a reason the data returned by the arv_buffer_get_data must be unchanged and therefor casting the const away is unsafe? Or can it be safely casted away? If yes wouldn't it be save to change the API and make the function returning non-const pointer?

Out of curiostity I have checked Basler's Pylon: their pyl::CGrabResultPtr::GetBuffer() returns non-const void pointer.

EmmanuelP, 2020-09-24:
Hi,

`const` is here to prevent a modification of the content of the data buffer outside of ArvBuffer API. Most of the time, there is no need to modify the data in place.

In C, it is mostly an 'advice', as if you know what you are doing, you always have the possibility to cast it to `void *`. That is the right workaround for a bogus API for example, or if you are applying in place modifications when you are the only owner of an ArvBuffer.

Cheers,

Emmanuel.

closed 2020-09-24 by EmmanuelP

casperroo, 2020-09-24:
I use C++ and casting away the const is ugly but I can live with it.
But I understand that writing to this buffer (after casting the const away) is actually safe, right?

EmmanuelP, 2020-09-24:
> But I understand that writing to this buffer (after casting the const away) is actually safe, right?

Yes, the data buffer used in ArvBuffer is not supposed to change, so the returned pointer is supposed to be valid during all the life of the parent ArvBuffer.