AravisProject/aravis #408: arv-viewer-0.8 showing black screen/ 0FPS in arv-camera-test

westonCoder, 2020-09-21:
**Describe the bug**
I have connected my gige vision camera (Imperx c2010 PoE 3MP CMOS camera) and I have installed aravis 0.8 and all dependencies. 

Bug 1:
when running ./arv-viewer-0.8 the camera populates a row correctly, but when I select it and press play button all I see is black screen. 

Bug 2:
when running ./arv-camera-test I get the following (0 FPS, no frames being captured):
Looking for the first available camera
vendor name = Imperx, Inc
model name = POE-C2010C-RC0000000
device id = <device serial number omitted>
image width = 2064
image height = 1544
horizontal binning = 0
vertical binning = 0
payload = 3186816 bytes
exposure = 0us
gain = 0 dB
gv n_stream channels = 1
gv current channel = 0
gv packet delay = 0ns
gv packet size = 7168 bytes
 0 frames/s - 0MiB/s
 0 frames/s - 0MiB/s
 0 frames/s - 0MiB/s
 0 frames/s - 0MiB/s
 0 frames/s - 0MiB/s
^C 0 frames/s - 0MiB/s
Completed buffers = 0
Failures = 0
Underruns = 0

**To Reproduce**
1. download aravis/ dependencies
2. build repo
3. connect "Imperx c2010 PoE 3MP CMOS camera" to ethernet port
4. run both arv-viewer-0.8, and arv-camera-test

**Expected behavior**
I expected to see frames being received (36FPS on manufacturer website), and to see video when running viewer.

**Camera description:**
 - Manufacturer = Imperx
 - Model = c2010
 - Interface = [ethernet (gigE vision)]

**Platform description:**
 - Aravis version = 0.8
 - OS = Ubuntu 18.04.4 LTS
 - Hardware = aarch64, running on jetson xavier AGX

**Additional context**
I have got the camera up and running (showing video) with the manufacturers SDK just fine. The reason I need aravis is I need to plug the stream into a gstreamer pipeline, and their SDK doesn't provide that functionality.

westonCoder, 2020-09-21:
attached these files I have piped all of the debug info to!
I hope this helps solve this mystery :)

[viewer_debug.txt](https://github.com/AravisProject/aravis/files/5258348/viewer_debug.txt)
[arv-camera-test_debug.txt](https://github.com/AravisProject/aravis/files/5258350/arv-camera-test_debug.txt)

EmmanuelP, 2020-10-29:
I'm seeing in the log that packet socket is enabled. Could you try to disable it ?

EmmanuelP, 2020-10-29:
Did you try aravis with this device on a intel platform ?

westonCoder, 2020-10-30:
Yes I had enabled it I believe. I could try disabling it on Monday!

I haven't tried it with an intel computer because my use case is on the Jetson. If it would help solve the issue, I could try it out on one.