AravisProject/aravis #412: Add support for Mono12Packed format

MarkRivers, 2020-09-23:
Some cameras offer support for PixelFormat=Mono12Packed.  For cameras with 10 or 12-bit ADCs this format can preserve the full resolution of the ADC while using only 75% of the network bandwidth required by PixelFormat=Mono16.  This allows higher frame rates for many cameras.

However, most higher-level applications need to have the Mono12Packed format unpacked to 16-bit integers on the host before they can handle it.   The Spinnaker SDK from FLIR and the Vimba SDK from Allied Vision Technologies provide functions to do this conversion.   It would be nice if aravis provided such a function as well.

abrock, 2021-04-30:
I'm debugging an issue with unpacking Mono12p and noticed that there is a difference between Mono12p and Mono12Packed which took me by surprise:

https://support.thinklucid.com/knowledgebase/pixel-formats/#mono12p

MarkRivers, 2021-04-30:
ADAravis now supports both Mono12p and Mono12Packed.  This was added in ADAravis R2-1 and R2-2.

These are controlled width the widgets in the lower-right of this screen shot:

[image]

abrock, 2021-05-01:
@MarkRivers thank you for providing the implementation. I'm trying to verify the correctness of two other implementations (for Mono12p). I don't know those "epics" types, if I replace "epicsUInt8" by "unsigned char" and "epicsUInt16" by "uint16_t" would your methods still work correctly? Do you have example data which can be used to verify the correctness of an implementation?

MarkRivers, 2021-05-01:
> I don't know those "epics" types, if I replace "epicsUInt8" by "unsigned char" and "epicsUInt16" by "uint16_t" would your methods still work correctly? 

Yes, epicsUInt8 is the same as unsigned char, and epicsUInt16 is the same as uint16_t.  EPICS supports some very old compilers (pre C99) that don't define uint16_t, for example, which is why I use the EPICS types.

> Do you have example data which can be used to verify the correctness of an implementation?

No, I don't think so.  I don't have code for a compressor, only for a decompressor.  I tested by using FLIR and Allied Vision Technology cameras that can send either Mono16, Mono12p, or Mono12Packed data.  I verified that the decompressed Mono12p and Mono12Packed data was the same as the Mono16.  I actually found some subtle differences for the FLIR camera, and asked their support team.  It turns out there are minor differences for their cameras when the camera is doing gamma correction, but that is an issue with their firmware, not with the decompressor.

MarkRivers, 2021-05-01:
@abrock I just realized that this issue is in the AravisProject, not in my EPICS ADAravis driver! https://github.com/areaDetector/ADAravis

I said:
> I should have closed this issue.

I should not have said that, because while the problem is fixed for my EPICS ADAravis driver, it is not fixed for aravis in general.

EmmanuelP, 2021-05-03:
For the record, here is the conversion code:

https://github.com/areaDetector/ADGenICam/blob/8c102bdcba242f6d0f96b3bf02c6331b8d7b27f3/GenICamApp/src/ADGenICam.cpp#L544-L589

>>    I should have closed this issue.
> I should not have said that, because while the problem is fixed for my EPICS ADAravis driver, it is not fixed for aravis in general.

@MarkRivers I don't really plan to work on format conversion in Aravis. So unless someone proposes and implements a format conversion API, it will not happen. As you have already fixed this issue for your use case, I'm closing it.

closed 2021-05-03 by EmmanuelP