AravisProject/aravis #415: arv-viewer-0.8 crash on start on macos

avilleret, 2020-10-08:
**Describe the bug**
arv-viewer-0.8 crash on start on macos

**To Reproduce**

```
git clone https://github.com/AravisProject/aravis
cd aravis
meson build
ninja -C build
cd build/viewer 
./arv-viewer-0.8
```

Then I got a segfault.
[arv-viewer-0.8.log](https://github.com/AravisProject/aravis/files/5347148/arv-viewer-0.8.log)

**Expected behavior**
A window should open.

**Camera description:**
No camera is present on the network.

**Platform description:**
 - Aravis version ARAVIS_0_8_1-9-g27ef1ef
 - OS: macos 11.0 Beta (20A5374i)
 - Hardware MacBook Pro (15-inch, 2019)

**Additional context**
The crash log is attached

EmmanuelP, 2020-10-08:
Hi Antoine,

Thanks for the report.

Looking at the crash backtrace, it looks like the crash happen in the user interface code, perhaps in gtk, may be in arvviewer.c.
Unfortunately, I don't have an OSX platform at hand to debug this issue. Any help in finding the root cause of the crash is welcome.

What is the gtk version you are using ? Do you have working gtk based application on your system ?

EmmanuelP, 2021-02-09:
I'm able to run the viewer on macOS Catalina, on a 2015 macbook pro. Gtk3 and GStreamer were installed using brew.

Please reopen if you still have the issue and the requested informations.

closed 2021-02-09 by EmmanuelP