AravisProject/aravis #416: Cleaning up to make mingw64 compilation straightfowards

eudoxos, 2020-10-09:
Far from being a fan of the Windows platform, I am under pressure to compile a code using Aravis for it. Using [crossroad](https://pypi.org/project/crossroad/) makes cross-compilation with gcc/mingw64 and meson (and installation of all dependencies via msys2 packages) feasible.

The compilation of aravis currently fails in several places, mostly because of platform-specific headers.  I was able to work around most of them, but would like to have some guidance and a go-ahead from the project so that those few changes can be merged later:

* Some header inclusion should be guarded (`#ifndef __MINGW32__` or similar; this concerns e.g. `sys/resource.h` and `sys/syscall.h`), similar to `#ifndef __APPLE__` which already appears in a few places.
* Some headers would conditionally use alternatives (`sys/socket.h` → `winsock.h`, for `setsockopt`).
* Some non-essential components using non-portable headers (`arvgvfakecamera.c`) would be excluded from the build.
* Some tests (`test/realtimetest.c`) would trivially fail.
* Features like documentation, viewer, introspection will not be cross-compilable (that is okay, for the start or forever).

The first step would be to make the code compile (for all platforms, incl. mingw), later test how functional it is on a native platform.

Could you suggest what preprocessor macros to use to guard platforms? It seems that mingw compiler unfortunately predefines `_POSIX_VERSION` (which (should normally be used)[https://stackoverflow.com/a/11351171/761090]) despite not providing `sys/...` headers. Perhaps combining `__APPLE__` and `__MINGW32__` (which is also defined by mingw64) would be sufficient, for the start.

NB: this is **not** about compiling Aravis using MSVC.

EmmanuelP, 2020-10-09:
Hi,

Thanks for your work on this.

I'm perfectly fine with a patch set that makes aravis compile on Windows, even with a reduced feature set. It would be nice if you could make also fakecamera compilable though.

There is a glib macro that detects Windows, G_OS_WIN32. I don't know if that help.

It would also be great if you could manage to configure a CI pipeline for the Windows build.

eudoxos, 2020-10-09:
The last missing bit seems to be `getifaddrs` which is non-portable. Some of the parts could be replaced by more portable `getaddrinfo`, except of the bit where netmask is queried (I don't know how to do that). What do you prefer? Use `getifaddrs` for unices and native windows API in windows, or do as much as possible via unified code with `getaddrinfo`?

eudoxos, 2020-10-09:
I am suggesting to have a single utility function enumerating network interfaces, returning GList/GArray of `struct { struct sockaddr addr, netmask, broadcast };`. It would have 2 implementations, one using `getifaddrs` for Linux/Mac, another one using [GetInterfaceInfo](https://docs.microsoft.com/en-us/windows/win32/api/iphlpapi/nf-iphlpapi-getinterfaceinfo) for Windows (I take care of that one). The code is in 3 cases exactly the same, including filtering (i.e. skipping ifaces without address, p2p, inactive, non-AF_INET); the last case (in arvgvstream) could consume the result as well, as far as I see.

This would put the platform-dependent enumeration in a single place and reduce code duplication as a bonus. Fakecamera would be automatically compatible as well.

Where should this utility function go? arvmisc?

These are the 4 places where interfaces are enumerated:

https://github.com/AravisProject/aravis/blob/8fa1071b835b5dea1221beb76038112a70505977/src/arvgvinterface.c#L97

https://github.com/AravisProject/aravis/blob/8fa1071b835b5dea1221beb76038112a70505977/src/arvgvinterface.c#L495

https://github.com/AravisProject/aravis/blob/d80d7b207771bc5605be2c488736040bba742248/src/arvgvstream.c#L857

https://github.com/AravisProject/aravis/blob/677b07af80b612f67d1b24b74e7eed780f1256dc/src/arvgvfakecamera.c#L485

EmmanuelP, 2020-10-09:
> I am suggesting to have a single utility function enumerating network interfaces

Good idea. Code factorization when it makes sense is always welcome.

> Where should this utility function go? arvmisc?

arvmisc.c, arvmiscprivate.h is the right place.

Thanks.

NerdToMars, 2020-10-27:
@eudoxos Hi, thanks for your contribution. I have successfully compiled the code in mingw64. My compiler is gcc.exe (Rev1, Built by MSYS2 project) 10.2.0;  I got this error: 

> FAILED: src/libaravis-0.8-0.dll.p/arvnetwork.c.obj
"cc" "-Isrc/libaravis-0.8-0.dll.p" "-Isrc" "-I../src" "-IC:/developing/msys2/mingw64/include/glib-2.0" "-IC:/developing/msys2/mingw64/lib/glib-2.0/include" "-IC:/developing/msys2/mingw64/include" "-IC:/developing/msys2/mingw64/include/libxml2" "-IC:/developing/msys2/mingw64/include/libusb-1.0" "-fdiagnostics-color=always" "-pipe" "-D_FILE_OFFSET_BITS=64" "-Wall" "-Winvalid-pch" "-g" "-Wduplicated-branches" "-Wimplicit-fallthrough" "-Wmisleading-indentation" "-Wstrict-prototypes" "-Wunused" "-Wdeclaration-after-statement" "-Wformat=2" "-Wimplicit-function-declaration" "-Winit-self" "-Wmissing-include-dirs" "-Wmissing-prototypes" "-Wpointer-arith" "-pthread" "-mms-bitfields" "-mms-bitfields" "-mms-bitfields" "-DARAVIS_COMPILATION" "-DARAVIS_DATA_DIR=\"C:/developing/msys2/mingw64/share/aravis-0.8\"" "-DARAVIS_API_VERSION=\"0.8\"" -MD -MQ src/libaravis-0.8-0.dll.p/arvnetwork.c.obj -MF "src/libaravis-0.8-0.dll.p/arvnetwork.c.obj.d" -o src/libaravis-0.8-0.dll.p/arvnetwork.c.obj "-c" ../src/arvnetwork.c
../src/arvnetwork.c: In function 'arv_enumerate_network_interfaces':
../src/arvnetwork.c:41:3: warning: #warning arv_enumerate_network_interface not yet implemented for WIN32 [-Wcpp]
   41 |  #warning arv_enumerate_network_interface not yet implemented for WIN32
      |   ^~~~~~~
../src/arvnetwork.c: At top level:
../src/arvnetwork.c:51:1: error: conflicting types for 'inet_ntop'
   51 | inet_ntop (int af, const void *src, char *dst, socklen_t cnt)
      | ^~~~~~~~~
In file included from C:/developing/msys2/mingw64/include/glib-2.0/gio/gnetworking.h:30,
                 from ../src/arvnetworkprivate.h:35,
                 from ../src/arvnetwork.c:23:
C:/developing/msys2/mingw64/x86_64-w64-mingw32/include/ws2tcpip.h:451:35: note: previous declaration of 'inet_ntop' was here
  451 | WINSOCK_API_LINKAGE LPCSTR WSAAPI InetNtopA(INT Family, LPCVOID pAddr, LPSTR pStringBuf, size_t StringBufSize);

Then I refer this link resolved this error: https://stackoverflow.com/questions/60894310/conflicting-types-for-inet-ntop

Is that because I used a different GCC compiler compare to yours? Or any other reasons?

eudoxos, 2020-10-27:
Thanks for testing, cool.

> Is that because I used a different GCC compiler compare to yours? Or any other reasons?
The declaration is in `ws2tcpip.h`, in a block guarded by `#if (_WIN32_WINNT >= 0x0600)`:

https://github.com/mirror/mingw-w64/blob/12b5d1344c23c5fdba7fb2c771d0476c2220db16/mingw-w64-headers/include/ws2tcpip.h#L317-L462

I also have it in my headers (`mingw-w64-x86-64-dev` Debian package) but somehow the `_WIN32_WINNT` version must differ.

In my installation, I changed the symlink `x86_64-w64-mingw32-gcc` to point to `x86_64-w64-mingw32-gcc-posix` (rather than `*-win32` using `update-alternatives` in Debian) to be able to use `std::thread` and others in c++. That might explain the difference.

The best might be to guard `inet_ntoa` in Aravis by `#if !defined(_WIN32_WINNT) || (_WIN32_WINNT < 0x0600)`.

eudoxos, 2020-11-03:
@NerdToMars: see #441, that should solve the problem. Fixes for the viewer (and more) are in #422 .

eudoxos, 2021-07-12:
This can be closed, I believe; https://github.com/AravisProject/aravis/issues/525 is a separate (and trivial) issue.

closed 2021-07-12 by eudoxos