AravisProject/aravis #418: FLIR A35 support

zyp, 2020-10-09:
Hi, I've got a FLIR A35 and I'm running into the same issue as in #403, but I'm filing it as a new issue since this is a different camera.

    ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Could not read camera "": node 'AcquisitionFrameRateAbs' not found

As far as I can see this is how this camera expects the framerate to be set:

    Category    : 'Mono8VideoControl'
        Enumeration : 'SensorVideoStandard'
            EnumEntry   : 'PAL50HZ'
            EnumEntry   : 'NTSC60HZ'
            EnumEntry   : 'PAL25Hz'
            EnumEntry   : 'NTSC30HZ'

For my use I'd be happy if the plugin just skipped setting the framerate when it can't find `AcquisitionFrameRateAbs`.

EmmanuelP, 2020-11-06:
> For my use I'd be happy if the plugin just skipped setting the framerate when it can't find AcquisitionFrameRateAbs.

That is implemented in #443. It would be great if you could test this patch.

EmmanuelP, 2021-09-16:
Please test Ararvis 0.8.17 and reopen this issue if it is not fixed.

closed 2021-09-16 by EmmanuelP