AravisProject/aravis #419: aravissrc blacklisted by gstreamer

deltatango2, 2020-10-09:
First off, thank you for this effort.  I hope to be able to use your gstpipeline and contribute to the project.

**Describe the bug**
gst-inspect-1.0 on aravissrc is blacklisted by gstreamer, so the pipeline cannot be formed.

**To Reproduce**
Downloaded  latest as of 10092020.
meson configure
meson build
ninja 
sudo PATH="$PATH" ninja install

Tested camera with Viewer successfully, found a previous issue resolved by reducing buffer to 256x256
ARV_UV_STREAM_MAXIMUM_TRANSFER_SIZE	65536

**Expected behavior**
Gstreamer should permit inspection of the element

**Camera description:**
 - Manufacturer: Allied Vision
 - Model: 1800U-319
 - Interface: USB3

**Platform description:**
 - Aravis version 0.8
 - OS: Ubuntu 18.04
 - Hardware Jetson Nano.

**Additional context**
The installation placed gstaravis in the wrong gstreamer lib so I copied the .so to the system gstreamer library.

EmmanuelP, 2020-10-29:
> The installation placed gstaravis in the wrong gstreamer lib so I copied the .so to the system gstreamer library.

You should configure the build to install in `/usr`, instead of default `/usr/local`.

```
meson --prefix=/usr build
```

I guess gstreamer is blacklisting the aravis plugin because it can not find the aravis library.

EmmanuelP, 2020-11-06:
I'm closing this issue. Please reopen if you think there is something wrong in aravis.

Cheers.

closed 2020-11-06 by EmmanuelP