AravisProject/aravis #431: USB3Vision device in Mingw64 Win10

NerdToMars, 2020-10-28:
Have successfully compiled the Aravis in Mingw64 Win10 (without viewer), but have some issues with the USB3 Balser camera. 

Try 1. At the very beginning, the Aravis cannot find the USB3 device among xxx devices (Found 0 USB3Vision device (among 4 USB devices)). I tracing the code, the error may be caused by libusb_open, it returns LIBUSB_ERROR_NOT_FOUND. 
I found the camera has been recognized as Basler device with Basler USB camera driver, so I follow this link: https://stackoverflow.com/questions/49327380/cannot-open-usb-device-with-libusb-1-0-in-cygwin and this https://stackoverflow.com/questions/17350177/libusb-open-returns-libusb-error-not-supported-on-windows-7, using Zadig to replace the driver to libusb;  

Try 2.  after the device driver replace by Zadig, The error in Try 1. disappeared but got a new error LIBUSB_ERROR_NOT_SUPPORTED when the code tried to `libusb_claim_interface (priv->usb_device, priv->data_interface); https://github.com/AravisProject/aravis/blob/d80d7b207771bc5605be2c488736040bba742248/src/arvuvdevice.c#L799`

3. Also tried WinUSB and libusbK provided by Zadig (Version 2.5 (Build 730))
 
Now I have no idea how to resolve this, Can you provide some suggestion? 

PS: the build machine (Microsoft Windows [Version 10.0.20241.1005]) and env:

> Project name: aravis
> Project version: 0.8.3
> C compiler for the host machine: cc (gcc 10.2.0 "cc (Rev1, Built by MSYS2 project) 10.2.0")
> C linker for the host machine: cc ld.bfd 2.35
> C++ compiler for the host machine: c++ (gcc 10.2.0 "c++ (Rev1, Built by MSYS2 project) 10.2.0")
> C++ linker for the host machine: c++ ld.bfd 2.35
> Host machine cpu family: x86_64
> Host machine cpu: x86_64
> Dependency glib-2.0 found: YES 2.64.5 (cached)
> Dependency gobject-2.0 found: YES 2.64.5 (cached)
> Dependency gio-2.0 found: YES 2.64.5 (cached)
> Dependency libxml-2.0 found: YES 2.9.10 (cached)
> Dependency zlib found: YES 1.2.11 (cached)
> Dependency libusb-1.0 found: YES 1.0.23 (cached)
> Library m found: YES
> Library ws2_32 found: YES
> Has header "linux/if_packet.h" : NO (cached)
> Dependency gobject-introspection-1.0 found: YES 1.64.1 (cached)
> Dependency gobject-introspection-1.0 found: YES 1.64.1 (cached)
> Program g_ir_scanner found: YES (C:/developing/msys2/mingw64/bin/g-ir-scanner.exe)
> Program g_ir_compiler found: YES (C:/developing/msys2/mingw64/bin/g-ir-compiler.exe)
> Program python3 found: YES (C:\developing\msys2\mingw64\bin/python3.EXE)
> Configuring pylaunch using configuration
> Configuring pylaunch-dbg using configuration
> Configuring jslaunch using configuration
> Configuring jslaunch-dbg using configuration
> Dependency gtk+-3.0 found: YES 3.24.22 (cached)
> Dependency gstreamer-base-1.0 found: YES 1.16.2 (cached)
> Dependency gstreamer-app-1.0 found: YES 1.16.2 (cached)
> Dependency gstreamer-video-1.0 found: YES 1.16.2 (cached)
> Dependency libnotify found: YES 0.7.8 (cached)
> Dependency gstreamer-base-1.0 found: YES 1.16.2 (cached)
> Dependency gstreamer-app-1.0 found: YES 1.16.2 (cached)
> Configuring gst-aravis-launch using configuration
> Configuring gst-aravis-inspect using configuration
> Program gtkdoc-scan found: NO
> Program xsltproc found: YES
> Build targets in project: 35

The USB3Vision camera model: Basler:acA4112-30um

eudoxos, 2020-11-03:
I assume you also read [How to use libusb on Windows](https://github.com/libusb/libusb/wiki/Windows#How_to_use_libusb_on_Windows)?

NerdToMars, 2020-11-05:
> I assume you also read [How to use libusb on Windows](https://github.com/libusb/libusb/wiki/Windows#How_to_use_libusb_on_Windows)?

Yes, have read this, but still don't know what's the root cause.

NerdToMars, 2020-11-05:
here are some further information:
1. libusbK driver installed by Zadig Version 2.5, the error arises when trying to open data interface(2), error code is -12 (LIBUSB_ERROR_NOT_SUPPORTED), the information shown in USBView:

> [Port17]  :  USB Composite Device
> 
> 
> Is Port User Connectable:         yes
> Is Port Debug Capable:            yes
> Companion Port Number:            1
> Companion Hub Symbolic Link Name: USB#ROOT_HUB30#4&2e69bf72&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}
> Protocols Supported:
>  USB 1.1:                         no
>  USB 2.0:                         no
>  USB 3.0:                         yes
> 
> Device Power State:               PowerDeviceD0
> 
>        ---===>Device Information<===---
> English product name: "Blackfly S BFS-U3-04S2M"
> 
> ConnectionStatus:                  
> Current Config Value:              0x01  -> Device Bus Speed: SuperSpeed
> Device Address:                    0x0C
> Open Pipes:                           4
> 
>           ===>Device Descriptor<===
> bLength:                           0x12
> bDescriptorType:                   0x01
> bcdUSB:                          0x0310
> bDeviceClass:                      0xEF  -> This is a Multi-interface Function Code Device
> bDeviceSubClass:                   0x02  -> This is the Common Class Sub Class
> bDeviceProtocol:                   0x01  -> This is the Interface Association Descriptor protocol
> bMaxPacketSize0:                   0x09 = (9) Bytes
> idVendor:                        0x1E10 = FLIR Integrated Imaging Solutions
> idProduct:                       0x4000
> bcdDevice:                       0x0000
> iManufacturer:                     0x01
>      English (United States)  "FLIR"
> iProduct:                          0x02
>      English (United States)  "Blackfly S BFS-U3-04S2M"
> iSerialNumber:                     0x03
>      English (United States)  "010B3D45"
> bNumConfigurations:                0x01
> 
>           ---===>Open Pipes<===---
> 
>           ===>Endpoint Descriptor<===
> bLength:                           0x07
> bDescriptorType:                   0x05
> bEndpointAddress:                  0x02  -> Direction: OUT - EndpointID: 2
> bmAttributes:                      0x02  -> Bulk Transfer Type
> wMaxPacketSize:                  0x0400
> bInterval:                         0x00
> 
>           ===>Endpoint Descriptor<===
> bLength:                           0x07
> bDescriptorType:                   0x05
> bEndpointAddress:                  0x82  -> Direction: IN - EndpointID: 2
> bmAttributes:                      0x02  -> Bulk Transfer Type
> wMaxPacketSize:                  0x0400
> bInterval:                         0x00
> 
>           ===>Endpoint Descriptor<===
> bLength:                           0x07
> bDescriptorType:                   0x05
> bEndpointAddress:                  0x83  -> Direction: IN - EndpointID: 3
> bmAttributes:                      0x02  -> Bulk Transfer Type
> wMaxPacketSize:                  0x0400
> bInterval:                         0x00
> 
[... 130 lines cut ...]
> 
>           ===>BOS Descriptor<===
> bLength:                           0x05
> bDescriptorType:                   0x0F
> wTotalLength:                      0x0016
> bNumDeviceCaps:                    0x02
> 
>           ===>USB 2.0 Extension Descriptor<===
> bLength:                           0x07
> bDescriptorType:                   0x10
> bDevCapabilityType:                0x02
> bmAttributes:                      0x0000F61E  -> Supports Link Power Management protocol
> 
>           ===>SuperSpeed USB Device Capability Descriptor<===
> bLength:                           0x0A
> bDescriptorType:                   0x10
> bDevCapabilityType:                0x03
> bmAttributes:                      0x00
> wSpeedsSupported:                  0x0C
>   -> Supports high-speed operation
>   -> Supports SuperSpeed operation
> bFunctionalitySupport:             0x02 -> lowest speed = high-speed
> bU1DevExitLat:                     0x0A -> less than 10 micro-seconds
> wU2DevExitLat:                     0x0200 -> less than 512 micro-seconds
>

NerdToMars, 2020-11-05:
While the USBView info shown in Ubuntu is like:

> Blackfly S BFS-U3-04S2M
> Manufacturer: FLIR
> Serial Number: 010B3D45
> Speed: 480Mb/s (high)
> Bus:   1
> Address:  18
> USB Version:  2.10
> Device Class: ef(misc )
> Device Subclass: 02
> Device Protocol: 01
> Maximum Default Endpoint Size: 64
> Number of Configurations: 1
> Vendor Id: 1e10
> Product Id: 4000
> Revision Number:  0.00
> 
> Config Number: 1
> 	Number of Interfaces: 3
> 	Attributes: 80
> 	MaxPower Needed: 224mA
> 
> 	Interface Number: 0
> 		Name: (none)
> 		Alternate Number: 0
> 		Class: ef(misc ) 
> 		Sub Class: 05
> 		Protocol: 00
> 		Number of Endpoints: 2
> 
> 			Endpoint Address: 02
> 			Direction: out
> 			Attribute: 2
> 			Type: Bulk
> 			Max Packet Size: 512
> 			Interval: 0ms
> 
> 			Endpoint Address: 82
> 			Direction: in
> 			Attribute: 2
> 			Type: Bulk
> 			Max Packet Size: 512
> 			Interval: 0ms
> 
> 	Interface Number: 1
> 		Name: (none)
> 		Alternate Number: 0
> 		Class: ef(misc ) 
> 		Sub Class: 05
> 		Protocol: 01
> 		Number of Endpoints: 1
> 
> 			Endpoint Address: 83
> 			Direction: in
> 			Attribute: 2
> 			Type: Bulk
> 			Max Packet Size: 512
> 			Interval: 0ms
> 
> 	Interface Number: 2
> 		Name: (none)
> 		Alternate Number: 0
> 		Class: ef(misc ) 
> 		Sub Class: 05
> 		Protocol: 02
> 		Number of Endpoints: 1
> 
> 			Endpoint Address: 81
> 			Direction: in
> 			Attribute: 2
> 			Type: Bulk
> 			Max Packet Size: 512
> 			Interval: 0ms

Fixstars-momoko, 2022-01-12:
I followed steps 1 and 2 from @NerdToMars , and I have the same issue.

Here is what I have seen after step 2.
The logs in `arv_uv_device_constructed` in `aravis\src\arvuvdevice.c` shows the values: 
```
$ ./src/arv-camera-test-0.8.exe
Looking for the first available camera

>>>---libusb_claim_interface (priv->usb_device, priv->control_interface)
        >>>---priv->control_interface = 0
        >>>---libusb_claim_interface (priv->usb_device, priv->control_interface) : 0

>>>---libusb_claim_interface (priv->usb_device, priv->data_interface)
        >>>---priv->data_interface = 1
        >>>---libusb_claim_interface (priv->usb_device, priv->data_interface) : -12

(arv-camera-test-0.8.exe:14736): GLib-CRITICAL **: 08:54:25.750: g_thread_join: assertion 'thread' failed
No camera found: Failed to claim USB interface to '(NULL)-(NULL)-(NULL)-26760158A14A': LIBUSB_ERROR_NOT_SUPPORTED
```
I am using version 0.8.20.

@EmmanuelP I'd like to work on and contribute to this, and seems like it is on To Do in Windows support.
Would you mind to share the progress and orientation of this issue?

EmmanuelP, 2022-01-12:
Hi @Fixstars-momoko 

> @EmmanuelP I'd like to work on and contribute to this, and seems like it is on To Do in Windows support.
> Would you mind to share the progress and orientation of this issue?

All the informations about this issue are in this report. May be there is some progress made by other contributors I am not aware of.

Any help in fixing this one is gladly welcome !

Fixstars-momoko, 2022-04-07:
Hi @EmmanuelP, @NerdToMars 

I'd share how to apply WinUSB to Basler (Composite Device) with Zadig.

Basically we follow what @NerdToMars has done in item 2, but apply WinUSB to the device, but to its Composite Parent.

When you list the device, you have to mark "List All Devices" and unmark "Ignore Hubs or Composite Parents" on Options as follows.

[image]

Then choose `<device name> (Composite Parent)`

With this driver replace, I confirmed that `arv-tool-0.8.exe` successfully listed up the device name (also with `features`, showed the GENICAM Feature) and `arv-camera-test-0.8.exe` obtained frames.

`arv-tool-0.8.exe features`

[image]

`arv-camera-test-0.8.exe`

[image]