AravisProject/aravis #432: GigEVision LLA mode vs DHCP

horstrand, 2020-10-28:
Everytime I connect my GigEV camera using aravis to my computer I have to fix the network IP, as it connects to the camera using the LLA mode. I have tried using DHCP but it is not assigning any IPv4 to the camera, so I'm not able to connect.

I have checked all related registers in my camera and seem to be right. Both LLA and DHCP are on:

GevCurrentIPConfigurationLLAReg: 1
GevCurrentIPConfigurationDHCPReg: 1
GevCurrentIPConfigurationPersistentIPReg: 0
GevSupportedIPConfigurationLLA: 1
GevSupportedIPConfigurationDHCP: 1
GevSupportedIPConfigurationPersistentIP: 1

Platform: 
-Arv version: 0.6.4
OS: Ubuntu 18.04.5 LTS, kernel 4.15.0-122-generic
Hw: x86_64

Camera:
 - Manufacturer: Specim
 - Model: FX17e
 - Interface: GigEV

Before I worked with the eBUS package from Pleora and I experienced the same behavior in an Ubuntu system, but on Windows the DHCP actually worked. So it might be operating system related, I'm not sure.

Thanks in advance for your support.
Pablo

EmmanuelP, 2020-10-29:
Hi Pablo,

It looks like a network configuration issue, not really related to Aravis. I guess there is no dhcp service running on your setup.

The simplest solution would be to assign a persistent IP to your device. Unfortunately there is not utility in aravis to help in this task, though it should be possible to assign an IP using arv-tool (I did not try).

I'm closing this issue. Don't hesitate to reopen if you think there is something wrong in aravis.

Thanks.

EmmanuelP, 2020-10-29:
I have just created #435 in order to track the implementation of an IP configuration tool

closed 2020-10-29 by EmmanuelP