AravisProject/aravis #436: Ubuntu 20.04 x86 install instructions

patrickwspanialy, 2020-10-29:
**Is your feature request related to a problem? Please describe.**
Setting up Aravis for a new user isn't as easy as it could be. Step-by-step install and example instructions would help to get started. For example, I'm not sure how to run the python examples. 

**Describe the solution you'd like**
An up-to-date set of markdown pages for installation on select distros and platforms, and examples in different programming languages.

**Describe alternatives you've considered**
Trial and error.

**Additional context**
For example, instructions for installing and running a python example on Ubuntu 20.04 on an x86 machine may include:

### Install pre-requisits
```
sudo apt install \
build-essential cmake \
libxml2-dev libusb-1.0-0-dev \
gobject-introspection libgirepository1.0-dev libgtk-3-dev \
libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev \
libnotify-dev gtk-doc-tools gstreamer1.0-plugins-bad
```

### Install aravis
```
wget http://ftp.gnome.org/pub/GNOME/sources/aravis/0.8/aravis-0.8.2.tar.xz
tar -xf aravis-0.8.2.tar.xz
cd aravis-0.8.2
meson build
cd build
ninja
sudo ninja install
sudo ldconfig
```
### Setup MTU for GigE
...

### Setup permissions USB 3
...

### Setup python bindings
(this is where I'm currently stuck)
```
$ python3 arv-buffer-test.py 
Traceback (most recent call last):
  File "arv-buffer-test.py", line 13, in <module>
    gi.require_version ('Aravis', '0.8')
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 129, in require_version
    raise ValueError('Namespace %s not available' % namespace)
```

#### Run python example

### Setup javascript bindings

#### Run javascript example

patrickwasp, 2020-11-02:
Looks like creating a link in `/usr/lib/x86_64-linux-gnu/girepository-1.0/` is what was needed. 

```
sudo ln -s /usr/local/lib/x86_64-linux-gnu/girepository-1.0/Aravis-0.8.typelib /usr/lib/x86_64-linux-gnu/girepository-1.0/Aravis-0.8.typelib
```

closed 2020-11-07 by patrickwasp

LeeYangLBLBCS, 2021-01-03:
I'm trying to build Aravis for the first time (ubuntu 18.04). I found this "closed issue" that's helpful.
I installed the prerequisites shown by @patrickwasp 
then proceeded to run "meson build", I got this error:
$meson build
The Meson build system
Version: 0.45.1
Source dir: /home/lyang/epics/aravis-0.8.5
Build dir: /home/lyang/epics/aravis-0.8.5/build
Build type: native build

meson_options.txt:3:0: ERROR: Unknown type feature.

Any help?

EmmanuelP, 2021-01-03:
Aravis requires meson >= 0.47, which probably explain your error.

You should be able to upgrade your meson version using pip.