AravisProject/aravis #438: Aravis icons license

cmarmo, 2020-10-30:
Hello,

I am packaging the last version of Aravis for Debian.
The license of the Aravis icons (`viewer/icons/src/*.svg`, `viewer/icons/gnome/*/*/*.svg`, `viewer/icons/gnome/*/*/*.png`) should be listed in the debian/copyright file. I assumed that the license was a creative commons one, as the svg contains the directive `xmlns:cc="http://creativecommons.org/ns#"`... perhaps I'm wrong.

May I ask you which license should I list? If it is a creative commons, do you mind specifying [which one](https://creativecommons.org/about/cclicenses/)?

Thanks a lot!
Chiara

EmmanuelP, 2020-10-30:
Bonjour Chiara,

Is CC BY-SA ok from the debian packaging point of view ?

What is the usual way to specify icon license in source packages ?

cmarmo, 2020-10-30:
> Bonjour Chiara,

Bonjour Emmanuel et merci pour ta réponse!

> Is CC BY-SA ok from the debian packaging point of view ?

I think so, they just ask to add the text of the license because it is not included in the `/usr/share/common-licenses` directory in debian.

> What is the usual way to specify icon license in source packages ?

euh... I don't know... :) sorry... I think the creative common is the right choice: I would probably specify which one and the attribution in the source svg using [Creative Commons Rights Expression Language](http://creativecommons.org/ns#).
Inkscape allows you to do that in the "Document Properties" -> "License".... or with a "license" file in `viewer/icons/` directory...

EmmanuelP, 2020-11-04:
I have added CC BY-SA license metadata to all icon source files.

Cheers,

    Emmanuel.

closed 2020-11-04 by EmmanuelP

cmarmo, 2020-11-04:
Thanks a lot! :)
Cheers,

Chiara

commit 0c00d172 (2020-11-04): viewer: set icon license

Fix #438