AravisProject/aravis #439: Imperx Bobcat 6620C Support

SaharGezer, 2020-11-02:
I have Imperx Bobcat 6620C GigE camera but I got an error when I tried to run a pipline via Gstreamer

ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Could not read camera "": node 'AcquisitionFrameRateAbs' not found

how can I fix this error?

Thanks

EmmanuelP, 2020-11-02:
Hi,

This error means your device has neither `AcquisitionFrameRate` nor `AcquisitionFrameRateAbs` features, making it impossible to set the desired frame rate.

There is 2 things we can do in aravis:

* Don't abort the gstreamer pipeline if frame rate setting failed
* Implement the frame rate setting for this device

What does the Imperx documentation says about frame rate setting ?

Emmanuel.

SaharGezer, 2020-11-03:
When I use aravis viewer I see the camera and can get frames, when I use aravis-gst + another gige camera like Filer I do not get any errors.

In Imperx documentation there is not any mention about frame rate

EmmanuelP, 2020-11-03:
Please attach the output of `arv-tool-0.8 features`.

SaharGezer, 2020-11-03:
Imperx, inc.-GEV_B6620C-500981 (145.8.9.104)
Category    : 'Root'
    Category    : 'DeviceInformation'
        StringReg   : 'DeviceVendorName'
        StringReg   : 'DeviceModelName'
        StringReg   : 'DeviceManufacturerInfo'
        StringReg   : 'DeviceVersion'
        StringReg   : 'DeviceID'
        Enumeration : 'DeviceScanType'
            EnumEntry   : 'Areascan'
        Command     : 'DeviceReset'
        Command     : 'CameraHeadReset'
    Category    : 'ImageSizeControl'
        Category    : 'AOI'
            Category    : 'Slave_AOI1'
                Enumeration : 'AOI1_Select'
                    EnumEntry   : 'Exclude'
                    EnumEntry   : 'Include'
                    EnumEntry   : 'Off'
                Integer     : 'AOI1_OffsetX'
                Integer     : 'AOI1_Width'
                Integer     : 'AOI1_OffsetY'
                Integer     : 'AOI1_Height'
            Category    : 'Slave_AOI2'
                Enumeration : 'AOI2_Select'
                    EnumEntry   : 'Exclude'
                    EnumEntry   : 'Include'
                    EnumEntry   : 'Off'
                Integer     : 'AOI2_OffsetX'
                Integer     : 'AOI2_Width'
                Integer     : 'AOI2_OffsetY'
                Integer     : 'AOI2_Height'
            Category    : 'Slave_AOI3'
                Enumeration : 'AOI3_Select'
                    EnumEntry   : 'Exclude'
                    EnumEntry   : 'Include'
                    EnumEntry   : 'Off'
                Integer     : 'AOI3_OffsetX'
                Integer     : 'AOI3_Width'
                Integer     : 'AOI3_OffsetY'
                Integer     : 'AOI3_Height'
            Category    : 'Slave_AOI4'
                Enumeration : 'AOI4_Select'
                    EnumEntry   : 'Exclude'
                    EnumEntry   : 'Include'
                    EnumEntry   : 'Off'
                Integer     : 'AOI4_OffsetX'
                Integer     : 'AOI4_Width'
                Integer     : 'AOI4_OffsetY'
                Integer     : 'AOI4_Height'
            Category    : 'Slave_AOI5'
                Enumeration : 'AOI5_Select'
                    EnumEntry   : 'Exclude'
                    EnumEntry   : 'Include'
                    EnumEntry   : 'Off'
                Integer     : 'AOI5_OffsetX'
                Integer     : 'AOI5_Width'
                Integer     : 'AOI5_OffsetY'
                Integer     : 'AOI5_Height'
            Category    : 'Slave_AOI6'
                Enumeration : 'AOI6_Select'
                    EnumEntry   : 'Exclude'
                    EnumEntry   : 'Include'
                    EnumEntry   : 'Off'
                Integer     : 'AOI6_OffsetX'
                Integer     : 'AOI6_Width'
                Integer     : 'AOI6_OffsetY'
                Integer     : 'AOI6_Height'
        Integer     : 'SensorWidth'
        Integer     : 'SensorHeight'
[... 1452 lines cut ...]
    Category    : 'UserSets'
        Category    : 'UserSetLoads'
            Command     : 'UserSetLoadDefault'
            Command     : 'UserSetLoadUserSet1'
        Category    : 'UserSetsAvailability'
            Boolean     : 'UserSet1Available'
        Enumeration : 'UserSetLoadStatus' (Not available)
            EnumEntry   : 'Success'
            EnumEntry   : 'Failure'
            EnumEntry   : 'InProgress'
        Enumeration : 'UserSetSelector'
              * UserSetLoad
              * UserSetSave
            EnumEntry   : 'UserSet1'
            EnumEntry   : 'Default'
        Command     : 'UserSetLoad'
        Command     : 'UserSetSave' (Not available)
        Enumeration : 'UserSetDefaultSelector'
            EnumEntry   : 'UserSet1'
            EnumEntry   : 'Default'
        Enumeration : 'UserSetLastLoaded'
            EnumEntry   : 'UserSet2'
            EnumEntry   : 'UserSet1'
            EnumEntry   : 'Default'
    Category    : 'TLParamsLockedCategory'

EmmanuelP, 2020-11-03:
There is no standard way to set the frame rate on this device, which means code in arvcamera.c must be modified in order to support it.

The FrameRate related features seems to be at least CurrentFrameRate and ConstantFrameRate.

I personally don't plan to work on this support, as I don't have the device at hand. But I will happily accept a patch...

SaharGezer, 2020-11-04:
Can you explain me what I need to add to arvcamera.c and where?
I tried to edit this code but I did not get any good results

EmmanuelP, 2020-11-04:
A new vendor and series should be added to `ArvCameraVendor` and `ArvCameraSeries` enums, and used in `arv_camera_constructed`.

Then support needs to be added in several functions:

```
arv_camera_is_frame_rate_available
arv_camera_set_frame_rate
arv_camera_get_frame_rate
arv_camera_get_frame_rate_bounds
```

SaharGezer, 2020-11-10:
> New vendor and series should be added to `ArvCameraVendor` and `ArvCameraSeries` enums, and used in `arv_camera_constructed()`.
>
> Then support needs to be added in several functions:
>
> ``` [...]

I made the above changes but I got this error :+1:  

node 'CurrentFrameRate' [ArvGcIntegerNode]  is not a ArvGcFloat

Can you tell me what is the problem?

EmmanuelP, 2020-11-10:
> node 'CurrentFrameRate' [ArvGcIntegerNode] is not a ArvGcFloat

You are trying to access an integer node using the float interface.

Could you share the modifications you have done.

SaharGezer, 2020-11-10:
in arv_camera_is_frame_rate_available  -> I added
		case ARV_CAMERA_VENDOR_IMPERX:
			return arv_camera_is_feature_available (camera, "CurrentFrameRate", error);

in arv_camera_set_frame_rate -> I added
		case ARV_CAMERA_VENDOR_IMPERX:
			if (local_error == NULL)
				arv_camera_set_string (camera, "TriggerMode", "Off", &local_error);
			if (local_error == NULL)
				arv_camera_set_integer (camera, "CurrentFrameRate", frame_rate, &local_error);
			break;

in arv_camera_get_frame_rate -> I added
case ARV_CAMERA_VENDOR_IMPERX:
			return arv_camera_get_integer (camera, "CurrentFrameRate", error);

in arv_camera_get_frame_rate_bounds -> I added
case ARV_CAMERA_VENDOR_IMPERX:
			arv_camera_get_float_bounds (camera, "CurrentFrameRate", min, max, error);
			break;

EmmanuelP, 2020-11-10:
> case ARV_CAMERA_VENDOR_IMPERX:
arv_camera_get_float_bounds (camera, "CurrentFrameRate", min, max, error);
break;

You should use get_integer_bounds here, with temporary guint64 variables for the storage of the min/max values.

By the way, what is returning `arv-tool-0.8 control CurrentFrameRate` ?

SaharGezer, 2020-11-11:
> arv_camera_get_float_bounds

I made this cahnge -> arv_camera_get_integer_bounds (camera, "CurrentFrameRate", 2, 3, error);

but I got an error -> 

#0  0x0000007f8fe6cce4 in __waitpid (pid=<optimized out>, stat_loc=0x7fcde58014, options=<optimized out>) at ../sysdeps/unix/sysv/linux/waitpid.c:30
#1  0x0000007f8fea82a0 in g_on_error_stack_trace ()
#2  0x0000005556455c3c in  ()
#3  0x0000005566317510 in  ()
Spinning.  Please run 'gdb gst-launch-1.0 12201' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.

when I run the command -> arv-tool-0.8 control CurrentFrameRate I get :

Imperx, inc.-GEV_B6620C-500980
CurrentFrameRate = 2

SaharGezer, 2020-11-18:
Hi,
I need your help, I got this error:

Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Could not set caps on camera "137.9.2.128": IntSwissKnife is read only
Additional debug info:
../gst/gstaravis.c(350): gst_aravis_set_caps (): /GstPipeline:pipeline0/GstAravis:aravis0
Execution ended after 0:00:00.065657984
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...

IntSwissKnife is read only -> How can I fix it?

thank you

EmmanuelP, 2020-11-18:
> I made this cahnge -> arv_camera_get_integer_bounds (camera, "CurrentFrameRate", 2, 3, error);

Parameter 2 and 3 are placeholders for the return value of min and max. You can not pass constant here.

EmmanuelP, 2020-11-18:
> ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Could not set caps on camera "137.9.2.128": IntSwissKnife is read only

The code in gstaravis tries to write to a read only feature. Try to isolate where it happens.

SaharGezer, 2020-11-18:
> > ERROR: from element /GstPipeline:pipeline0/GstAravis:aravis0: Could not set caps on camera "137.9.2.128": IntSwissKnife is read only
> 
> The code in gstaravis tries to write to a read only feature. Try to isolate where it happens.

Thank you for your reply, I found the bug and now it works with my Imperx Camera.

EmmanuelP, 2020-11-18:
> Thank you for your reply, I found the bug and now it works with my Imperx Camera.

Could you share the modifications you have made ?

SaharGezer, 2020-11-19:
> > Thank you for your reply, I found the bug and now it works with my Imperx Camera.
> 
> Could you share the modifications you have made ?

Yes sure , but before that I need to add another GIGE camera to aravis gst