AravisProject/aravis #440: Support Cognex cic-1300

Gunob17, 2020-11-02:
Hello 
first of all great project.

im trying to get a cognex cic-1300 gige to communicate with a nvidea jetson nano. 
the cic-1300 is according to cognex a rebrand of the basler asa-1300. 
when i open the aravis viewer i can see it sends out the dicovery cmd in wireshark and a return acknowledgement comes from the camera but nothing shows up in the viewer. i have also tried getting it to work in pylon but also without any luck. 
Could i missed something obvious or can it just be cognex that have made so many changes to the firmware that it doesn't work?  
```
Frame 70: 298 bytes on wire (2384 bits), 298 bytes captured (2384 bits) on interface 0
    Interface id: 0 (eth0)
        Interface name: eth0
    Encapsulation type: Ethernet (1)
    Arrival Time: Nov  2, 2020 15:22:39.054472789 CET
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1604326959.054472789 seconds
    [Time delta from previous captured frame: 0.000344907 seconds]
    [Time delta from previous displayed frame: 0.000344907 seconds]
    [Time since reference or first frame: 36.144034623 seconds]
    Frame Number: 70
    Frame Length: 298 bytes (2384 bits)
    Capture Length: 298 bytes (2384 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:ip:udp:gvcp]
    [Coloring Rule Name: UDP]
    [Coloring Rule String: udp]
Ethernet II, Src: Basler_35:d2:b3 (00:30:53:35:d2:b3), Dst: Nvidia_eb:fa:21 (00:04:4b:eb:fa:21)
    Destination: Nvidia_eb:fa:21 (00:04:4b:eb:fa:21)
        Address: Nvidia_eb:fa:21 (00:04:4b:eb:fa:21)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: Basler_35:d2:b3 (00:30:53:35:d2:b3)
        Address: Basler_35:d2:b3 (00:30:53:35:d2:b3)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: IPv4 (0x0800)
Internet Protocol Version 4, Src: 10.3.0.2, Dst: 10.3.0.4
    0100 .... = Version: 4
    .... 0101 = Header Length: 20 bytes (5)
    Differentiated Services Field: 0x00 (DSCP: CS0, ECN: Not-ECT)
        0000 00.. = Differentiated Services Codepoint: Default (0)
        .... ..00 = Explicit Congestion Notification: Not ECN-Capable Transport (0)
    Total Length: 284
    Identification: 0x0000 (0)
    Flags: 0x0000
        0... .... .... .... = Reserved bit: Not set
        .0.. .... .... .... = Don't fragment: Not set
        ..0. .... .... .... = More fragments: Not set
        ...0 0000 0000 0000 = Fragment offset: 0
    Time to live: 255
    Protocol: UDP (17)
    Header checksum: 0xa6c5 [validation disabled]
    [Header checksum status: Unverified]
    Source: 10.3.0.2
    Destination: 10.3.0.4
User Datagram Protocol, Src Port: 3956, Dst Port: 43266
    Source Port: 3956
    Destination Port: 43266
    Length: 264
    Checksum: 0x16f7 [unverified]
    [Checksum Status: Unverified]
    [Stream index: 7]
GigE Vision Control Protocol
    Acknowledge Header: DISCOVERY_ACK
        Status: GEV_STATUS_SUCCESS (0x0000)
        Acknowledge: DISCOVERY_ACK (0x0003)
        Payload Length: 0x00f8 (248)
        Request ID: 0xffff (65535)
        DISCOVERY_ACK Payload
            Spec Version: 0x00010001
                0000 0000 0000 0001 .... .... .... .... = Version Major: 0x0001
                .... .... .... .... 0000 0000 0000 0001 = Version Minor: 0x0001
            Device Mode: 0x80000001
                1... .... .... .... .... .... .... .... = Endianess: True
                .000 .... .... .... .... .... .... .... = Device Class: Transmitter (0x0)
                .... ..00 .... .... .... .... .... .... = Current Link Configuration: Single Link (0x0)
                .... .... .... .... .... .... .... 0001 = Character Set: UTF-8 Character Set (0x1)
            Device MAC Address: Basler_35:d2:b3 (00:30:53:35:d2:b3)
            Supported IP Configuration: 0x00000007
                0... .... .... .... .... .... .... .... = IP Config Can Handle Pause Frames: False
                .0.. .... .... .... .... .... .... .... = Can Generate Pause Frames: False
                .... .... .... .... .... .... .... .1.. = LLA: True
                .... .... .... .... .... .... .... ..1. = DHCP: True
                .... .... .... .... .... .... .... ...1 = Persistent IP: True
            Current IP Configuration: 0x00000005
                0... .... .... .... .... .... .... .... = IP Config Can Handle Pause Frames: False
                .0.. .... .... .... .... .... .... .... = Can Generate Pause Frames: False
                .... .... .... .... .... .... .... .1.. = LLA: True
                .... .... .... .... .... .... .... ..0. = DHCP: False
                .... .... .... .... .... .... .... ...1 = Persistent IP: True
            Current IP: 10.3.0.2
            Subnet Mask: 255.255.0.0
            Default Gateway: 0.0.0.0
            Manufacturer Name: COGNEX
            Model Name: CIC-1300
            Device Version: 106552-08
            Manufacturer Specific Info: none
            Serial Number: 23527347
            User-defined Name: 

```

EmmanuelP, 2020-11-03:
Did you try `tests/arv-camera-test -d all` ?

Gunob17, 2020-11-04:
thanks i just tryed it and i get: 
it timesout on both the read_memory and write_register and ends op giving the following error:  
`No camera found: Invalid Genicam data`

That gives me the sneeking suspicion that Cognex have changed so much of the firmware that its not usable. 
Is there any way i somehow can add it to Genicam myself if i can get the information needed from Cognex?

EmmanuelP, 2020-11-04:
It would help if you could capture the network traffic when the camera is accessed using cognex software. I'm a bit surprised it does not work at all, as the documentation seems to refer to GeniCam.

Gunob17, 2020-11-04:
okay here you go. if the image isn't enough then the zip file contains wireshark capture up until it the cognex software just keeps repeating the same readreg_cmd and is returned the same readreg_ack 

[image]

[cic-1300comunication.zip](https://github.com/AravisProject/aravis/files/5487274/cic-1300comunication.zip)

EmmanuelP, 2020-11-04:
I should have asked for it at the same time: could you also attach a capture during `tests/arv-camera-test` ?

Gunob17, 2020-11-04:
of cause here you go
 
[image]

[avaviscapture.zip](https://github.com/AravisProject/aravis/files/5487493/avaviscapture.zip)

Gunob17, 2020-11-04:
okay abit of progress i got it to continue communication it turned out that ubuntu keept changing my subnet mask which broke communication. i now have a new problem though now when i run `test/arv-camera-test -d all` I now get "cant create stream without control access"
"can't create stream thread: controller privileges required for streaming control" 
i have tried running it with root access in case that were the problem, but got the same message.

EmmanuelP, 2020-11-04:
If you got `"cant create stream without control access"`, that means aravis went past Genicam data retrieval.
This error may happen if the camera was not correctly shut down, and the camera is tied to a machine port. Make sure you don't have another implementation communicating with the devices.

Gunob17, 2020-11-04:
okay great thanks i will look into that

Gunob17, 2020-11-04:
That was it, it works now thanks alot and stay safe

closed 2020-11-04 by Gunob17

EmmanuelP, 2020-11-05:
Ok, great news.

> okay abit of progress i got it to continue communication it turned out that ubuntu keept changing my subnet mask which broke communication.

For the record, could explain how you have detected this bad behaviour and how you fixed it ?

Gunob17, 2020-11-05:
well when it comes to the bad subnet mask i was confused as to why i could establish a connection when a secondary ip was connected and not with out, it turns out that ubuntu/jetpack will drop any changes set with `ip addr add X.X.X.X/24 dev eth0` if network configuration has not been changed from automatic to manual. i'm not sure in what file this should be changed in is called but it can be changed in settings in the gui 

settings -->Network -->Wired-->IPv4 Settings 

i found out that was the problem by comparing the subnet masks in the wireshark dumps above.

for the second part after the control access problem i did a restart of the jetson which fixed the problem, it was from when i was trying out pylon, which apparently had left the camera hanging and had locked aravis out.

EmmanuelP, 2020-11-05:
Thanks for the feedback.

Cheers.