AravisProject/aravis #442: Can't connect to two Gigev cameras simultaneously

horstrand, 2020-11-03:
Hi,
I'm having troubles trying to connect to two cameras with Aravis. If I run the aravis tool it only detects one device. But with the arv_get_n_devices() function it detects two and I can also get the device id of each camera by invoking arv_get_device_id (i).

Later in the code when I try to create a new camera with arv_camera_new(device_id), it just works for one (and always the same). The device_id is the string that the arv_get_device_id function is returning. 
If I unplug the camera that was connecting fine and try with the other, then it is alright and I can connect. When I plug again the second camera, again I can only connect to one.
I tried a different way by opening two devices instead, using arv_interface_open_device, but the same happens, it just works for one of them, the other one returns NULL as when trying with the camera initialization.

The cameras are connected to two different ethernet ports in the PC over LLA. The subnet mask is set to 255.255.0.0 and the IP of both connections in the PC side are set manually to the values:
169.254.0.1
169.254.0.2

The cameras get any address in the range 169.254.xx.xx, and always a different one from another.

Platform:
-Arv version: 0.6.4
OS: Ubuntu 18.04.5 LTS, kernel 4.15.0-122-generic
Hw: x86_64

Cameras:
Manufacturer: Specim
Model: FX17e and FX10e
Interface: GigEV

Thanks for the support.

EmmanuelP, 2020-11-03:
What gives `arv-tool-0.8 --debug=all` ?

horstrand, 2020-11-03:
It is actually arv-tool-0.6 --debug=all, as I am using version 0.6.4.

There it is:

Found 0 USB3Vision device (among 10 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 169.254.0.1
[GvDiscoverSocket::new] Add interface 169.254.0.2
[GvInterface::discovery] Device 'Specim-FX10e-077000034676' found (interface 169.254.0.1) user_id ' ' - MAC 'xx:xx:xx:xx:xx:xx'
[GvInterface::discovery] Device 'Specim-FX17e-076900048460' found (interface 169.254.0.2) user_id ' ' - MAC 'xx:xx:xx:xx:xx:xx'
Found 0 USB3Vision device (among 10 USB devices)
[GvDevice::new] Interface address = 169.254.0.1
[GvDevice::new] Device address = 169.254.146.234
[GvDevice::new] Failed to load genicam data
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 169.254.0.1
[GvDiscoverSocket::new] Add interface 169.254.0.2
[GvInterface::discovery] Device 'Specim-FX17e-076900048460' found (interface 169.254.0.2) user_id ' ' - MAC 'xx:xx:xx:xx:xx:xx'
[GvInterface::discovery] Device 'Specim-FX10e-077000034676' found (interface 169.254.0.1) user_id ' ' - MAC 'xx:xx:xx:xx:xx:xx'
[GvDevice::new] Interface address = 169.254.0.1
[GvDevice::new] Device address = 169.254.146.234
[GvDevice::new] Failed to load genicam data
Found 0 USB3Vision device (among 10 USB devices)
[GvDevice::new] Interface address = 169.254.0.2
[GvDevice::new] Device address = 169.254.133.177
[GvDevice::load_genicam] xml url = 'Local:FX17e.zip;400000;bf65' at 0x200
[GvDevice::load_genicam] Xml address = 0x400000 - size = 0xbf65 - FX17e.zip
[GvDevice::load_genicam] Zipped xml data
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (Visibility)
... (there are many unknown tags here, which I believe are irrelevant for this issue)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (ImposedAccessMode)
[Genicam::create_element] Unknown tag (Visibility)
[ArvDevice::set_string_feature_value] Node 'GevGVSPExtendedIDMode' is not a string
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
Specim-FX17e-076900048460 (169.254.133.177)

EmmanuelP, 2020-11-03:
> [GvDevice::new] Failed to load genicam data

Something goes wrong when loading the Genicam data. Before I try to debug an old version, could you please try latest stable aravis ?

EmmanuelP, 2020-11-03:
Specim-FX10e is the one that does not work. Could you attach the output of `arv-tool -n Specim-FX10e-077000034676 --debug=all:3` ?

horstrand, 2020-11-03:
Yes the FX10 is the one that does not work (in this situation). 
Here it goes the output of the command you suggested:

Found 0 USB3Vision device (among 10 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 169.254.0.1
[GvDiscoverSocket::new] Add interface 169.254.0.2
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = Specim
name         =  
model        = FX17e
address      = 169.254.133.177
00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 80 00 00 01  ................
00000010 00 00 70 f8 e7 b0 04 8a 00 00 00 07 00 00 00 06  ..p.............
00000020 00 00 00 00 00 00 00 00 00 00 00 00 a9 fe 85 b1  ................
00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000050 53 70 65 63 69 6d 00 29 74 0d 02 00 00 00 04 00  Specim.)t.......
00000060 00 00 00 00 00 98 00 00 98 fe 7f 00 00 00 00 29  ...............)
00000070 46 58 31 37 65 00 00 00 00 00 00 00 00 11 00 00  FX17e...........
00000080 00 00 00 00 00 00 00 29 74 0d 02 00 00 00 00 29  .......)t......)
00000090 56 65 72 73 69 6f 6e 20 31 2e 35 00 72 73 69 6f  Version 1.5.rsio
000000a0 6e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  n...............
000000b0 53 70 65 63 69 6d 00 00 02 98 00 00 00 40 d6 29  Specim.......@.)
000000c0 74 0d 02 00 00 b9 00 00 00 00 00 00 00 00 01 00  t...............
000000d0 00 00 00 00 00 50 01 29 74 0d 02 00 00 00 00 00  .....P.)t.......
000000e0 30 37 36 39 30 30 30 34 38 34 36 30 00 00 00 00  076900048460....
000000f0 20 00 00 00 00 00 00 00 e8 08 01 00 00 00 00 00   ...............

[GvInterface::discovery] Device 'Specim-FX17e-076900048460' found (interface 169.254.0.2) user_id ' ' - MAC
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = Specim
name         =  
model        = FX10e
address      = 169.254.146.234
00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 80 00 00 01  ................
00000010 00 00 70 f8 e7 b0 01 2c 00 00 00 07 00 00 00 06  ..p....,........
00000020 00 00 00 00 00 00 00 00 00 00 00 00 a9 fe 92 ea  ................
00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000050 53 70 65 63 69 6d 00 80 40 00 0c 76 05 00 04 00  Specim..@..v....
00000060 00 00 00 50 00 d0 0b 00 00 05 00 01 10 04 00 00  ...P............
00000070 46 58 31 30 65 00 00 80 40 00 0c 76 05 00 04 00  FX10e...@..v....
00000080 00 00 00 50 00 d0 0b 00 00 05 00 01 10 04 00 00  ...P............
00000090 56 65 72 73 69 6f 6e 20 30 2e 33 00 05 00 04 00  Version 0.3.....
000000a0 00 00 00 50 00 d0 0b 00 00 05 00 01 10 04 00 00  ...P............
000000b0 53 70 65 63 69 6d 00 80 40 00 0c 76 05 00 04 00  Specim..@..v....
000000c0 00 00 00 50 00 d0 0b 00 00 05 00 01 10 04 00 00  ...P............
000000d0 0f 00 00 00 08 00 01 00 04 00 00 04 00 00 00 10  ................
000000e0 30 37 37 30 30 30 30 33 34 36 37 36 00 00 00 00  077000034676....
000000f0 20 00 00 00 01 2c 08 80 40 00 0c 76 05 00 04 00   ....,..@..v....

[GvInterface::discovery] Device 'Specim-FX10e-077000034676' found (interface 169.254.0.1) user_id ' ' - MAC '70:f8:e7:b0:01:2c'
[GvDevice::new] Interface address = 169.254.0.1
[GvDevice::new] Device address = 169.254.146.234
packet_type  = cmd
command      = read-memory-cmd
size         = 8
id           = 65301
address      =        512 (0x00000200)
size         =        512 (0x00000200)
00000000 42 01 00 84 00 08 ff 15 00 00 02 00 00 00 02 00  B...............

packet_type  = cmd
[... 90 lines cut ...]
size         = 8
id           = 65313
address      =       2560 (0x00000a00)
value        =          0 (0x00000000)
00000000 42 01 00 82 00 08 ff 21 00 00 0a 00 00 00 00 00  B......!........

packet_type  = cmd
command      = write-register-cmd
size         = 8
id           = 65314
address      =       2560 (0x00000a00)
value        =          0 (0x00000000)
00000000 42 01 00 82 00 08 ff 22 00 00 0a 00 00 00 00 00  B......"........

packet_type  = cmd
command      = write-register-cmd
size         = 8
id           = 65315
address      =       2560 (0x00000a00)
value        =          0 (0x00000000)
00000000 42 01 00 82 00 08 ff 23 00 00 0a 00 00 00 00 00  B......#........

Device 'Specim-FX10e-077000034676' not found

Regarding the update I'll start now and post again when it is done.

horstrand, 2020-11-03:
I updated to the latest aravis stable version: 0.8.2

This is the output when I run: arv-tool-0.8 --debug=all

Found 0 USB3Vision device (among 10 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 169.254.0.1
[GvDiscoverSocket::new] Add interface 169.254.0.2
[GvDiscoverSocket::new] Add interface 10.13.22.71
[GvInterface::discovery] Device 'Specim-FX17e-076900048460' found (interface 169.254.0.2) user_id ' ' - MAC ' '
[GvInterface::discovery] Device 'Specim-FX10e-077000034676' found (interface 169.254.0.1) user_id ' ' - MAC ' '
Found 0 USB3Vision device (among 10 USB devices)
[GvDevice::new] Interface address = 169.254.0.1
[GvDevice::new] Device address = 169.254.181.33
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
...
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
...
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
Failed to open device 'Specim-FX10e-077000034676': Invalid Genicam data
Found 0 USB3Vision device (among 10 USB devices)
[GvDevice::new] Interface address = 169.254.0.2
[GvDevice::new] Device address = 169.254.116.199
[GvDevice::load_genicam] xml url = 'Local:FX17e.zip;400000;bf65' at 0x200
[GvDevice::load_genicam] Xml address = 0x400000 - size = 0xbf65 - FX17e.zip
[GvDevice::load_genicam] Zipped xml data
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
...
[Genicam::create_element] Unknown tag (Streamable)
[Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[Gc::set_default_node_data] Add 'GevSCSP'
[Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
[GvDevice::new] Legacy endianness handling = no
Specim-FX17e-076900048460 (169.254.116.199)

Please let me know if you need any additional information.
Thanks again for the support.

EmmanuelP, 2020-11-03:
Thanks. Aravis 0.8.2 is a bit more verbose. The FX10e camera does not seem to answer to read memory command. If I recall correctly, read memory should be mandatory.

What we could try is a network traffic capture using wireshark. Could you give it a go ?

horstrand, 2020-11-05:
I have tried wireshark on both network connections and this is the result when I run: arv-tool-0.8

On FX17 (now this camera is the one that is not connecting) I get the following result:

No.     Time           Source                Destination           Protocol Length Info
      1 0.000000000    169.254.0.2           255.255.255.255       GVCP     50     > DISCOVERY_CMD 

Frame 1: 50 bytes on wire (400 bits), 50 bytes captured (400 bits) on interface 0
Ethernet II, Src: AsustekC_50:45:e2 (9c:5c:8e:50:45:e2), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
Internet Protocol Version 4, Src: 169.254.0.2, Dst: 255.255.255.255
User Datagram Protocol, Src Port: 37065, Dst Port: 3956
GigE Vision Control Protocol

No.     Time           Source                Destination           Protocol Length Info
      2 0.042004913    169.254.141.162       169.254.0.2           GVCP     298    < DISCOVERY_ACK (Specim, 076900048460)

Frame 2: 298 bytes on wire (2384 bits), 298 bytes captured (2384 bits) on interface 0
Ethernet II, Src: Photonfo_04:8a (70:f8:e7:b0:04:8a), Dst: AsustekC_50:45:e2 (9c:5c:8e:50:45:e2)
Internet Protocol Version 4, Src: 169.254.141.162, Dst: 169.254.0.2
User Datagram Protocol, Src Port: 3956, Dst Port: 37065
GigE Vision Control Protocol

On FX10 I get the following result (shortened as it prints a lot of text):

No.     Time           Source                Destination           Protocol Length Info
      1 0.000000000    169.254.0.1           224.0.0.251           MDNS     78     Standard query 0x0000 PTR _privet._tcp.local, "QM" question

Frame 1: 78 bytes on wire (624 bits), 78 bytes captured (624 bits) on interface 0
Ethernet II, Src: AsustekC_50:45:e1 (9c:5c:8e:50:45:e1), Dst: IPv4mcast_fb (01:00:5e:00:00:fb)
Internet Protocol Version 4, Src: 169.254.0.1, Dst: 224.0.0.251
User Datagram Protocol, Src Port: 5353, Dst Port: 5353
Multicast Domain Name System (query)

No.     Time           Source                Destination           Protocol Length Info
      2 1.001293611    169.254.0.1           224.0.0.251           MDNS     78     Standard query 0x0000 PTR _privet._tcp.local, "QM" question

Frame 2: 78 bytes on wire (624 bits), 78 bytes captured (624 bits) on interface 0
Ethernet II, Src: AsustekC_50:45:e1 (9c:5c:8e:50:45:e1), Dst: IPv4mcast_fb (01:00:5e:00:00:fb)
Internet Protocol Version 4, Src: 169.254.0.1, Dst: 224.0.0.251
User Datagram Protocol, Src Port: 5353, Dst Port: 5353
Multicast Domain Name System (query)

No.     Time           Source                Destination           Protocol Length Info
      3 3.002309689    169.254.0.1           224.0.0.251           MDNS     78     Standard query 0x0000 PTR _privet._tcp.local, "QM" question

Frame 3: 78 bytes on wire (624 bits), 78 bytes captured (624 bits) on interface 0
Ethernet II, Src: AsustekC_50:45:e1 (9c:5c:8e:50:45:e1), Dst: IPv4mcast_fb (01:00:5e:00:00:fb)
Internet Protocol Version 4, Src: 169.254.0.1, Dst: 224.0.0.251
User Datagram Protocol, Src Port: 5353, Dst Port: 5353
Multicast Domain Name System (query)

No.     Time           Source                Destination           Protocol Length Info
      4 6.405833312    169.254.0.1           255.255.255.255       GVCP     50     > DISCOVERY_CMD 

Frame 4: 50 bytes on wire (400 bits), 50 bytes captured (400 bits) on interface 0
Ethernet II, Src: AsustekC_50:45:e1 (9c:5c:8e:50:45:e1), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
Internet Protocol Version 4, Src: 169.254.0.1, Dst: 255.255.255.255
User Datagram Protocol, Src Port: 51157, Dst Port: 3956
GigE Vision Control Protocol

No.     Time           Source                Destination           Protocol Length Info
      5 6.535851878    169.254.53.116        169.254.0.1           GVCP     298    < DISCOVERY_ACK (Specim, 077000034676)

Frame 5: 298 bytes on wire (2384 bits), 298 bytes captured (2384 bits) on interface 0
Ethernet II, Src: Photonfo_01:2c (70:f8:e7:b0:01:2c), Dst: AsustekC_50:45:e1 (9c:5c:8e:50:45:e1)
Internet Protocol Version 4, Src: 169.254.53.116, Dst: 169.254.0.1
User Datagram Protocol, Src Port: 3956, Dst Port: 51157
GigE Vision Control Protocol

No.     Time           Source                Destination           Protocol Length Info
[... 138 lines cut ...]
Ethernet II, Src: AsustekC_50:45:e1 (9c:5c:8e:50:45:e1), Dst: IPv6mcast_fb (33:33:00:00:00:fb)
Internet Protocol Version 6, Src: fe80::5e56:28f3:2d89:7660, Dst: ff02::fb
User Datagram Protocol, Src Port: 5353, Dst Port: 5353
Multicast Domain Name System (query)

No.     Time           Source                Destination           Protocol Length Info
    203 17.653980674   169.254.0.1           224.0.0.251           MDNS     160    Standard query 0x0000 PTR _ftp._tcp.local, "QM" question PTR _nfs._tcp.local, "QM" question PTR _afpovertcp._tcp.local, "QM" question PTR _smb._tcp.local, "QM" question PTR _sftp-ssh._tcp.local, "QM" question PTR _webdavs._tcp.local, "QM" question PTR _webdav._tcp.local, "QM" question

Frame 203: 160 bytes on wire (1280 bits), 160 bytes captured (1280 bits) on interface 0
Ethernet II, Src: AsustekC_50:45:e1 (9c:5c:8e:50:45:e1), Dst: IPv4mcast_fb (01:00:5e:00:00:fb)
Internet Protocol Version 4, Src: 169.254.0.1, Dst: 224.0.0.251
User Datagram Protocol, Src Port: 5353, Dst Port: 5353
Multicast Domain Name System (query)

No.     Time           Source                Destination           Protocol Length Info
    204 18.186376484   AsustekC_50:45:e1     Broadcast             ARP      42     Who has 169.254.141.162? Tell 169.254.0.2

Frame 204: 42 bytes on wire (336 bits), 42 bytes captured (336 bits) on interface 0
Ethernet II, Src: AsustekC_50:45:e1 (9c:5c:8e:50:45:e1), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
Address Resolution Protocol (request)

So just the discovery cmd and ack goes through the other connection. Other read cmds and acks are not.

Regards, 
Pablo

EmmanuelP, 2020-11-05:
Thanks Pablo. Could you attach the capture file, please ? It is easier to read, using wireshark, than the text output.

Also, could you give us some details on your network setup (configuration, how things are physically connected).

horstrand, 2020-11-05:
I have made a few screenshots. I hope they are visible.

FX17 (not connecting one):

[image]

FX10:

[image]

[image]

Regarding the network setup.  I have the asus z10pe-d16 ws mother board on my PC which has two network adapters (gigabit).
Both cameras are connected to these two ports.
I have set a manual IP to the connections:

169.254.0.1
169.254.0.2

EmmanuelP, 2020-11-05:
Please attach the pcap files.

horstrand, 2020-11-05:
Here they are:

[pcap_files.zip](https://github.com/AravisProject/aravis/files/5493066/pcap_files.zip)

EmmanuelP, 2020-11-05:
In the fx17 non working case, I don't see the readmem packets which should be emitted from the host to the camera. Is the trace complete ? Is there a firewall on your machine that could block them ?

horstrand, 2020-11-05:
That's right, just the discovery.
The trace is complete.
I have checked the status of my firewall and it is inactive. Anyway, I don't think this is a firewall related issue, because as soon as I unplug the FX10, the FX17 starts working fine.

EmmanuelP, 2020-11-05:
How do you configure the network capture ? I'm a bit puzzled nothing is sent from aravis, even to a wrong direction, after the reception of the discovery, while aravis logs show it tries to send readmem packets.

horstrand, 2020-11-05:
Sorry I did not understand what you meant by configuring the network capture.

Where do you see the aravis logs showing that it tries to send readmem packets? In the wireshark I see that the readmem packets are sent to only one ip address (169.254.53.116).

EmmanuelP, 2020-11-05:
> Sorry I did not understand what you meant by configuring the network capture.

You can configure capture filters or display filters, which means everything is not displayed or saved in the capture file. Normally, a lot of traffic happen on a network interface. I guess you have captured the traffic on a given interface. If aravis sends a packet on an another interface, it will not appear in the trace.

> Where do you see the aravis logs showing that it tries to send readmem packets? In the wireshark I see that the readmem packets are sent to only one ip address (169.254.53.116).

Well, that was based on the console output you have sent previously, where we see aravis sending readmem packets but never receive acknowledges.

horstrand, 2020-11-05:
Thanks for the clarification. Now I understand what you mean. I'm quite sure there are no filters applied for the network capture on the FX17 adapter, the trace is complete as it is. 

You are right that it is certainly puzzling. I ran again the command you suggested before: arv-tool-0.8 -n Specim-FX17e-076900048460 --debug=all:3

And obtained the following output.

Found 0 USB3Vision device (among 10 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 169.254.0.1
[GvDiscoverSocket::new] Add interface 169.254.0.2
[GvDiscoverSocket::new] Add interface 10.13.22.71
packet_type  = ack
command      = discovery-ack
size         = 248
id           = 65535
manufacturer = Specim
name         =  
model        = FX17e
address      = 169.254.155.64
00000000 00 00 00 03 00 f8 ff ff 00 01 00 02 80 00 00 01  ................
00000010 00 00 70 f8 e7 b0 04 8a 00 00 00 07 00 00 00 06  ..p.............
00000020 00 00 00 00 00 00 00 00 00 00 00 00 a9 fe 9b 40  ...............@
00000030 00 00 00 00 00 00 00 00 00 00 00 00 ff ff 00 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000050 53 70 65 63 69 6d 00 29 74 0d 02 00 00 00 04 00  Specim.)t.......
00000060 00 00 00 00 00 98 00 00 98 fe 7f 00 00 00 00 29  ...............)
00000070 46 58 31 37 65 00 00 00 00 00 00 00 00 11 00 00  FX17e...........
00000080 00 00 00 00 00 00 00 29 74 0d 02 00 00 00 00 29  .......)t......)
00000090 56 65 72 73 69 6f 6e 20 31 2e 35 00 72 73 69 6f  Version 1.5.rsio
000000a0 6e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  n...............
000000b0 53 70 65 63 69 6d 00 00 02 98 00 00 00 40 d6 29  Specim.......@.)
000000c0 74 0d 02 00 00 b9 00 00 00 00 00 00 00 00 01 00  t...............
000000d0 00 00 00 00 00 50 01 29 74 0d 02 00 00 00 00 00  .....P.)t.......
000000e0 30 37 36 39 30 30 30 34 38 34 36 30 00 00 00 00  076900048460....
000000f0 20 00 00 00 00 00 00 00 e8 08 01 00 00 00 00 00   ...............

[GvInterface::discovery] Device 'Specim-FX17e-076900048460' found (interface 169.254.0.2) user_id ' ' - MAC '70:f8:e7:b0:04:8a'
[GvDevice::new] Interface address = 169.254.0.2
[GvDevice::new] Device address = 169.254.155.64
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::read_memory] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
Device 'Specim-FX17e-076900048460' not found: Invalid Genicam data

I checked both interfaces and realized that on the FX17 nothing was being printed (just the discovery again) and on the other one it keeps asking "Who has 169.254.155.64", somehow overlapped in time with the send_commands from aravis.

Here is the pcap file: 

[fx17_network_adapter_of_fx10.zip](https://github.com/AravisProject/aravis/files/5493700/fx17_network_adapter_of_fx10.zip)

EmmanuelP, 2020-11-05:
Please try: `./src/arv-tool-0.8 control DeviceModelName -d cp:4`

And attach the console output and the capture of the 'any' input, which should capture the traffic on all the network adapters.
Make sure you don't have other data transfer over the network during the capture, and stop the capture as soon as the command is finished.

horstrand, 2020-11-05:
After executing: arv-tool-0.8 control Specim-FX17e-076900048460 -d cp:4
with both cameras connected, I get the following capture on Wireshark (selecting any):

[any_capture.zip](https://github.com/AravisProject/aravis/files/5494685/any_capture.zip)

And this is the log I get from aravis:

[aravis.log](https://github.com/AravisProject/aravis/files/5494686/aravis.log)

EmmanuelP, 2020-11-05:
Thanks for the traces.

I did not see anything obvious that would explain why the readmem packet is not even sent to the camera. Instead we got an ICMP packet telling the host is unreachable.

What is the network configuration for each interface ? Are they in a different subnet ?

horstrand, 2020-11-20:
Sorry for my late reply.

I was finally able to check if the cameras were in the same subnet and if this was causing the problem and it did actually!
I had to change to a persistent configuration and different subnets for each camera and now is working.

Apologies for the troubles caused. Thanks for the support.

I'll close the issue with this comment.

closed 2020-11-20 by horstrand