AravisProject/aravis #451: arv-viewer frames not displayed, how to debug.

bpinsard, 2020-11-30:
**Describe the bug**
I have  a problem with a camera where the frames are not displayed in arv-viewer, while it seems that frames are received.
This is what it looks like after stretching the window. 
[image]

Arv-camera-test statistics look good.
It could very well be a problem with the camera, but I am looking for a way to understand what is the problem.
Trying to use `-d debug` with arv-viewer does not produce further info.
Is there a way to set arv-viewer to a more verbose mode? 
Do you have any clue why this could be happening? (network settings, buffering issue....?)
Thanks.

**To Reproduce**
Steps to reproduce the behavior: depends on hardware

**Expected behavior**
I would expect debug to provide more verbose output. 

**Camera description:**
 - Manufacturer : MRC https://www.mrc-systems.de/en/products/mr-compatible-cameras 
 - Model: HiSpeed https://www.mrc-systems.de/en/products/mr-compatible-cameras#hispeed-mrcam 
 - Interface: Ethernet

**Platform description:**
 - Aravis version 0.8.5 (tested previous version as well)
 - OS:Ubuntu 18.04
 - Hardware x86_64

EmmanuelP, 2020-11-30:
It is probably an issue related to the gstreamer video output embedded in the gtk3 interface. I guess you have the same issue if you select the fake camera.

Are you using an X server or wayland ?

bpinsard, 2020-12-01:
As it is on Ubuntu 18 I would say that it is X server, and consider it a duplicate of #345 
Thanks for the pointer and explanations.

closed 2020-12-01 by bpinsard