AravisProject/aravis #454: Fail to get& set camera gain with aravis0.8

JiangXL, 2020-12-02:
**Describe the bug**
I use Dahua A3600CU60 U3V camera with aravis0.8(on fedora 33), but fail to get & set gain.

**To Reproduce**
1. connect camera to USB3 port.
2. run follow python code.
`
```python3
In [1]: import gi
        gi.require_version('Aravis', '0.8')
        gi.repository import Aravis
In [2]: camera =  Aravis.Camera.new('Dahua Technology-XXXXX')
In [3]: camera.set_frame_rate(10)

In [4]: camera.get_exposure_time()
Out[4]: 10000.0

In [5]: camera.get_gain()
----------------------------------------------------------------------
Error                                Traceback (most recent call last)
<ipython-input-5-c45f10c9a239> in <module>
----> 1 camera.get_gain()

Error: arv-device-error-quark: node 'GainRaw' [ArvGcFloatNode]  is not a ArvGcInteger (0)
In [6]: camera.get_gain_bounds()
---------------------------------------------------------------------------
Error                                     Traceback (most recent call last)
<ipython-input-6-8d27cfba1ab0> in <module>
----> 1 camera.get_gain_bounds()

Error: arv-device-error-quark: node 'GainRaw' [ArvGcFloatNode]  is not a ArvGcInteger (0)
```
3. If I open with arv-viewer, gain still doesn't work.

**Expected behavior**
As above.

**Camera description:**
 - Manufacturer Dahua Technology
 - Model A3600CU60
 - Interface USB3

**Platform description:**
 - Aravis version 0.8.5
 - OS: Fedora 33
 - Hardware [e.g. x86_64]

EmmanuelP, 2020-12-02:
Hi,

Please try this patch: https://github.com/AravisProject/aravis/pull/455

JiangXL, 2020-12-02:
Thanks! I will try it later.

closed 2020-12-04 by JiangXL