AravisProject/aravis #458: Provide a simple way to get an ArvCamera from an ArvDevice

lovesegfault, 2020-12-16:
**Is your feature request related to a problem? Please describe.**
In my code I need to, during setup, store the camera's IP and mac address. I've done this by using `arv_interface_get_device_serial_nbr` on all indices from 0 to `arv_interface_get_n_devices` until I find the index with the serial corresponding to the camera I want. I then use `arv_interface_get_physical_id` and `_get_device_address` to store the ip and mac addresses.

At this point I could, in theory, just call `_open_device` and from there get an `ArvCamera`, but currently there is no easy way of doing this. I ended up using `arv_camera_new(mySerial)` which is suboptimal since, as I understand it, it redoes part of the above logic to find the correct/desired camera.

**Describe the solution you'd like**
I wish there was a `arv_device_get_camera()` method that allowed for easy `ArvDevice` -> `ArvCamera` conversion.

**Describe alternatives you've considered**
I wish there was a simple way to query an `ArvCamera`'s IP and MAC.

lovesegfault, 2020-12-17:
While investigating this, and attempting to use `g_initable_new`. With @sdroege's help we found this:
https://github.com/AravisProject/aravis/blob/c2f21a085dff256fa5d1aa8da225ac6b4d22eb17/src/arvcamera.c#L3029

Which should actually be `g_value_dup_object()`, it seems. This ended up causing issues when using aravis from the Rust glib APIs.

sdroege, 2020-12-17:
Currently it would steal the reference from the caller, but `set_property()` should never ever do that. Just like it currently already duplicates the string a couple of lines below.

Fixing this needs some checking if there any callers that would start leaking if this is now changed.

EmmanuelP, 2020-12-18:
Hi,

I'm tempted to just fix aravis and use g_value_dup_object instead of g_value_get_object. This property was not really meant to be used outside of aravis.

EmmanuelP, 2020-12-18:
About the issue itself:

> I wish there was a arv_device_get_camera() method that allowed for easy ArvDevice -> ArvCamera conversion.

We could add a arv_camera_new_with_device method. This naming is common in glib/gtk.

arv_device_get_camera would imply there is an underlying camera object in device, while it is the other way around.

EmmanuelP, 2020-12-18:
Should be fixed by 93fdb3873119148f1f85d99586fcf4c0f3bfc517

closed 2020-12-18 by EmmanuelP

lovesegfault, 2020-12-21:
Thanks Emmanuel!