AravisProject/aravis #460: arv_camera_set_trigger fails if TriggerSelector does not have an entry AcquisitionStart

TIS-Tim, 2021-01-18:
**Describe the bug**
`arv_camera_set_trigger` fails if `TriggerSelector` does not have an entry `AcquisitionStart`.

**To Reproduce**
Call `arv_camera_set_trigger` on a camera lacking said entry

**Expected behavior**
Trigger mode being activated as expected.

**Camera description:**
The Imaging Source DFK 33GX273 [GigEVision]

**Platform description:**
 - Aravis 0.8

**Additional context**
Since the `AcquisititionStart` entry is optional, I think aravis should only try disabling that specific trigger if it is available, and not abort the whole function.
As a side note, `TriggerActivation` is optional as well (although I think I never saw a camera without it), failing to set should not let the whole function fail.

EmmanuelP, 2021-01-21:
Hi,

Thanks for the report. Looking at Genicam SNFC document, I have found: « This feature is generally mandatory for transmitters and transceivers of most Transport Layers. »

I fail to see why a camera would not implement this feature. In your device, how are we supposed to tell the device it can start acquisition ?

TIS-Tim, 2021-01-21:
Hi,

I think you are confusing the AcquisitonStart command (which is of course mandatory) with the TriggerSelector enumeration entry AquisitionStart (which is the way to execute that feature via a trigger).

The issue I think we found is related to the latter.

EmmanuelP, 2021-01-21:
Ok, sorry for the misunderstanding. I should have looked at the code first.

EmmanuelP, 2021-01-21:
Please let me know if the patch fixes the issue.

TIS-Tim, 2021-01-21:
It does. Thank you very much!

PR #461 (2021-01-21): camera_set_trigger: ignore error if AcquisitionStart entry is not found
Fix #460
commit b698b72b (2021-01-21): camera_set_trigger: ignore error if AcquisitionStart entry is not found (#461)

Fix #460
closed 2021-01-21 by EmmanuelP