AravisProject/aravis #463: Failure to compile on macOS 10.15.7

Louis-Demers, 2021-01-23:
**Describe the bug**

Latest OS, source code as of 2021-01-23
"meson build" fails complaining glib-2.0 not found

**To Reproduce**
download the code and cd to the directory
brew install gettext intltool gtk-doc libxml2 meson libusb
brew install gtk+3 gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad libnotify gnome-icon-theme
meson configure -Dviewer=enabled
meson build

fails with  the following
[louis-demers@Louis-Demers:Aravis] meson build        
The Meson build system
Version: 0.56.2
Source dir: /Users/louis-demers/Development/External/Aravis
Build dir: /Users/louis-demers/Development/External/Aravis/build
Build type: native build
Project name: aravis
Project version: 0.8.6
Using 'CFLAGS' from environment with value: '-Wshadow -arch x86_64'
C compiler for the host machine: cc (clang 11.0.3 "Apple clang version 11.0.3 (clang-1103.0.32.62)")
C linker for the host machine: cc ld64 556.6
C++ compiler for the host machine: c++ (clang 11.0.3 "Apple clang version 11.0.3 (clang-1103.0.32.62)")
C++ linker for the host machine: c++ ld64 556.6
Using 'CFLAGS' from environment with value: '-Wshadow -arch x86_64'
Host machine cpu family: x86_64
Host machine cpu: x86_64
Compiler for C supports arguments -Wduplicated-branches: NO 
Compiler for C supports arguments -Wimplicit-fallthrough: YES 
Compiler for C supports arguments -Wmisleading-indentation: NO 
Compiler for C supports arguments -Wstrict-prototypes: YES 
Compiler for C supports arguments -Wunused: YES 
Compiler for C supports arguments -Wdeclaration-after-statement: YES 
Compiler for C supports arguments -Wformat=2: YES 
Compiler for C supports arguments -Wimplicit-function-declaration: YES 
Compiler for C supports arguments -Winit-self: YES 
Compiler for C supports arguments -Wmissing-include-dirs: YES 
Compiler for C supports arguments -Wmissing-prototypes: YES 
Compiler for C supports arguments -Wpointer-arith: YES 
Compiler for C supports arguments -Wformat-nonliteral: YES 
Compiler for C supports arguments -Wenum-conversion: YES 
Found pkg-config: /usr/local/bin/pkg-config (0.29.2)
Found CMake: /usr/local/bin/cmake (3.18.2)
Run-time dependency glib-2.0 found: NO (tried pkgconfig, framework and cmake)

meson.build:40:0: ERROR: Dependency "glib-2.0" not found, tried pkgconfig, framework and cmake

A full log can be found at /Users/louis-demers/Development/External/Aravis/build/meson-logs/meson-log.txt

**Expected behavior**
compile ?

**Camera description:**

didn't make it  there. Was hoping to test with
- BFS-PGE-31S4M-C: 3.2 MP, Mono, 35 FPS, Sony IMX265
- Pleora iPort Analog-Pro

**Platform description:**
 - Aravis latent
 - OS: MacOS 10.15.7, Xcode tools 11.7 (11E801a) , latest brew !
 - MacBook Pro (16-inch, 2019), 2.3 GHz 8-Core Intel Core i9, 64 GB 2667 MHz DDR4

**Additional context**
putting a lot of hope that I can make this work !

Louis-Demers, 2021-01-23:
Ok, progress, brew did install glib-2.0 but claimed it was not link, suggesting I do "brew link glib". This filled complaining about permission. a search on stakcoverflwo suggest "sudo chown -R "$USER":admin /usr/local/*" and brew was happy. Bit of a sludge ...

Starting from scratch, "meson build" yields the 4 red flags

```
Compiler for C supports arguments -Wduplicated-branches: NO 
Compiler for C supports arguments -Wmisleading-indentation: NO 
Has header "linux/if_packet.h" : NO 
Run-time dependency gtk+-3.0 found: NO (tried pkgconfig, framework and cmake)

```
but gtk+-3.0 is indeed installed by brew ???

continuing on ninja -C build yield 14 warnings about declarations shadowing variable...

ninja -C build install produced 

```
ERROR: Error in gtkdoc helper script:

ERROR: ['/usr/local/bin/gtkdoc-mkhtml', '--path=/Users/louis-demers/Development/External/Aravis/docs/reference/aravis:/Users/louis-demers/Development/External/Aravis/build/docs/reference/aravis', 'aravis', '../aravis-docs.xml'] failed with status 5
I/O error : Attempt to load network entity http://docbook.sourceforge.net/release/xsl/current/html/chunk.xsl
warning: failed to load external entity "http://docbook.sourceforge.net/release/xsl/current/html/chunk.xsl"
compilation error: file /usr/local/Cellar/gtk-doc/1.32_2/share/gtk-doc/data/gtk-doc.xsl line 11 element import
xsl:import : unable to load http://docbook.sourceforge.net/release/xsl/current/html/chunk.xsl

```
I checked with curl and the file pointed by the url is accessible ... So no doc for me...

So now I have arv-tool and It sees my Flir BlackFly...

Next trying to compile with the viewer...

Louis-Demers, 2021-01-23:
Ok, next, the compilation yield 

```
 Run-time dependency gtk+-3.0 found: NO (tried framework and cmake)

../meson.build:77:0: ERROR: Could not generate cargs for gtk+-3.0:
Package harfbuzz was not found in the pkg-config search path.
Perhaps you should add the directory containing `harfbuzz.pc'
to the PKG_CONFIG_PATH environment variable
Package 'harfbuzz', required by 'pango', not found

A full log can be found at ...l/Aravis/build/meson-logs/meson-log.txt
FAILED: build.ninja 
/usr/local/Cellar/meson/0.56.2/libexec/bin/meson --internal regenerate /Users/louis-demers/Development/External/Aravis /Users/louis-demers/Development/External/Aravis/build --backend ninja
ninja: error: rebuilding 'build.ninja': subcommand failed

```
now that the permissions have been corrected in the previous steps,  I "brew instal harfbuzz" which asked me to relink, "brew link harfbuzz" and then the viewer compiled. 

Viewer sees my Blackfly and the fake device. The viewer displays image sequence from the fake camera, but only one image from the Blackfly. the bottom right displays "1 images/0 error". Maybe the Blackfly is single acquisition by default !

At least some progress. Hope I didn't bore anyone with my tribulation and that it might help someone...

Louis-Demers, 2021-01-24:
well, just discovered I could have just done

`brew install aravis 
`

which claims to support bigSur on Apple M1 cpus ! Will check on Monday at the office.

Louis-Demers, 2021-01-26:
Tested the brew install on Mac-mini M1. Seems to work (executables in tests directory) On Both 10.15.7 and 11.1, arv-viewer runs, detects cameras but often does faits at capturing and displaying image. Works 1 in 20 times....

often displays 

` WARNING **: 13:37:01.230: Failed to make stream thread high priority
`

arc-camera-test and arv-acquisition-test seem to work.

EmmanuelP, 2021-02-08:
Hi,

Compilation of aravis core (library and command line tools) is under continuous integration. Please have a look at .travis.yml, it contains the different step needed for the compilation. Unfortunately, I did not take the time to include viewer building.

> Tested the brew install on Mac-mini M1. Seems to work (executables in tests directory) On Both 10.15.7 and 11.1, arv-viewer runs, detects cameras but often does faits at capturing and displaying image. Works 1 in 20 times....

While gtk3+gstreamer performance on macOS is quite bad, my experience is not as bad as yours, running arv-viewer on a macbook pro from 2015. The maximum framerate is quite low compared to a similar machine running linux, but at least the viewer should be able to connect to the camera and show a video stream.

> WARNING **: 13:37:01.230: Failed to make stream thread high priority

This warning is harmless, there no realtime/high priority support for macOS in aravis code (and yes, Aravis should be fixed to not display it :( ).

Louis-Demers, 2021-02-08:
Hello, 

My interest in the viewer is only to validate the functionality of the Aravis package. I will integrate it in our own apps. so the performance hit because of gtk3+gstreamer is not of concern. If I can reliably suck images out of Aravis , I'll feed them to the native macOS display pipe.

Thanks for your hints.

EmmanuelP, 2021-02-09:
> While gtk3+gstreamer performance on macOS is quite bad, my experience is not as bad as yours, running arv-viewer on a macbook pro from 2015. The maximum framerate is quite low compared to a similar [...]

I have just retested, and indeed the viewer is not able to display anything from a GigEVision camera, while working quite well with an USB3Vision device. Using tests/arv-camera-test, I'm able to receive an almost full 1Gb/s bandwith from a GigEVision camera, which means the stream receiving code works fine. It looks like there is a priority conflict between the thread receiving the stream data, and the other arv-viewer threads.

>>  WARNING **: 13:37:01.230: Failed to make stream thread high priority
>
>This warning is harmless, there no realtime/high priority support for macOS in aravis code (and yes, Aravis should be fixed to not display it :( ).

It is removed in the latest release.

EmmanuelP, 2021-02-09:
> I have just retested, and indeed the viewer is not able to display anything from a GigEVision camera, while working quite well with an USB3Vision device. Using tests/arv-camera-test, I'm able to [...]

Wrong explanation. I had WIFI enabled as well as the ethernet adapter connected. For some reason, arv-camera-test used the ethernet port, while the viewer used the WIFI. With WIFI disabled, the viewer is able to display a 2048x1536 RGB8 video stream at 30Hz, with a sustained 94MB/s data stream.

[image]

I'm closing this issue.

Please reopen if you retry to compile aravis by yourself and still have issues. Make sure to have a look at aravis travis CI configuration and also at the brew recipe.

Cheers.

closed 2021-02-09 by EmmanuelP

Louis-Demers, 2021-02-11:
Thanks for the confirmation...

Louis-Demers, 2021-02-13:
Ok, back working on this and indeed, turning off WiFi, the viewer works great.  And on we go.

Thank you.