AravisProject/aravis #465: View camera simulator from another host

lililla, 2021-02-08:
Hi,
Thank you for this project first of all. This is my first post here.

I have two Centos 8 computers with aravis 0.8 with the idea that one uses it as a camera with the Aravis-Fake simulator running and the other visualizes it. Well in this case, from the client side, in the arv-viewer I see the camera but I cannot reproduce it, the play button disabled (arv-viewer very slow) and from the gst-launch I get the following error. “Could not read camera ‘Aravis-Fake-GV02’: Invalid Genicam data" 

On the other hand, if from the same client I also start its Aravis-Fake-GV01 camera, from the arv-viewer I can see both cameras and now I can view the camera of the other host (GV02) without problem and with arv-viewer without the slowness of before. With gstreamer I can view it without problem too.

That could be happening? Is there any other solution to get what I want to do? 

Thank you very much.

EmmanuelP, 2021-02-08:
Hi,

Thanks for you report.

> On the other hand, if from the same client I also start its Aravis-Fake-GV01 camera, from the arv-viewer I can see both cameras and now I can view the camera of the other host (GV02) without problem [...]

I have reproduced your issue. It is stranger that I though, when you are able to connect GV02, it is in fact GV01. If you chnage the pixel format for GV01, you get the same pixel format with GV02.

Definitely an issue.

EmmanuelP, 2021-02-08:
Closing as duplicate of #464 

Whatever the interface is set using -i parameter, arv-fake-gv-camera always llisten to 127.0.0.1.

closed 2021-02-08 by EmmanuelP