AravisProject/aravis #467: Device support: Emergent Vision polarisation camera (HR-12000SPM)

abrock, 2021-02-08:
**Describe the bug**

All images result in a timeout, no success getting any images out of the camera.

**To Reproduce**
Steps to reproduce the behavior:

Start arv-viewer-0.8, select the camera and try to get images.

**Expected behavior**

Images from the camera displayed by arv-viewer.

**Camera description:**
 - Manufacturer Emergent Vision
 - Model  [HR-12000SPM](https://emergentvisiontec.com/products/area-scan-cameras/10-gige-area-scan-cameras-hr-series/hr-12000/)
 - Interface GigE-Vision (SFP+ 10GigE)

**Platform description:**
 - Aravis version 6b1d0c77d636d82e28e7046afaf22cda2a743ed1
 - OS: Ubuntu 20.04 LTS
 - Hardware x86_64

**Additional context**

I have a Emergent Vision polarisation camera (HR-12000SPM) I'd like to use but I have problems.

I compiled and installed 6b1d0c77d636d82e28e7046afaf22cda2a743ed1 and I ran ninja test (OK: 8). I also ran the tests under valgrind:

1/6 evaluator                               OK       0.92 s
2/6 buffer                                  OK       0.82 s
3/6 misc                                    OK       0.77 s
4/6 fake                                    OK       2.67 s
5/6 fakegv                                  FAIL    14.55 s (killed by signal 5 SIGTRAP)
6/6 genicam                                 OK       1.47 s

I attached a zip file with the output of
`arv-tool-0.8 -d all:3 > arv-tool-log`
and the corresponding wireshark file

I think that works ok, I also used `arv-tool-0.8 genicam` to download the GeniCam XML file which looks ok.

Then I tried to capture some images using `arv-viewer-0.8 -d all:3 > arv-viewer-log` but the viewer shows lots of errors and no success.
I stopped the wireshark capture early to avoid large files for the attachment, I also ran a test were I captured more packages and I didn't found anything interesting after the first couple hundred.

abrock, 2021-02-08:
I clearly forgot to attach the [logs.zip](https://github.com/AravisProject/aravis/files/5946692/logs.zip)

EmmanuelP, 2021-02-08:
Hi,

Could you attach the output of `./tests/arv-camera-test -d all`, please ?

abrock, 2021-02-08:
Thank you for the super quick response, I'll do that on Wednesday.

abrock, 2021-02-10:
I let `./tests/arv-camera-test -d all` run until it stopped by itself which happened after 15 seconds. I recorded all data using wireshark and exported all packets from start to the first frame and from the last frame to the end: 
[logs2.zip](https://github.com/AravisProject/aravis/files/5957408/logs2.zip)

This is my setup for the network interface:

```
ens4f0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9000 qdisc mq state UP group default qlen 1000
    link/ether 6c:b3:11:3c:94:d0 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.2/24 scope global ens4f0
       valid_lft forever preferred_lft forever
    inet 192.168.1.2/24 scope global ens4f0
       valid_lft forever preferred_lft forever
    inet 192.168.2.2/24 scope global ens4f0
       valid_lft forever preferred_lft forever
    inet6 fe80::6eb3:11ff:fe3c:94d0/64 scope link 
       valid_lft forever preferred_lft forever
```

```
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
192.168.0.0     0.0.0.0         255.255.255.0   U     0      0        0 ens4f0
192.168.1.0     0.0.0.0         255.255.255.0   U     0      0        0 ens4f0
192.168.2.0     0.0.0.0         255.255.255.0   U     0      0        0 ens4f0
```

EmmanuelP, 2021-02-10:
Hi,

I have just had a quick look at the log files. I'm puzzled. There is some complaints about invalid address access, while it corresponds to a required register (GevSCSP0) which is the stream source port. Also, the frame ids don't make sense.

I need to spend more time on it. Don't know when though.