AravisProject/aravis #470: Problem receiving frame with a Pleora device

lililla, 2021-02-15:
Hi I’m a little new to this, I’m testing with a pleora simulator and aravis8-0-6. From arv-viewer and arv-tools-0.8 I can see the device correctly.

The problem I have is that I cannot display the image, I can only control the configuration from arv-tools-0.8 control to the device without any problem.

The simulator contains PixelMono8, PixelYUV422 and PixelBGR8 so I think this is not the problem.

It seems that the problem is that I am not receiving the stremeing, from ./arv-camera-test I check that I do not receive any frame.

I have configured the simulator with these ports, GVCP 3956 and in GVSP 20202. Firewalld, selinux and MTU disabled and configured.

Is the GVSP configuration correct?

Do I have to control any settings on the device using arv-tool-8.0?

Any other idea?

Thanks a lot.
Miguel Ángel.

EmmanuelP, 2021-02-22:
Hi,

Please run `arv-camera-test -d all` for a few seconds and attach the console output.

lililla, 2021-02-22:
Hi, this is the output of `arv-camera-test -d all`:
[output_pleora_camera_test.txt](https://github.com/AravisProject/aravis/files/6021344/output_pleora_camera_test.txt)

Thanks for your answer.

EmmanuelP, 2021-03-06:
Is the pleora simulator freely available ?

Could you attach the output of `arv-tool-8 values` ?

c10ud, 2021-09-09:
Hello, not sure if this is related but I cannot see video coming from a Pleora iPort-SB-GigE device connected to a Sony Block

Here is the output of arv-tool-0.8 values: https://pastebin.com/eCLQEszp
And the output of arv-camera-test-0.8 -d all: https://pastebin.com/Jaku5CiJ
And the output of 
GST_DEBUG=aravissrc:6 gst-launch-1.0 aravissrc camera-name="Pleora Technologies Inc.-iPORT-SB-GigE-PT01-SN4IP01-128x-00:11:1c:03:a1:70" ! video/x-raw,format=GRAY8,width=1280,height=720,framerate=30/1 ! videoconvert ! autovideosink

https://pastebin.com/P7JbSHZn

Thanks