AravisProject/aravis #479: Add an API for GigEVision IP settings

EmmanuelP, 2021-03-06:
We should add an API to manage the IP settings of GigEVision devices.

Here is an example on how to set a fixed IP address: https://aravis-project.discourse.group/t/is-it-possible-to-change-the-ip-of-a-connected-camera/226/3

Without the API, user has to convert the IP address and mask to an integer value. Also we need to add GigEVision standard feature for IP settings (GevPersistentIPAddress, GevPersistentSubnetMask, GevPersistentDefaultGateway, GevCurrentIPConfigurationDHCP, GevCurrentIPConfigurationPersistentIP, GevCurrentIPConfigurationLLA) in the Genicam data if they are not resent in device XML.

The API would be:

```
arv_camera_gv_set_persistent_ip_from_string (ArvCamera *camera, const char *ip, const char *mask, const char *gateway);
arv_camera_gv_set_persistent_ip (ArvCamera *camera, GInetAddress *ip, GInetAddressMask *mask, GInetAddress *gateway);
arv_camera_gv_set_ip_configuration_mode (ArvCamera *camera, ArvIPMode mode);
```

mshr-h, 2022-02-10:
Is anyone working on it?
If not, we'll try to implement it.

EmmanuelP, 2022-02-10:
Hi @mshr-h 

> Is anyone working on it? If not, we'll try to implement it.

No one to my knowledge.

mshr-h, 2022-02-17:
Hi @EmmanuelP 

I'm a bit confused about Aravis/GigE Vision/GenICam stuff. What do you mean by "if they are not resent in device XML"?

I thought we just have to add `arv_camera_gv_set_persistent_ip_from_string`, `arv_camera_gv_set_persistent_ip`, `arv_camera_gv_set_ip_configuration_mode` to [aravis/arvcamera.c](https://github.com/AravisProject/aravis/blob/main/src/arvcamera.c).

Is there anything we have to do except for that?

EmmanuelP, 2022-02-17:
Hi,

> What do you mean by "if they are not resent in device XML"?

It is a typo, the correct sentence is « if they are not present in the device XML data ».

Camera XML data don't always include all the feature definitions corresponding to the mandatory GigEVision registers. This is handled here:

https://github.com/AravisProject/aravis/blob/6c054bc7d8ace5961fcdfcdea93b1382e46f163a/src/arvgvdevice.c#L906-L1141

mshr-h, 2022-02-18:
Hi @EmmanuelP 

Thanks for your reply.

My understanding is that IP settings (GevPersistentIPAddress, GevPersistentSubnetMask, GevPersistentDefaultGateway, GevCurrentIPConfigurationDHCP, GevCurrentIPConfigurationPersistentIP, GevCurrentIPConfigurationLLA) are all optional features in [GenICam Standard Features Naming Convention](https://www.emva.org/wp-content/uploads/GenICam_SFNC_2_0_0.pdf). So we don't have to handle whether they're present in the Camera Description File or not. Is my understanding correct?

EmmanuelP, 2022-02-18:
They are optional in the Genicam SNFC as this standard applies also to USB3Vision and CameraLink protocols. I think they are mandatory in the GigEVision standard, but I don't know for sure, as we are not authorized to use the GigEVision standard documents for the development of Aravis.

PR #681 (2022-06-10): gigevision: add APIs for GigE Vision IP setting
This pull request adds some APIs for GigE Vision IP setting.

Fix #479
closed 2022-06-10 by EmmanuelP