AravisProject/aravis #480: cannot set feature in cyclingPresetMode, Teledyne DALSA Nano-C2050

ariprog, 2021-03-09:
**Describe the bug**
Cannot set some features of cyclingPresetMode.

`int main( int argc, char **argv )
{
	ArvCamera *camera;
	ArvDevice *device;
    ArvGc *genicam;
    ArvGcNode *node;
	ArvStream *stream;
	int i;

	arv_debug_enable( arv_option_debug_domains );

	camera = arv_camera_new( NULL, &error );
    check_error( "arv_camera_new" );

	if( ARV_IS_CAMERA( camera ) ) {
		void (*old_sigint_handler)(int);
		gint payload;

        printf( "vendor name %s\n", arv_camera_get_vendor_name( camera, NULL ) );
        printf( "model name %s\n", arv_camera_get_model_name( camera, NULL ) );

        arv_camera_set_string( camera, "cyclingPresetMode", "Off", &error );
        check_error( "cyclingPresetMode" );
        arv_camera_set_string( camera, "TriggerMode", "Off", &error );
        check_error( "TriggerMode" );

        arv_camera_set_integer( camera, "cP_PresetConfigurationSelector", 1, &error );
        check_error( "cP_PresetConfigurationSelector" );

        arv_camera_set_string( camera, "cP_LineSelector", "Line3", &error );
        check_error( "cP_LineSelector" );
        arv_camera_set_string( camera, "cP_OutputLineSource", "Off", &error );
        check_error( "cP_OutputLineSource" );
        arv_camera_set_string( camera, "cP_LineSelector", "Line4", &error );
        check_error( "cP_LineSelector" );
        arv_camera_set_string( camera, "cP_OutputLineSource", "SoftwareControlled", &error );
        check_error( "cP_OutputLineSource" );
        printf( "Line3 %s\n", arv_camera_get_string( camera, "cP_OutputLineSource", &error ) );
        check_error( "cP_OutputLineSource" );
        printf( "Line4 %s\n", arv_camera_get_string( camera, "cP_OutputLineSource", &error ) );
        check_error( "cP_OutputLineSource" );

        arv_camera_set_string( camera, "cP_LineSelector", "Line3", &error );
        check_error( "cP_LineSelector" );
        arv_camera_set_string( camera, "cP_OutputLineSource", "SoftwareControlled", &error );
        check_error( "cP_OutputLineSource" );
        arv_camera_set_string( camera, "cP_LineSelector", "Line4", &error );
        check_error( "cP_LineSelector" );
        arv_camera_set_string( camera, "cP_OutputLineSource", "Off", &error );
        check_error( "cP_OutputLineSource" );
        printf( "Line3 %s\n", arv_camera_get_string( camera, "cP_OutputLineSource", &error ) );
        check_error( "cP_OutputLineSource" );
        printf( "Line4 %s\n", arv_camera_get_string( camera, "cP_OutputLineSource", &error ) );
        check_error( "cP_OutputLineSource" );

        arv_camera_set_integer( camera, "cP_ExposureTime", 11111, &error );
        check_error( "cP_ExposureTime" );
        printf( "cP_ExposureTime %d\n", arv_camera_get_integer( camera, "cP_ExposureTime", &error ) );
        check_error( "cP_ExposureTime" );

        arv_camera_set_integer( camera, "cP_ExposureDelay", 33333, &error );
        check_error( "cP_ExposureDelay" );
        printf( "cP_ExposureDelay %d\n", arv_camera_get_integer( camera, "cP_ExposureDelay", &error ) );
        check_error( "cP_ExposureDelay" );

		g_object_unref (camera);
	}

	return 0;
}
`

The output is this:
```
vendor name Teledyne DALSA
model name Nano-C2050
Line3 SoftwareControlled # should be Off
Line4 SoftwareControlled
Line3 Off
Line4 Off # should be SoftwareControlled
cP_ExposureTime 0 # should be 1111
cP_ExposureDelay 33333
```

**To Reproduce**
See above program.

**Expected behavior**
Line3 and Line4 are the same but they should be different.
cP_ExposureTime should be 1111 insted I read 0 indipendent of what the value I set.

**Camera description:**
 - Manufacturer
 - Model
 - Interface [e.g. USB3]

**Platform description:**
 - Aravis version 0.8.6 compiled from source
 - OS: Manjaro Arm aarch64
 - Hardware: raspberry pi 4, 8 GB RAM
 - Camera: Teledyne DALSA Nano-C2050

**Additional context**
If I install their libraries it works.
The application, actually I'm using the python bindings, was working very well with aravis.
I mean I could set features, read streaming etc.
Then my client asked me to use the cyclingPresetMode and then I have this problem.

Thank you very much.
If you need more infos or tests, just ask.

EmmanuelP, 2021-03-09:
I have spotted this warning:

```
[GcPropertyNode::set_int64] Invalid linked node 'cnv_cP_ExposureTime'
[GcPropertyNode::get_int64] Invalid node 'cnv_cP_ExposureTime'
```
Could you attach the Genicam data given by `arv-tool-0.8 genicam` ?

EmmanuelP, 2021-03-09:
Please try https://github.com/AravisProject/aravis/pull/481

That should fix the cP_ExposureTime issue.

ariprog, 2021-03-09:
Wow, it works!!!
The cP_ExposureTime now works.
Many thanks.

EmmanuelP, 2021-03-09:
On the same branch, I have pushed another commit that should fix cP_OutputLineSource setting.

closed 2021-03-09 by ariprog