AravisProject/aravis #482: Getting errors while acquiring frames

Shorlaks, 2021-03-09:
**Describe the bug**
I am able to connect to the camera, but I don't get any frames.
Iv'e ran `arv-camera-test` and this is the result:

[image]

The camera works at 265 Hz, so basically every pocket I receive is broken.

**To Reproduce**
I followed the setup steps on a clean raspbian distro.
Also tried changing MTU from 1500 to 4088 (max), didn't do anything.

**Camera description:**
I tried 3 different cameras, same problem occurs on all 3 of them.

**Platform description:**
 - Aravis 0.8.6
 - OS: Raspbian latest version
 - Hardware rPi4 8

Thank you very much.

EmmanuelP, 2021-03-09:
Please retry with `--debug=all`.

Shorlaks, 2021-03-09:
Hi EmmanuelP.
Thank you for the quick response.
Iv'e ran it for a few seconds and pressed Ctrl+C

`pi@raspberrypi:~/Downloads/aravis-0.8.6/build/tests $ ./arv-camera-test --debug=all
Looking for the first available camera
Found 0 USB3Vision device (among 6 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 10.53.148.116
[GvDiscoverSocket::new] Add interface 169.254.94.131
[GvInterface::discovery] Device 'GigaLinx-GigE_Connect-10001' found (interface 169.254.94.131) user_id '' - MAC '00:50:c2:8c:f0:87'
[GvDevice::new] Interface address = 169.254.94.131
[GvDevice::new] Device address = 169.254.0.8
[GvDevice::load_genicam] xml url = 'Local:Rafael_Flir_GEV_v5_my.xml;C00F8000;617E' at 0x200
[GvDevice::load_genicam] Xml address = 0xc00f8000 - size = 0x617e - Rafael_Flir_GEV_v5_my.xml
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[Gc::set_default_node_data] Add 'DeviceVendorName'
[Gc::set_default_node_data] Add 'DeviceModelName'
[Gc::set_default_node_data] Add 'DeviceVersion'
[Gc::set_default_node_data] Add 'DeviceManufacturerInfo'
[Gc::set_default_node_data] Add 'DeviceID'
[Gc::set_default_node_data] Add 'GevStreamChannelCount'
[Gc::set_default_node_data] Add 'GevTimestampTickFrequency'
[Gc::set_default_node_data] Add 'GevSCPHostPort'
[Gc::set_default_node_data] Add 'GevSCPSDoNotFragment'
[Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[Gc::set_default_node_data] Add 'GevSCDA'
[Gc::set_default_node_data] Add 'GevSCSP'
[Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
[GvDevice::new] Legacy endianness handling = yes
vendor name           = GigaLinx
model name            = GigE_Connect
device id             = 10001
image width           = 480
image height          = 385
horizontal binning    = 0
vertical binning      = -1231105060
payload               = 369600 bytes
exposure              = 0 µs
gain                  = 0 dB
[GvDevice::create_stream] Number of stream channels = 1
[GvDevice::auto_packet_size] No GevSCPSFireTestPacket feature found
[GvStream::stream_new] Packet size = 8036 byte(s)
[GvStream::stream_new] Destination stream port = 42443
[GvStream::stream_new] Source stream port = 0
[GvStream::stream_thread] Packet timeout = 40 ms
[GvStream::stream_thread] Frame retention = 200 ms
[GvStream::loop] Standard socket method
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 384 ns
gv packet size        = 8036 bytes
[GvStream::missing_packet_check] Maximum number of requests reached at dt = 3333, n_packet_requests = 47 (49 packets/frame), frame_id = 2392
[GvStream::missing_packet_check] Maximum number of requests reached at dt = 3384, n_packet_requests = 47 (49 packets/frame), frame_id = 2393
[GvStream::missing_packet_check] Maximum number of requests reached at dt = 3416, n_packet_requests = 47 (49 packets/frame), frame_id = 2394
[GvStream::missing_packet_check] Maximum number of requests reached at dt = 3425, n_packet_requests = 47 (49 packets/frame), frame_id = 2395
[GvStream::missing_packet_check] Maximum number of requests reached at dt = 3486, n_packet_requests = 47 (49 packets/frame), frame_id = 2396
[GvStream::missing_packet_check] Maximum number of requests reached at dt = 3325, n_packet_requests = 47 (49 packets/frame), frame_id = 2397
[GvStream::missing_packet_check] Maximum number of requests reached at dt = 3432, n_packet_requests = 47 (49 packets/frame), frame_id = 2398
[GvStream::missing_packet_check] Maximum number of requests reached at dt = 3384, n_packet_requests = 47 (49 packets/frame), frame_id = 2399
[GvStream::missing_packet_check] Maximum number of requests reached at dt = 3478, n_packet_requests = 47 (49 packets/frame), frame_id = 2400
[GvStream::missing_packet_check] Maximum number of requests reached at dt = 3395, n_packet_requests = 47 (49 packets/frame), frame_id = 2401
[... 559 lines cut ...]
min     ;    1972
max     ;  105715
last max
at:     ;    2601
Counter =     2659
[GvStream::finalize] n_completed_buffers    = 0
[GvStream::finalize] n_failures             = 241
[GvStream::finalize] n_timeouts             = 241
[GvStream::finalize] n_aborteds             = 27
[GvStream::finalize] n_underruns            = 0
[GvStream::finalize] n_missing_frames       = 0
[GvStream::finalize] n_size_mismatch_errors = 0
[GvStream::finalize] n_received_packets     = 535
[GvStream::finalize] n_missing_packets      = 11568
[GvStream::finalize] n_error_packets        = 0
[GvStream::finalize] n_ignored_packets      = 0
[GvStream::finalize] n_resend_requests      = 0
[GvStream::finalize] n_resent_packets       = 0
[GvStream::finalize] n_resend_ratio_reached = 267
[GvStream::finalize] n_duplicated_packets   = 0
[Stream::finalize] Flush 22 buffer[s] in input queue
[Stream::finalize] Flush 28 buffer[s] in output queue
`

Thanks you, Shorlaks

EmmanuelP, 2021-03-09:
The packet size is probably to big, aravis only receives leader and trailer packets.

Try to set the packet size to 1500 (`-i 1500`)

Shorlaks, 2021-03-09:
Is that how i set the packet size?

[image]

Thanks,
Shorlaks

EmmanuelP, 2021-03-09:
Yes. What gives the debug output ?

EmmanuelP, 2021-03-10:
The log is missing the last part, just after you pressed ctrl+c.

EmmanuelP, 2021-03-10:
Also, did you try to run aravis from a standard machine ?

Shorlaks, 2021-03-10:
Oh, I accidentally pressed ctrl+c twice and missed the log.
I didn't try running aravis from a standard machine because it has to run on a raspberryPi.
Ill try running it on ubuntu instead of raspbian, maybe their drivers are better or abit different.

```
  bins  ;Buffer r
       0;       1
    2000;       0
    4000;       1
    6000;       0
    8000;       1
   10000;       0
   12000;       1
   14000;       0
   16000;       1
   18000;       0
   20000;       1
   22000;       0
   24000;       1
   26000;       1
   28000;       0
   30000;       1
   32000;       0
   34000;       1
   36000;       0
   38000;       1
   40000;       0
   42000;       1
   44000;       0
   46000;       1
   48000;       1
   50000;       0
   52000;       0
   54000;       1
   56000;       0
   58000;       1
   60000;       0
   62000;       1
   64000;       1
   66000;       0
   68000;       1
   70000;       0
   72000;       1
   74000;       0
   76000;       1
   78000;       0
   80000;       1
   82000;       0
   84000;       1
   86000;       0
   88000;       1
   90000;       1
   92000;       0
   94000;       1
   96000;       0
   98000;       1
  100000;       2
  102000;      10
  104000;       4
  106000;       5
  108000;       4
  110000;       4
  112000;       3
  114000;       5
  116000;       1
  118000;       8
  120000;       7
  122000;       6
  124000;       4
  126000;       8
  128000;       4
  130000;      12
  132000;       9
  134000;       7
  136000;      10
  138000;      11
  140000;      13
  142000;      13
  144000;      14
  146000;      18
  148000;      16
  150000;      13
  152000;      24
  154000;      18
  156000;      24
  158000;      19
  160000;      26
  162000;      20
  164000;      29
  166000;      39
  168000;      32
  170000;      34
  172000;      29
  174000;      32
  176000;      26
  178000;      29
  180000;      36
  182000;      39
  184000;      29
  186000;      35
  188000;      38
  190000;      33
  192000;      35
  194000;      32
  196000;      33
  198000;      27
-------------
>=200000;     209
<      0;       0
min     ;    1980
max     ;  224599
last max
at:     ;     415
Counter =     1289
[GvStream::finalize] n_completed_buffers    = 0
[GvStream::finalize] n_failures             = 1137
[GvStream::finalize] n_timeouts             = 1137
[GvStream::finalize] n_aborteds             = 34
[GvStream::finalize] n_underruns            = 1690
[GvStream::finalize] n_missing_frames       = 40
[GvStream::finalize] n_size_mismatch_errors = 0
[GvStream::finalize] n_received_packets     = 25274
[GvStream::finalize] n_missing_packets      = 286898
[GvStream::finalize] n_error_packets        = 0
[GvStream::finalize] n_ignored_packets      = 1695
[GvStream::finalize] n_resend_requests      = 1737861
[GvStream::finalize] n_resent_packets       = 1982
[GvStream::finalize] n_resend_ratio_reached = 331
[GvStream::finalize] n_duplicated_packets   = 18
[Stream::finalize] Flush 16 buffer[s] in input queue
[Stream::finalize] Flush 34 buffer[s] in output queue
[GvDevice::read_register] Ack reception timeout
[GvDevice::read_register] Ack reception timeout

```

EmmanuelP, 2021-03-10:
What is the load on your raspberry ?

There a lot of underruns, which should indicate a system overload.

Also, a lot of missing packets.

You may try to decrease the frame rate (-f), enable auto socket buffer size (-a), enable realtime (--realtime) or high priority (--high-priority), or try the packet socket running arv-camera-test as root.

Shorlaks, 2021-03-10:
OK, I've done some tests:
To your question, the rPi CPU load is 5%, memory load is 0.1%.
I tried running it on Ubuntu, I'm getting the same results (MTU is capped at 1500);
Tried setting auto socket buffer, enabling real-time and setting high priority, non of them did anything.

I lowered the camera's frequency from within the camera from 265 to 100, and it helped, I'm getting full frames now.
Only problem is I need 265 fps, not 100.

I guess the low MTU creates a bottleneck.
Do you have any ideas what should I try?
Thank you,
Shorlaks

Shorlaks, 2021-03-10:
Direct wired connection.
The raspberry's NIC MTU is capped at 1500, cant go any higher.

When I run 'arv-camera-test' I can limit the pocket size. (-i).
How can I limit the pocket size from within the arv-viewer?

EmmanuelP, 2021-03-10:
The packet size setting is not available in arv-viewer. As a workaround, you can set it using arv-tool or arv-camera-test before launching the viewer. If your camera supports it, you can save a default packet size on the camera. Please refer to your camera documentation.

I fear I have exhausted my list of tweaks related to performance issue, and what I'm able to do remotely.

Shorlaks, 2021-03-10:
Ok, Thank you very much, I appreciate it.
I'll keep tinkering with it, and ill post here my findings.

Shorlaks, 2021-03-15:
Hello EmmanuelP, I got some news.
The raspberry's internal network adapter supports up to 16k MTU, but the driver only supports up to 1.5k MTU.
A solution to this is patching some files and recompiling the kernel, a guide to this can be found [here](https://www.jeffgeerling.com/blog/2020/setting-9000-mtu-jumbo-frames-on-raspberry-pi-os).

Although the MTU now is 9k, I still experience some weird behavior.
The default packet size from the camera side is 8036
```
pi@raspberrypi:~/aravis-0.8.6/build/tests $ ./arv-camera-test 
Looking for the first available camera
vendor name           = GigaLinx
model name            = GigE_Connect
device id             = 10001
image width           = 480
image height          = 385
horizontal binning    = 0
vertical binning      = 1460294008
payload               = 369600 bytes
exposure              = 0 µs
gain                  = 0 dB
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 384 ns
gv packet size        = 8036 bytes
  0 frame/s  -       0 MiB/s - 238 errors
  0 frame/s  -       0 MiB/s - 265 errors
  0 frame/s  -       0 MiB/s - 266 errors
  0 frame/s  -       0 MiB/s - 265 errors
  0 frame/s  -       0 MiB/s - 265 errors
  0 frame/s  -       0 MiB/s - 266 errors
  0 frame/s  -       0 MiB/s - 265 errors
  0 frame/s  -       0 MiB/s - 265 errors
  0 frame/s  -       0 MiB/s - 266 errors
^C  0 frame/s  -       0 MiB/s - 265 errors
Completed buffers = 0
Failures          = 2626
Underruns         = 0
```
If I change it to 4088, I'm getting full frames but after a while it brakes.
```
pi@raspberrypi:~/aravis-0.8.6/build/tests $ ./arv-camera-test -i 4088
Looking for the first available camera
vendor name           = GigaLinx
model name            = GigE_Connect
device id             = 10001
image width           = 480
image height          = 385
horizontal binning    = 0
vertical binning      = 2027508088
payload               = 369600 bytes
exposure              = 0 µs
gain                  = 0 dB
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 384 ns
gv packet size        = 4088 bytes
263 frames/s -    97.2 MiB/s - 2 errors
265 frames/s -    97.9 MiB/s
266 frames/s -    98.3 MiB/s
265 frames/s -    97.9 MiB/s
264 frames/s -    97.6 MiB/s - 1 error
266 frames/s -    98.3 MiB/s
265 frames/s -    97.9 MiB/s
265 frames/s -    97.9 MiB/s
265 frames/s -    97.9 MiB/s
266 frames/s -    98.3 MiB/s
264 frames/s -    97.6 MiB/s - 1 error
265 frames/s -    97.9 MiB/s
265 frames/s -    97.9 MiB/s - 1 error
264 frames/s -    97.6 MiB/s - 1 error
264 frames/s -    97.6 MiB/s - 1 error
[... 218 lines cut ...]
<      0;       0
min     ;    3169
max     ;  122210
last max
at:     ;    1710
Counter =     2124
[GvStream::finalize] n_completed_buffers    = 2112
[GvStream::finalize] n_failures             = 12
[GvStream::finalize] n_timeouts             = 12
[GvStream::finalize] n_aborteds             = 0
[GvStream::finalize] n_underruns            = 0
[GvStream::finalize] n_missing_frames       = 0
[GvStream::finalize] n_size_mismatch_errors = 0
[GvStream::finalize] n_received_packets     = 199174
[GvStream::finalize] n_missing_packets      = 786
[GvStream::finalize] n_error_packets        = 0
[GvStream::finalize] n_ignored_packets      = 4
[GvStream::finalize] n_resend_requests      = 1036
[GvStream::finalize] n_resent_packets       = 812
[GvStream::finalize] n_resend_ratio_reached = 12
[GvStream::finalize] n_duplicated_packets   = 68
[Stream::finalize] Flush 50 buffer[s] in input queue
[Stream::finalize] Flush 0 buffer[s] in output queue
```
What is happening here?

EmmanuelP, 2021-03-15:
Hi,

> The default packet size from the camera side is 8036

To understand what is going wrong in this case, a capture of the network traffic may help. 

> If I change it to 4088, I'm getting full frames but after a while it brakes.

Some devices don't like to be asked too much for packet resend, and stop emitting resent packets. May be that is what is happening here.

You may try to decrease the packet request ratio threshold 

Also, a network capture with this packet size would help.

https://www.wireshark.org/docs/wsug_html_chunked/ChCustCommandLine.html

Shorlaks, 2021-03-17:
Hi EmmanuelP,
I was in the middle of logging the errors and taking screenshots from wire-shark and I stumbled upon a solution.
If you follow the guide i mentioned before for recompiling the kernel and you set the packet size to 6130 (-i 6130) everything works fine and there's no lost frames at all.
I've tested several cameras, their frame rates ranges from 30 to 265 with different pixel formats (all streams were around 100Mb/s) and I experience zero problems.
Thank you very much for Aravis and your help.

closed 2021-03-17 by Shorlaks