AravisProject/aravis #485: Image colours are biased towards yellow and green

martenf9, 2021-03-12:
Hello,

I'm new to Aravis. I want to publish live images from a **TIS Camera** to a ROS topic with camera_aravis package. But with one camera image **colours are biased** towards yellow, with another camera image colours are biased towards green. 

Images shown in **rqt_image_view** are biased as well as images shown in **Aravis Viewer**.
In **Tcam-Capture** (Software for recording with TIS Camera) the image colours are without any bias.

Is something wrong with colour space?

Chosen and only option in Tcam-Capture is **"Format: rggb"**
Chosen Pixel Format in Aravis Viewer is **"BayerRG8"**
Should there be any other option to choose? Because at the moment I have to decide between Mono8, Mono16 and BayerRG8.

I'm working on **Ubuntu 20.04** and installed **Aravis Version 0.6.4** from here https://download.gnome.org/sources/aravis/0.6/

Camera_aravis package for **ROS Noetic** is installed from following repository: https://github.com/fkie-forks/camera_aravis

Cameras are following:
The Imaging Source DFK Z12GP031
The Imaging Source DFK Z30GP031

I am attaching two images: One without any bias (taken with Tcam-Capture) and one with green biased colours (Taken with rqt_image_view)

I would really appreciate any help.
Thanks in advance.
[image]
[image]

makkarpov, 2021-03-12:
You should check output of `arv-tool-0.8 features` and then adjust features related to white balance. Probably your software does this automatically.

EmmanuelP, 2021-03-15:
@martenf9 Did you try to tweak the color balance ?

Could you attach the raw image data as saved using aravis viewer ? And the output of `arv-tool-0.8 features` ?

There is may be an issue with the debayerisation of the image.

martenf9, 2021-04-23:
Thanks @makkarpov, it helped!

closed 2021-04-23 by EmmanuelP