AravisProject/aravis #486: USB3Vision: GENCP_ACCESS_DENIED not supported

ArkadiuszRaj, 2021-03-14:
**Describe the bug**
GenICAM GenCP result code GENCP_ACCESS_DENIED (0x8006) is not supported.
In ARV_DEBIG log every time this result code is received, a [[UvDevice::write_memory] Unexpected answer (0x8006) is logged.
Then Aravis tries to resend command again and egain and evemtually fails with Timeout status.

ArvGvcpError enum defines this result code as ARV_GVCP_ERROR_ACCESS_DENIED but with different value.
https://www.emva.org/wp-content/uploads/GenCP_1.1.pdf states it shall be 0x8006

Anyway I see that for USB3Vision (arvuvcpprivate.h) there is no GenCP errors definion as it is for GIGe (arvgvcpprivate.h)
Is this a source of the problem?

**To Reproduce**
Hard to say. I have camera FLIR Blackfly S BFS-U3-200S6M which replays with such a results.

**Expected behavior**
GENCP_ACCESS_DENIED shall be supported and properly understood by Aravis.

**Camera description:**
 - Manufacturer : FLIR 
 - Model : Blackfly S BFS-U3-200S6M 
 - Interface : USB3

**Platform description:**
 - Aravis version : HEAD
 - OS: Ubuntu 20.04.2
 - Hardware x86_64

**Additional context**
Example log, setting exposure time to 80us and gain to 27dB
```
[Evaluator::set_double_variable] FROM = 80
[Evaluator::evaluate_as_double] Expression = 'FROM'
[Evaluator::parse_expression] FROM
[Evaluator::parse_expression] Found 1 items in expression
[Evaluator::parse_expression] 0 items in garbage list
[Evaluator::parse_expression] 1 items in token list
[Evaluator::evaluate_as_double] Parsing status = 0
(var) FROM = 80
[Evaluator::evaluate] Result = (double) 80
[GcRegisterNode::_set_integer_value] address = 0xc1004, value = 0x50
packet_type  = cmd
command      = write-memory-cmd
size         = 12
id           = 65497
address      = 0x00000000000c1004
00000000 55 33 56 43 00 40 02 08 0c 00 d9 ff 04 10 0c 00  U3VC.@..........
00000010 00 00 00 00 50 00 00 00                          ....P...        
packet_type  = (null)
command      = write-memory-ack
size         = 4
id           = 65497
written      =          0 (0x00000000)
00000000 55 33 56 43 06 80 03 08 04 00 d9 ff 00 00 00 00  U3VC............

[[UvDevice::write_memory] Unexpected answer (0x8006)
packet_type  = cmd
command      = write-memory-cmd
size         = 12
id           = 65498
address      = 0x00000000000c1004
00000000 55 33 56 43 00 40 02 08 0c 00 da ff 04 10 0c 00  U3VC.@..........
00000010 00 00 00 00 50 00 00 00                          ....P...        
packet_type  = (null)
command      = write-memory-ack
size         = 4
id           = 65498
written      =          0 (0x00000000)
00000000 55 33 56 43 06 80 03 08 04 00 da ff 00 00 00 00  U3VC............

[[UvDevice::write_memory] Unexpected answer (0x8006)
packet_type  = cmd
command      = write-memory-cmd
size         = 12
id           = 65499
address      = 0x00000000000c1004
00000000 55 33 56 43 00 40 02 08 0c 00 db ff 04 10 0c 00  U3VC.@..........
00000010 00 00 00 00 50 00 00 00                          ....P...        
packet_type  = (null)
command      = write-memory-ack
size         = 4
id           = 65499
written      =          0 (0x00000000)
00000000 55 33 56 43 06 80 03 08 04 00 db ff 00 00 00 00  U3VC............

[[UvDevice::write_memory] Unexpected answer (0x8006)
packet_type  = cmd
command      = write-memory-cmd
size         = 12
id           = 65500
address      = 0x00000000000c1004
00000000 55 33 56 43 00 40 02 08 0c 00 dc ff 04 10 0c 00  U3VC.@..........
00000010 00 00 00 00 50 00 00 00                          ....P...        
packet_type  = (null)
command      = write-memory-ack
size         = 4
id           = 65500
written      =          0 (0x00000000)
00000000 55 33 56 43 06 80 03 08 04 00 dc ff 00 00 00 00  U3VC............

[[UvDevice::write_memory] Unexpected answer (0x8006)
packet_type  = cmd
command      = write-memory-cmd
size         = 12
id           = 65501
address      = 0x00000000000c1004
00000000 55 33 56 43 00 40 02 08 0c 00 dd ff 04 10 0c 00  U3VC.@..........
00000010 00 00 00 00 50 00 00 00                          ....P...        
packet_type  = (null)
command      = write-memory-ack
size         = 4
id           = 65501
written      =          0 (0x00000000)
00000000 55 33 56 43 06 80 03 08 04 00 dd ff 00 00 00 00  U3VC............

[[UvDevice::write_memory] Unexpected answer (0x8006)
[Evaluator::set_double_variable] FROM = 27
[Evaluator::evaluate_as_double] Expression = 'FROM*(2**14)/6.020599913'
[Evaluator::parse_expression] FROM*(2**14)/6.020599913
[Evaluator::parse_expression] Found 9 items in expression
[Evaluator::parse_expression] 2 items in garbage list
[Evaluator::parse_expression] 7 items in token list
[Evaluator::evaluate_as_double] Parsing status = 0
(var) FROM = 27
(int64) 2
(int64) 14
(operator) **
(operator) *
(double) 6.0206
(operator) /
[Evaluator::evaluate] Result = (double) 73475.7
packet_type  = cmd
command      = read-memory-cmd
size         = 12
id           = 65502
address      = 0x0000000000104400
size         =          4 (0x00000004)
00000000 55 33 56 43 00 40 00 08 0c 00 de ff 00 44 10 00  U3VC.@.......D..
00000010 00 00 00 00 00 00 04 00                          ........        
packet_type  = ack
command      = read-memory-ack
size         = 4
id           = 65502
00000000 55 33 56 43 00 00 01 08 04 00 de ff 01 00 00 f8  U3VC............

```

EmmanuelP, 2021-03-16:
Please try #489

EmmanuelP, 2021-03-19:
Should be fixed by #489. Please reopen if the issue persists.

closed 2021-03-19 by EmmanuelP