AravisProject/aravis #487: Add support for ExposureMode

ArkadiuszRaj, 2021-03-14:
**Describe the solution you'd like**
We develop the firmware that need very precise start & stop of exposition. Defined by external GPS synced clock.
So the application is aware of the frame timestamp with requested precision.

We have a FLIR camera that supports ExposureMode setting with ability to steer start & stop of light acquisition.

**Describe alternatives you've considered**
SpinnakerAPI

**Additional context**
Add new enum ArvExposureMode with values like TIMED (using ExposureTime) and TRIGGER_WIDTH (using Frame Trigger signal pulse width).

Add new function:
```
void arv_camera_set_exposure_mode (ArvCamera *camera, ArvExposureMode value);
```

EmmanuelP, 2021-03-15:
Please try #488

EmmanuelP, 2021-03-19:
Sould be fixed #488. Please reopen if there is an issue in the implementation.

closed 2021-03-19 by EmmanuelP