AravisProject/aravis #490: Dalsa Calibir - can not set packet size

Shorlaks, 2021-03-17:
**Describe the bug**
I'm trying to connect to a **Teledyne DALSA-Calibir GXM640** camera.
I want to set the packet size, but I cant set a packet size higher then 64 bytes.

```
pi@raspberrypi:~/aravis-0.8.6/build/tests $ ./arv-camera-test -i 6130
Looking for the first available camera
vendor name           = Teledyne DALSA
model name            = Calibir GXM640
device id             = S1241494
image width           = -2147352576
image height          = -536805376
horizontal binning    = 0
vertical binning      = -2124787336
payload               = 0 bytes
exposure              = 0 µs
gain                  = 0 dB
gv n_stream channels  = 16777216
gv current channel    = 0
gv packet delay       = 0 ns
gv packet size        = 64 bytes
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
^C  0 frame/s  -       0 MiB/s
Completed buffers = 0
Failures          = 0
Underruns         = 0
Control lost
```
With debug
```
pi@raspberrypi:~/aravis-0.8.6/build/tests $ ./arv-camera-test -i 6130 --debug=all
Looking for the first available camera
Found 0 USB3Vision device (among 6 USB devices)
[GvDiscoverSocket::new] Add interface 127.0.0.1
[GvDiscoverSocket::new] Add interface 10.53.148.116
[GvDiscoverSocket::new] Add interface 169.254.94.131
[GvInterface::discovery] Device 'Teledyne DALSA-Calibir GXM640-S1241494' found (interface 169.254.94.131) user_id '' - MAC '00:01:0d:c5:51:01'
[GvDevice::new] Interface address = 169.254.94.131
[GvDevice::new] Device address = 169.254.2.81
[GvDevice::load_genicam] xml url = 'Local:teledynedalsa_calibir_0_2_53330331.xml;a000;4871a' at 0x200
[GvDevice::load_genicam] Xml address = 0xa000 - size = 0x4871a - teledynedalsa_calibir_0_2_53330331.xml
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[Genicam::create_element] Unknown tag (IsDeprecated)
[Genicam::create_element] Unknown tag (IsDeprecated)
[Genicam::create_element] Unknown tag (IsDeprecated)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (ValidValueSet)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (IsDeprecated)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)[Gc::set_default_node_data] Add 'DeviceID'
[Gc::set_default_node_data] Add 'GevTimestampTickFrequency'
[Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[GvDevice::new] Packet resend = yes
[GvDevice::new] Write memory = yes
[GvDevice::new] Legacy endianness handling = no
vendor name           = Teledyne DALSA
model name            = Calibir GXM640
device id             = S1241494
image width           = -2147352576
image height          = -536805376
horizontal binning    = 0
vertical binning      = 1454395768
payload               = 0 bytes
exposure              = 0 µs
gain                  = 0 dB
[GvDevice::create_stream] Number of stream channels = 16777216
[GvDevice::auto_packet_size] Try packet size = 1000
[GvDevice::auto_packet_size] Try packet size = -2147483116
[GvDevice::auto_packet_size] Try packet size = -2147483148
[GvDevice::auto_packet_size] Packet size set to 64 bytes
[GvStream::stream_new] Packet size = 64 byte(s)
[GvStream::stream_new] Destination stream port = 33172
[GvStream::stream_new] Source stream port = 0
[GvStream::stream_thread] Packet timeout = 40 ms
[GvStream::stream_thread] Frame retention = 200 ms
[GvStream::loop] Standard socket method
gv n_stream channels  = 16777216
gv current channel    = 0
gv packet delay       = 0 ns
gv packet size        = 64 bytes
  0 frame/s  -       0 MiB/s
[GvDevice::read_register] Ack reception timeout
[GvDevice::read_register] Ack reception timeout
[GvDevice::read_register] Ack reception timeout
[GvDevice::read_register] Ack reception timeout
  0 frame/s  -       0 MiB/s
Completed buffers = 0
Failures          = 0
Underruns         = 0
[GvDevice::write_memory] Ack reception timeout
[GvDevice::write_memory] Ack reception timeout
  bins  ;Buffer r
       0;       0
-------------
>=  2000;       0
<      0;       0
min     ;     n/a
max     ;     n/a
last max
at:     ;       0
Counter =        0
[GvStream::finalize] n_completed_buffers    = 0
[GvStream::finalize] n_failures             = 0
[GvStream::finalize] n_timeouts             = 0
[GvStream::finalize] n_aborteds             = 0
[GvStream::finalize] n_underruns            = 0
[GvStream::finalize] n_missing_frames       = 0
[GvStream::finalize] n_size_mismatch_errors = 0
[GvStream::finalize] n_received_packets     = 0
[GvStream::finalize] n_missing_packets      = 0
[GvStream::finalize] n_error_packets        = 0
[GvStream::finalize] n_ignored_packets      = 0
[GvStream::finalize] n_resend_requests      = 0
[GvStream::finalize] n_resent_packets       = 0
[GvStream::finalize] n_resend_ratio_reached = 0
[GvStream::finalize] n_duplicated_packets   = 0
[Stream::finalize] Flush 50 buffer[s] in input queue
[Stream::finalize] Flush 0 buffer[s] in output queue
[GvDevice::write_register] Ack reception timeout
[GvDevice::write_register] Ack reception timeout
```
I also included a wireshark log:
[wireshark_log.zip](https://github.com/AravisProject/aravis/files/6155214/wireshark_log.zip)

**Platform description:**
 - Aravis 0.8.6
 - OS: Raspbian
 - Hardware RaspberryPi 4

Is this problem occurs on my side? or maybe the camera implements some protocols differently?
Have you ever stumbled across anything like this? and what can I do?
Thank you,
Shorlaks

thompd27, 2021-04-30:
I have the same issue with the previous generation of Calibir cameras. I've never been able to get them to open with Aravis.

EmmanuelP, 2021-05-01:
This is another occurence of the Genicam legacy mode issue. Please try this workaround: https://github.com/AravisProject/aravis/issues/444#issuecomment-730263537

Please attach the Genicam data of your camera, obtained using `arv-tool-0.8 genicam`.

Shorlaks, 2021-05-01:
Hi @EmmanuelP
Here is the xml: 
[genicam.zip](https://github.com/AravisProject/aravis/files/6409114/genicam.zip)

I think I know what the problem is, the xml has some strange characters like `’`, `°` and `µ` and the genicam's xmlParser is not able to handle them

EmmanuelP, 2021-05-03:
> I think I know what the problem is, the xml has some strange characters like ’, ° and µ and the genicam's xmlParser is not able to handle them

As long as these characters are UTF-8 encoded, the XML parser should not have issue with them.

Did you try the legacy workaround ?

Shorlaks, 2021-05-03:
> Did you try the legacy workaround ?

Yes, it worked, thank you very much

EmmanuelP, 2021-05-03:
One last thing. Could you please  update Aravis to the latest version, and attach the output of `arv-camera-test-0.8 -d all` ?

That will provide some additional informations about the legacy issue.

Shorlaks, 2021-05-03:
```
pi@raspberrypi:~/aravis-0.8.9/build/src $ ./arv-camera-test-0.8 -d all
Looking for the first available camera
[11:55:06.275] ❷ interface> Found 0 USB3Vision device (among 5 USB devices)
[11:55:06.276] ❷ interface> [GvDiscoverSocket::new] Add interface 127.0.0.1
[11:55:06.277] ❷ interface> [GvDiscoverSocket::new] Add interface 169.254.76.176
[11:55:06.278] ❷ interface> [GvInterface::discovery] Device 'Teledyne DALSA-Calibir GXM640-S1241494' found (interface 169.254.76.176) user_id '' - MAC '00:01:0d:c5:51:01'
[11:55:06.278] ❷ device> [GvDevice::new] Interface address = 169.254.76.176
[11:55:06.278] ❷ device> [GvDevice::new] Device address = 169.254.2.81
[11:55:06.279] ❷ device> [GvDevice::load_genicam] xml url = 'Local:teledynedalsa_calibir_0_2_53330331.xml;a000;4871a' at 0x200
[11:55:06.279] ❷ device> [GvDevice::load_genicam] Xml address = 0xa000 - size = 0x4871a - teledynedalsa_calibir_0_2_53330331.xml
[11:55:06.365] ❷ dom> [Genicam::create_element] Unknown tag (IsDeprecated)
[11:55:06.365] ❷ dom> [Genicam::create_element] Unknown tag (IsDeprecated)
[11:55:06.365] ❷ dom> [Genicam::create_element] Unknown tag (IsDeprecated)
[11:55:06.390] ❷ dom> [Genicam::create_element] Unknown tag (ValidValueSet)
[11:55:06.393] ❷ dom> [Genicam::create_element] Unknown tag (IsDeprecated)
[11:55:06.424] ❷ dom> [Genicam::create_element] Unknown tag (IsDeprecated)
[11:55:06.431] ❷ dom> [Genicam::create_element] Unknown tag (IsDeprecated)
[11:55:06.434] ❷ dom> [Genicam::create_element] Unknown tag (IsDeprecated)
[11:55:06.440] ❷ genicam> [Gc::set_default_node_data] Add 'DeviceID'
[11:55:06.440] ❷ genicam> [Gc::set_default_node_data] Add 'GevTimestampTickFrequency'
[11:55:06.441] ❷ genicam> [Gc::set_default_node_data] Add 'GevSCPSBigEndian'
[11:55:06.441] ❷ genicam> [Gc::set_default_node_data] Add 'GevSCPAddrCalc'
[11:55:06.442] ❷ device> [GvDevice::new] Device endianness = big
[11:55:06.442] ❷ device> [GvDevice::new] Packet resend     = yes
[11:55:06.442] ❷ device> [GvDevice::new] Write memory      = yes
[11:55:06.442] ❷ device> [GvDevice::new] Legacy endianness handling = no
vendor name           = eleT
model name            = ilaC
device serial number  = 421S
image width           = 640
image height          = 480
payload               = 307392 bytes
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 0 ns
gv packet size        = 1500 bytes
[11:55:06.445] ❷ device> [GvDevice::create_stream] Number of stream channels = 1
[11:55:06.446] ❷ device> [GvStream::stream_new] Packet size = 1500 byte(s)
[11:55:06.446] ❷ stream> [GvStream::stream_new] Destination stream port = 44475
[11:55:06.446] ❷ stream> [GvStream::stream_new] Source stream port = 49152
[11:55:06.447] ❷ stream-thread> [GvStream::stream_thread] Packet timeout = 20 ms
[11:55:06.447] ❷ stream-thread> [GvStream::stream_thread] Frame retention = 100 ms
[11:55:06.447] ❷ stream> [GvStream::loop] Standard socket method
  0 frame/s  -       0 MiB/s
  0 frame/s  -       0 MiB/s
[11:55:09.261] ❷ stream-thread> [GvStream::missing_packet_check] Maximum number of requests reached at dt = 4405, n_packet_requests = 61 (212 packets/frame), frame_id = 9
[11:55:09.264] ❷ stream-thread> [GvStream::process_packet] Error packet at dt = 207410, packet id = 174 frame id = 3
[11:55:09.265] ❷ sp> packet_type  = packet-unavailable (0x800c)
content_type = data-block (0x0003)
frame_id     =        3 
packet_id    =      174
data_size    =        0

[11:55:09.265] ❷ stream-thread> [GvStream::process_packet] Error packet at dt = 208344, packet id = 175 frame id = 3
[11:55:09.266] ❷ sp> packet_type  = packet-unavailable (0x800c)
content_type = data-block (0x0003)
frame_id     =        3 
packet_id    =      175
data_size    =        0

[11:55:09.266] ❷ stream-thread> [GvStream::process_packet] Error packet at dt = 208924, packet id = 192 frame id = 3
[11:55:09.266] ❷ sp> packet_type  = packet-unavailable (0x800c)
content_type = data-block (0x0003)
frame_id     =        3 
packet_id    =      192
data_size    =        0

[11:55:09.266] ❷ stream-thread> [GvStream::process_packet] Error packet at dt = 209455, packet id = 197 frame id = 3
[11:55:09.267] ❷ sp> packet_type  = packet-unavailable (0x800c)
[... 436 lines cut ...]
<      0;       0
min     ;    4733
max     ;  291663
last max
at:     ;       7
Counter =      137
[11:55:13.585] ❷ stream> [GvStream::finalize] n_completed_buffers    = 130
[11:55:13.585] ❷ stream> [GvStream::finalize] n_failures             = 7
[11:55:13.585] ❷ stream> [GvStream::finalize] n_timeouts             = 7
[11:55:13.585] ❷ stream> [GvStream::finalize] n_aborteds             = 0
[11:55:13.585] ❷ stream> [GvStream::finalize] n_underruns            = 0
[11:55:13.585] ❷ stream> [GvStream::finalize] n_missing_frames       = 0
[11:55:13.585] ❷ stream> [GvStream::finalize] n_size_mismatch_errors = 0
[11:55:13.585] ❷ stream> [GvStream::finalize] n_received_packets     = 28467
[11:55:13.585] ❷ stream> [GvStream::finalize] n_missing_packets      = 1162
[11:55:13.585] ❷ stream> [GvStream::finalize] n_error_packets        = 49
[11:55:13.586] ❷ stream> [GvStream::finalize] n_ignored_packets      = 0
[11:55:13.586] ❷ stream> [GvStream::finalize] n_resend_requests      = 4225
[11:55:13.586] ❷ stream> [GvStream::finalize] n_resent_packets       = 2026
[11:55:13.586] ❷ stream> [GvStream::finalize] n_resend_ratio_reached = 3
[11:55:13.586] ❷ stream> [GvStream::finalize] n_duplicated_packets   = 0
[11:55:13.586] ❷ stream> [Stream::finalize] Flush 50 buffer[s] in input queue
[11:55:13.586] ❷ stream> [Stream::finalize] Flush 0 buffer[s] in output queue

```

commit 1a9dbcd3 (2023-06-06): gv: add TeledyneDALSA ICE to legacy endianness mechanism list

Fix #490
closed 2023-06-06 by EmmanuelP