AravisProject/aravis #491: Reacting to events sent by camera

abrock, 2021-03-18:
**Is your feature request related to a problem? Please describe.**

Some cameras send event notifications for events like RisingEdge on some signal input, frame start etc.
For an example see [Baumer VLXT documentation](https://www.baumer.com/medias/__secure__/Baumer_UsersGuide-VLXT.I.FO-v18_EN_20210310_MNL.pdf?mediaPK=9000800649246), section 7.11 on page 107.
I have problems with triggering those cameras and I'd like to debug it.
I searched the Aravis source code for "event" but didn't find anything that looked like it allowed me to get notified if a camera sends one of those events.

**Describe the solution you'd like**

I'd like to have some kind of mechanism where I specify a callback function which is then called when an event notification is received. Perhaps similar to the current mechanism for receiving frames `g_signal_connect(stream,"new-buffer", G_CALLBACK(my_buffer_handler), &my_cam_data_struct)`

1873505600, 2023-10-08:
Hello, is the event or event data already in the plan, or is it somewhere else in the source code that I didn't find relevant

EmmanuelP, 2023-10-08:
Event support is not implemented, but it is definitely in Aravis project scope. Merge request welcome !

1873505600, 2023-10-08:
> Event support is not implemented, but it is definitely in Aravis project scope. Merge request welcome !

Maybe I can try to add the related source code, but I need a genicam protocol file to determine the protocol version. I haven't found the related file link in the wiki and readme file, could you please give me a related file link

EmmanuelP, 2023-10-08:
The genicam specification does only define an API for event. The interesting parts are in the GigEVision and USB3Vision protocols, but the use of these specifications for an open source implementation is forbidden, everything has to be done by reverse engineering. If you have read these documents, you can not contribute to the related code in aravis (arv_uv_* and arv_gv_* source code).

To reverse engineer the protocol, one has to capture the data exchanged between the device and a proprietary sdk, using wireshark, and find the packet related to events.

1873505600, 2023-10-08:
> The genicam specification does only define an API for event. The interesting parts are in the GigEVision and USB3Vision protocols, but the use of these specifications for an open source [...]

Thank you for your answer. I have seen the GigEVision 2.0 standard, but since this standard cannot be used directly for aravis, and I do not have equipment for reverse analysis at the moment, the merge request has to be postponed

dkogan, 2024-03-13:
Hello. I'm trying to use bulk transfers for my camera (https://aravis-project.discourse.group/t/how-to-execute-bulk-serial-transfers-in-aravis/728, https://github.com/AravisProject/aravis/issues/883).

I have the "correct" packet trace from running the vendor's code, and it's using an Event to get the data back. Should I look at implementing this, or do you want to collaborate? I'm not an expert in this, and it would go much faster if I had help.

Thanks

EmmanuelP, 2024-03-13:
Hi Dima,

Any help in trying to implement events would be appreciated. I don't really have time currently to work on this, but I can help in understanding the protocol and designing the API.

If you are willing to work on this feature, it is better to open a draft merge request as soon as possible, don't wait to have something working before. Even a failed attempt can be useful to someone who want to give it a try later.