AravisProject/aravis #492: Flir camera only starting after Spinview

rogerartigaspursals, 2021-03-20:
I have succesfully integrated aravis with Baumer and Sentech USB3 cameras. Recently, I started to work with a Flir Camera, model BFS-U3-16S2. The camera is recognized, but no images are streamed neither, on arv-viewer or my software. After executing SpinView from FLIR, the camera works, and it can stream in arv-viewer and my software.

I have gone through all the installation process, and the aravis.rules files.

Any suggestion to avoid goring through the SpinView everythime I restart the computer?

Just for additional information: I’m using a Jetson Xavier NX.

This is the output after executing  "arv-camera-test -d all" just after restarting the Jetson (Still no Spinview executed)

Looking for the first available camera
Found 1 USB3Vision device (among 7 USB devices)
[UvDevice::new] Vendor  = FLIR
[UvDevice::new] Product = Blackfly S BFS-U3-16S2M
[UvDevice::new] S/N     = 0138C343
[UvDevice::new] Using control endpoint 2, interface 0
[UvDevice::new] Using data endpoint 1, interface 2
Get genicam
MANUFACTURER_NAME =        'FLIR'
MAX_DEVICE_RESPONSE_TIME = 0x000000c8
DEVICE_CAPABILITY        = 0x0000000000000f0d
SRBM_ADDRESS =             0x0000000000010000
MANIFEST_TABLE_ADDRESS =   0x0000000000040000
U3VCP_CAPABILITY =         0x00000003
MAX_CMD_TRANSFER =         0x00000400
MAX_ACK_TRANSFER =         0x00000400
SIRM_OFFSET =              0x0000000000020000
SI_INFO =                  0x03000000
SI_CONTROL =               0x00000000
SI_REQ_PAYLOAD_SIZE =      0x0000000000100000
SI_REQ_LEADER_SIZE =       0x00000034
SI_REQ_TRAILER_SIZE =      0x00000024
SI_MAX_LEADER_SIZE =       0x00000000
SI_PAYLOAD_SIZE =          0x00000000
SI_PAYLOAD_COUNT =         0x00000000
SI_TRANSFER1_SIZE =        0x00000000
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_TRAILER_SIZE =      0x00000000
MANIFEST_N_ENTRIES =       0x0000000000000001
MANIFEST ENTRY
00000000 00 00 00 01 00 04 01 01 00 00 00 60 00 00 00 00  ...........`....
00000010 1b 0a 01 00 00 00 00 00 17 35 1a d8 06 b4 91 73  .........5.....s
00000020 d8 08 e8 e4 41 73 cd b8 4c eb 67 08 00 00 00 00  ....As..L.g.....
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

genicam address =          0x0000000060000000
genicam size    =          0x0000000000010a1b
zip file =                 public_camxml.xml
[GcFeature::set_attribute] Unknown attribute 'xmlns:xi'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (ImposedAccessMode)
[Genicam::create_element] Unknown tag (Representation)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (ImposedAccessMode)
[Genicam::create_element] Unknown tag (Representation)
[Genicam::create_element] Unknown tag (Visibility)
...
...
...
ArvDevice::set_status] Status changed ('[ArvDevice::write_memory] Timeout')
[ArvDevice::set_integer_feature_value] Node 'DeviceLinkThroughputLimitMode' is not an integer
vendor name           = FLIR
model name            = Blackfly S BFS-U3-16S2M
device id             = 20497219
image width           = 1024
image height          = 1024
horizontal binning    = 1
vertical binning      = 1
payload               = 1048576 bytes
exposure              = 2051 µs
gain                  = 0 dB
uv bandwidth limit     = 380000000 [0..500000000]
SI_INFO            =       0x03000000
SI_REQ_PAYLOAD_SIZE =      0x0000000000100000
SI_REQ_LEADER_SIZE =       0x00000034
SI_REQ_TRAILER_SIZE =      0x00000024
Required alignment =       8
SI_PAYLOAD_SIZE =          0x00010000
SI_PAYLOAD_COUNT =         0x00000010
SI_TRANSFER1_SIZE =        0x00000000
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_LEADER_SIZE =       0x00000038
SI_MAX_TRAILER_SIZE =      0x00000028
USB transfer error: LIBUSB_ERROR_TIMEOUT
USB transfer error: LIBUSB_ERROR_IO
USB transfer error: LIBUSB_ERROR_IO
USB transfer error: LIBUSB_ERROR_IO
USB transfer error: LIBUSB_ERROR_IO
...
...
...

AFTER EXECUTING SPINVIEW:

Looking for the first available camera
Found 1 USB3Vision device (among 7 USB devices)
[UvDevice::new] Vendor  = FLIR
[UvDevice::new] Product = Blackfly S BFS-U3-16S2M
[UvDevice::new] S/N     = 0138C343
[UvDevice::new] Using control endpoint 2, interface 0
[UvDevice::new] Using data endpoint 1, interface 2
Get genicam
MANUFACTURER_NAME =        'FLIR'
MAX_DEVICE_RESPONSE_TIME = 0x000000c8
DEVICE_CAPABILITY        = 0x0000000000000f0d
SRBM_ADDRESS =             0x0000000000010000
MANIFEST_TABLE_ADDRESS =   0x0000000000040000
U3VCP_CAPABILITY =         0x00000003
MAX_CMD_TRANSFER =         0x00000400
MAX_ACK_TRANSFER =         0x00000400
SIRM_OFFSET =              0x0000000000020000
SI_INFO =                  0x03000000
SI_CONTROL =               0x00000001
SI_REQ_PAYLOAD_SIZE =      0x0000000000100000
SI_REQ_LEADER_SIZE =       0x00000034
SI_REQ_TRAILER_SIZE =      0x00000024
SI_MAX_LEADER_SIZE =       0x00000038
SI_PAYLOAD_SIZE =          0x00000000
SI_PAYLOAD_COUNT =         0x00000000
SI_TRANSFER1_SIZE =        0x00100000
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_TRAILER_SIZE =      0x00000028
MANIFEST_N_ENTRIES =       0x0000000000000001
MANIFEST ENTRY
00000000 00 00 00 01 00 04 01 01 00 00 00 60 00 00 00 00  ...........`....
00000010 1b 0a 01 00 00 00 00 00 17 35 1a d8 06 b4 91 73  .........5.....s
00000020 d8 08 e8 e4 41 73 cd b8 4c eb 67 08 00 00 00 00  ....As..L.g.....
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................

genicam address =          0x0000000060000000
genicam size    =          0x0000000000010a1b
zip file =                 public_camxml.xml
[GcFeature::set_attribute] Unknown attribute 'xmlns:xi'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (ImposedAccessMode)
[Genicam::create_element] Unknown tag (Representation)
[Genicam::create_element] Unknown tag (Visibility)
[Genicam::create_element] Unknown tag (ImposedAccessMode)
[Genicam::create_element] Unknown tag (Representation)
[Genicam::create_element] Unknown tag (Visibility)
...
...
...

[ArvDevice::set_status] Status changed ('[ArvDevice::write_memory] Timeout')
[ArvDevice::set_integer_feature_value] Node 'DeviceLinkThroughputLimitMode' is not an integer
vendor name           = FLIR
model name            = Blackfly S BFS-U3-16S2M
device id             = 20497219
image width           = 1024
image height          = 1024
horizontal binning    = 1
vertical binning      = 1
payload               = 1048576 bytes
exposure              = 2051 µs
gain                  = 0 dB
uv bandwidth limit     = 380000000 [0..500000000]
SI_INFO            =       0x03000000
SI_REQ_PAYLOAD_SIZE =      0x0000000000100000
SI_REQ_LEADER_SIZE =       0x00000034
SI_REQ_TRAILER_SIZE =      0x00000024
Required alignment =       8
SI_PAYLOAD_SIZE =          0x00010000
SI_PAYLOAD_COUNT =         0x00000010
SI_TRANSFER1_SIZE =        0x00000000
SI_TRANSFER2_SIZE =        0x00000000
SI_MAX_LEADER_SIZE =       0x00000038
SI_MAX_TRAILER_SIZE =      0x00000028
Frame rate = 262 Hz
Frame rate = 244 Hz
Frame rate = 238 Hz
Frame rate = 239 Hz
Frame rate = 238 Hz
^CFrame rate = 238 Hz
Completed buffers = 1459
Failures          = 0
Underruns         = 0
[UvStream::finalize] n_completed_buffers    = 1459
[UvStream::finalize] n_failures             = 0
[UvStream::finalize] n_underruns            = 0
[Stream::finalize] Flush 49 buffer[s] in input queue
[Stream::finalize] Flush 1 buffer[s] in output queue

EmmanuelP, 2021-03-21:
Hi Roger,

Thanks for the report.

Could you checkout this branch:

https://github.com/AravisProject/aravis/tree/issue/492%2Fcheck-stream-enable

And attach the output of `arv-camera-test -d all:3` before and after spinview please ?

rogerartigaspursals, 2021-03-21:
Hi Emmanuel

Here is the output with this version. Something strange is happening: the output of the file is very large. After executing the arv-camera-test for the first time, the output of the file is very large. After executing spinview, the camera is neither recognized. I need to remove the USB cable to restart the camera and execute the Spinview to recognize the camera.

Without the -d all:3 command line, the output is much smaller 😂
Looking for the first available camera
vendor name           = FLIR
model name            = Blackfly S BFS-U3-16S2M
device id             = 20497219
image width           = 1024
image height          = 1024
horizontal binning    = 1
vertical binning      = 1
payload               = 1048576 bytes
exposure              = 2051 µs
gain                  = 0 dB
uv bandwidth limit    = 380000000 [0..500000000]

 0 frame/s  -       0 MiB/s (before Spinview)
135 frames/s -     142 MiB/s (after Spinview)

With the arv-camera-test -d all:3 command, the output is

Before Spinview

Looking for the first available camera
Found 1 USB3Vision device (among 7 USB devices)
[UvDevice::new] Vendor  = FLIR
[UvDevice::new] Product = Blackfly S BFS-U3-16S2M
[UvDevice::new] S/N     = 0138C343
[UvDevice::new] Using control endpoint 2, interface 0
[UvDevice::new] Using data endpoint 1, interface 2
Get genicam
flags        = 0x4000
command      = read-memory-cmd
size         = 12
id           = 65301
address      = 0x0000000000000004
size         =         64 (0x00000040)
00000000 55 33 56 43 00 40 00 08 0c 00 15 ff 04 00 00 00  ***@***.***
00000010 00 00 00 00 00 00 40 00                          ......@.
status       = success
command      = read-memory-ack
size         = 64
id           = 65301
00000000 55 33 56 43 00 00 01 08 40 00 15 ff 46 4c 49 52  ***@***.***FLIR
00000010 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000020 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000030 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00  ................
00000040 00 00 00 00 00 00 00 00 00 00 00 00              ............
MANUFACTURER_NAME =        'FLIR'
flags        = 0x4000
command      = read-memory-cmd
size         = 12
id           = 65302
address      = 0x00000000000001d8
size         =          8 (0x00000008)
00000000 55 33 56 43 00 40 00 08 0c 00 16 ff d8 01 00 00  ***@***.***
00000010 00 00 00 00 00 00 08 00                          ........
status       = success
command      = read-memory-ack
size         = 8
id           = 65302
00000000 55 33 56 43 00 00 01 08 08 00 16 ff 00 00 01 00  U3VC............
00000010 00 00 00 00                                      ....
flags        = 0x4000
command      = read-memory-cmd
size         = 12
id           = 65303
address      = 0x00000000000001cc
[... 323 lines cut ...]
size         =          4 (0x00000004)
00000000 55 33 56 43 00 40 00 08 0c 00 25 ff 20 00 02 00  ***@***.***%. ...
00000010 00 00 00 00 00 00 04 00                          ........
status       = success
command      = read-memory-ack
size         = 4
id           = 65317
00000000 55 33 56 43 00 00 01 08 04 00 25 ff 00 00 00 00  U3VC......%.....

[http://www.sensofar.com/email/Logo_Met.png]<http://www.sensofar.com/metrology>

[http://www.sensofar.com/email/ic_linkedin.jpg]<https://es.linkedin.com/company/sensofar> [http://www.sensofar.com/email/ic_youtube.jpg] <http://www.youtube.com/user/sensoVID>  [http://www.sensofar.com/email/ic_twitter.jpg] <https://twitter.com/sensofar>

Roger Artigas
President & CTO

T.    +34 937 001 492
M.   +34 639 766 414
[http://www.sensofar.com/email/ic_location.png]<http://www.sensofar.com/metrology/company/location/>   HQ Spain  UTC +1

[cid:bd5ed10f-9c4c-46d0-b711-c42b372f2a96.jpg]

[http://www.sensofar.com/email/banner-met.png]<https://www.sensofar.com/events/tradeshows/>

https://github.com/AravisProject/aravis/tree/issue/492%2Fcheck-stream-enable

EmmanuelP, 2021-03-21:
Thanks for the informations. Unfortunately, I did not spot anything obvious.

The last thing we can try is a capture of the USB data using wireshark. A capture of aravis before and after spinview, as well as during spinview could may be help to find what is missing or wrong in aravis.

rogerartigaspursals, 2021-03-22:
Hi Emmanueal

I have no idea about wireshark. I will spend sometime learning how to install and get data from it. If I find the solution I will post it.

Regards
Roger

[http://www.sensofar.com/email/Logo_Met.png]<http://www.sensofar.com/metrology>

[http://www.sensofar.com/email/ic_linkedin.jpg]<https://es.linkedin.com/company/sensofar> [http://www.sensofar.com/email/ic_youtube.jpg] <http://www.youtube.com/user/sensoVID>  [http://www.sensofar.com/email/ic_twitter.jpg] <https://twitter.com/sensofar>

Roger Artigas
President & CTO

T.    +34 937 001 492
M.   +34 639 766 414
[http://www.sensofar.com/email/ic_location.png]<http://www.sensofar.com/metrology/company/location/>   HQ Spain  UTC +1

[cid:bd5ed10f-9c4c-46d0-b711-c42b372f2a96.jpg]

[http://www.sensofar.com/email/2020-S-wide-Signature.gif]<https://www.sensofar.com/metrology/products/swide?utm_source=Signature&utm_medium=Email&utm_campaign=Web%20I%20Products%20I%20S%20wide>

Reply to: AravisProject/aravis ***@***.***>
Date: Sunday, 21 March 2021 at 19:45

eusebi, 2021-06-21:
Hi, rogerartigaspursals, 
have you found a solution to this problem ?  I am running into the same problem and any hint is appreciated

rogerartigaspursals, 2021-06-22:
Hi Eusebi,

No, I’m sorry. I could not find a solution for it. I decided to change the programming library from Aravis to FLIR-Spinnaker to be able to use the camera.

Regards
Roger

[http://www.sensofar.com/email/Logo_Met.png]<http://www.sensofar.com/metrology>

[http://www.sensofar.com/email/ic_linkedin.jpg]<https://es.linkedin.com/company/sensofar> [http://www.sensofar.com/email/ic_youtube.jpg] <http://www.youtube.com/user/sensoVID>  [http://www.sensofar.com/email/ic_twitter.jpg] <https://twitter.com/sensofar>

Roger Artigas
President & CTO

T.    +34 937 001 492
M.   +34 639 766 414
[http://www.sensofar.com/email/ic_location.png]<http://www.sensofar.com/metrology/company/location/>   HQ Spain  UTC +1

[cid:bd5ed10f-9c4c-46d0-b711-c42b372f2a96.jpg]

[http://www.sensofar.com/email/2020-S-wide-Signature.gif]<https://www.sensofar.com/metrology/products/swide?utm_source=Signature&utm_medium=Email&utm_campaign=Web%20I%20Products%20I%20S%20wide>

Find out more information about our Privacy Policy at www.sensofar.com/privacy<https://www.sensofar.com/privacy/>

Date: Monday, 21 June 2021 at 20:27

boilerbots, 2021-09-22:
I can say that I have been using Aravis with BlackFly USB3 cameras for years and they do start and will work if you configure them properly, however if you program any registers outside of their "legal" values you can lock up the camera or get garbage out. The cameras can be recovered using a script such as this:

```
arv-tool-0.6 | while read -r line; do
    serial=`echo $line | awk '{print $1}'`
    echo "Resetting $serial"
    arv-tool-0.6 -n $serial control DeviceReset=1
done
```

timohueser, 2023-02-24:
I'm running into the same problem on the Jetson Orin developement Kit, using a Flir BlackFly S camera. LibUSB times out during the call to **libusb_bulk_transfer** as soon as I try to acquire images from the camera. This happens both in the arv-viewer and the [02-multiple-acquisition.c](https://github.com/AravisProject/aravis-c-examples/blob/main/02-multiple-acquisition.c) from the aravis-c-examples. Interestingly I sometimes get  few frames to show up before it times out, but this happens irregularly and most of the time it's just a single frame.
The issue completely goes away as soon as the camera is initialized using the Spinnaker SDK. All that is needed is a call to the init() function of the camera thorugh the Spinnaker SDK and Aravis starts streaming the images as expected.

This issue does not occur when I use the exact same camera on my Ubuntu 22.04 desktop PC, so I suspect the problem is Jetson/ARM related. I tried examining the USB traffic between the camera and the host using wireshark, and the traffic looks identical on my desktop and the Jetson Orin. 
I also tried upgrading libUSB to the newest release (1.0.26) without any luck. 

One last thing that might be worth noting is that even after the image streaming gets stuck on the timeout all camera settings are still accessible and get applied to the camera correctly. I tried a bunch of combinations of framerates, pixel formats and resolutions, all with pretty much the same result. Very small resolutions (16x16 in Mono8) did stop the libusb_bulk_transfer function from timing out, but there were still no images transferred to the buffers.

I would really appreciate any help or suggestions!

EmmanuelP, 2023-02-24:
Hi Timo,

Like I said in comment https://github.com/AravisProject/aravis/issues/492#issuecomment-803639754 , some wireshark captures would help to understand the issue.

timohueser, 2023-02-24:
Hi Emmanuel,

I will try to do some captures this afternoon! Since I'm new to using wireshark, what format (e.g. csv or raw capture etc.) is most useful to you? And do I need to configure it in any special way to give you the data you need?

Thanks for your help!

timohueser, 2023-02-24:
Hi again,

Here is a collection of Wireshark captures:

- **02-multiple-acquisition**: Running the example from the aravis-c-examples, both with and without having the cameras initialized using the Spinnaker SDK (success/fail respectively).
- **arv_camera_test**: Running the arc-camera-test utility, both with and without having the cameras initialized using the Spinnaker SDK (success/fail respectively). I also included the logs produced at debug level 3, like you suggested in an earlier comment.
- **FLIR_Spinview_Init**: Capture during launch of the SpinView software and selecting the camera in the camera list. This is enough to initialize the camera and make it work with aravis.
- **FLIR_Minimal_Init_Deinit**: Capture of the Spinnaker SDK being used to init and deinit the camera. No settings are changed on the cameras and no Acquisition is started. This also is sufficient to make the camera work with aravis afterwards.
- **FLIR_Acquisition_Example**: Capture of the unedited Acquisition sample from the Spinnaker SDK as a refernce.

I hope this is helpful, please let me know if you need anything else.
[Wireshark_Captures.zip](https://github.com/AravisProject/aravis/files/10824609/Wireshark_Captures.zip)

timohueser, 2023-02-26:
Some additional observations I made this weekend:

- Using the asynchronous mode (e.g. with --usb-mode=async in arv-camera-test) makes it to work for 1000-10000 frames, after which the framerate abruptly drops to 0. Curiosuly there are no libusb errors reportet in the async mode, it just silently stops working. After initializing the camera using the Spinnaker SDK it works indefinitely, both in sync and async mode.
- It seems like the only write command that the Spinnaker API issues over the U3V protocoll to the camera during the init() call is setting the EI_Control register. As far as I can tell the value it sets is the default value that camera is initialized with anyways, so this should have no effect. The only other non-parameter-related thing I could spot are some URB_BULK transfers to interface 3 of the camera (this interface is not used with aravis as far as I can tell). I have no idea what these do though and if they are related to the problem at all.
- To make sure it the Spinnaker init() call changes the configuration on the camera, and not some USB related settings on the jetson itself I tried powering the camera externally, initializing it with the Spinnaker SDK on a second computer and then plugging it into the Jetson. And indeed this has the same effect as initializing it on the Jetson directly, so whatever the Spinnaker SDK does it has to be related to the camera itself.

If anyone has any suggestions on what else I could try to troubleshoot this please let me know!

EmmanuelP, 2023-03-01:
Hi Timo,

Thanks for the capture. In the FLIR_Spinview_Init.pcapng file, I have noticed this command:

```
No.	Time	Source	Destination	Protocol	Length	Info
1313	39.785737	host	2.5.2	U3V	88	> WRITEMEM_CMD [EI_Control]: 4 bytes
1314	39.786836	2.5.2	host	USB	64	URB_BULK out
1315	39.786901	host	2.5.2	USB	64	URB_BULK in
1316	39.787015	2.5.2	host	U3V	80	< WRITEMEM_ACK [EI_Control]
```

```
Frame 1313: 88 bytes on wire (704 bits), 88 bytes captured (704 bits) on interface usbmon0, id 0
USB URB
USB 3 Vision
    CCD: WRITEMEM_CMD
    SCD: WRITEMEM_CMD
        Address: 0x0000000000028000 [EI_Control]
        Event Interface Control: 1
```

Which enables device events. Aravis does not implement event support yet, and does nothing about them. I'm not sure how this command could make aravis work though.

timohueser, 2023-03-01:
Hi Emmanuel,

Thanks for looking through the captures! I tried adding the EI_Control command to aravis in the _bootstrap method and confirmed that it issues the same write command as the Spinnaker SDK using wireshark. Unfortunately it has no effect on the behaviour of the camera. Do you thin the issue could still somehow be related to device events? 

Another thing I noticed is that I also get LIBUSB_ERROR_IO errors when I change the length parameter in the libusb_bulk_transfer call that is responsible for reading the frame data from the camera (as far I can tell). This is just a wild guess, but could it be that the camera sends some packets with a length different from the expected one? 

I also contacted the FLIR support about the issue in the hopes of getting some more info about what happens during the Spinnaker init() call. I'll update this issue if I get any more info.

boilerbots, 2023-03-03:
Just a wild guess but try disable USB3 low power mode on ports and hubs that you connect the FLIR cameras to.

eusebi, 2023-05-10:
@timohueser Have you found the solution to this spinview magic initialization ? This detail still plagues our code...

joshuashepherd29-art, 2025-09-05:
[CFR_FLIR-FLIR-C8.pdf](https://github.com/user-attachments/files/22181486/CFR_FLIR-FLIR-C8.pdf)