AravisProject/aravis #494: Fetched frames are corrupted (don't look as they should)

Shorlaks, 2021-04-05:
**Describe the bug**
I'm using Aravis to receive frames from my camera and save them locally to a binary file.
The problem is that the frame's I'm getting look like this

[image]

When they suppose to look like this (using eBus).

[image]

This is the python code I'm using to do so: (my camera is 265Hz, so I'm basically recording 10 seconds)
```
import sys
import gi
import time
gi.require_version ('Aravis', '0.8')
from gi.repository import Aravis

        
try:
    if len(sys.argv) > 1:
            camera = Aravis.Camera.new (sys.argv[1])
    else:
            camera = Aravis.Camera.new (None)
except TypeError:
	print ("No camera found")
	exit ()
	
camera.gv_set_packet_size(6130)
payload = camera.get_payload ()
[x,y,width,height] = camera.get_region ()

print ("Camera vendor : %s" %(camera.get_vendor_name ()))
print ("Camera model  : %s" %(camera.get_model_name ()))
print ("ROI           : %dx%d at %d,%d" %(width, height, x, y))
print ("Payload       : %d" %(payload))
print ("Pixel format  : %s" %(camera.get_pixel_format_as_string ()))

stream = camera.create_stream (None, None)
for i in range(10):
    stream.push_buffer (Aravis.Buffer.new_allocate (payload))
camera.start_acquisition()
file = open("rec.Raw", "wb")
t = time.time()
for i in range(0,2650):
    image = stream.pop_buffer()
    data = image.get_data()
    if image:
        file.write(data)
        stream.push_buffer (image)

print(time.time() - t)
camera.stop_acquisition ()
file.close()
```

**Platform description:**
 - Aravis version: 8.0.6
 - OS: Raspbian latest version
 - Hardware: Raspberrypi4 8Gb

Any ideas on what's going on or how to fix it?
Thanks, Shorlaks

EmmanuelP, 2021-04-06:
Hi,

Are both resulting images built from the raw buffer of aravis and ebus ?

Aravis does not transform what is in the buffer data, so as long as the buffer status is SUCCESS, the data should be exactly what was sent by the camera.

Shorlaks, 2021-04-06:
Hi @EmmanuelP 
Yes, both images from the raw buffers are 369600 bytes, the pixel values seems to be identical (the raw data looks the same) and I save and view it the same way.

Although Aravis doesn't alter the data, it seems like every row is shifted 1 pixel to the right.
The camera's resolution is 480*385.
This behavior maybe can be explained if instead of 480*385 bytes, Aravis gets 480*385 bytes but with 479*385 resolution and when I reshape it I get the shifted rows.
What do you think?

EmmanuelP, 2021-04-06:
Did you check the actual width and height settings, and the buffer height and width properties ?

EmmanuelP, 2021-04-06:
> 480X385 at 0,0

Is that the configured size, or the buffer one ?

On some devices, there is a set of feature for the size setting, and another one for the actual size. In your software, do you actually set the image size ?

Shorlaks, 2021-04-06:
It's the size I get from `camera.get_region()`. The device is configured to 480X385, how do I check the buffers size?
In my software I do not configure the size since I want the default size.

After checking the buffer status with `image.get_status()` I get alot of `ARV_BUFFER_STATUS_TIMEOUT`, can it be related?

Shorlaks, 2021-04-06:
@EmmanuelP I've ran the same script on an intel's NUC and everything works fine, Aravis works 100%.
It has something to do with the raspberrypi

EmmanuelP, 2021-04-12:
I'm closing this issue. Please reopen if you think there is something wrong in aravis.

closed 2021-04-12 by EmmanuelP