AravisProject/aravis #497: problems with the viewer

enarasanz, 2021-04-20:
Hello, 
I have installed the Aravis Project on a jetson Nano. I am new with this and I am having problems. I connected a Photonfocus camera through the ethernet cable. When I run arv-viewer I can see that it detects the camera and I can start the video acquisition but when I do so, it shows me a black window and I tried saving a shot into a folder and the next message appears: 

(arv-viewer:12527): Aravis-CRITICAL **:13:48:07.381: snapshot_cb: assertion 'ARV_IS_BUFFER (viewer->last_buffer)' failed

Trying to solve it I came across a command to check the camera, ./arv-camera-test and run it. When I do it the next appears:

Looking for the first available camera
vendor name           = Photonfocus AG
model name            = MV1-D2048x1088-C102-96-G2-10
device id             = 075700031493
image width           = 2048
image height          = 1088
horizontal binning    = 1
vertical binning      = 1
payload               = 4456448 bytes
exposure              = 349000 µs
gain                  = 5 dB
gv n_stream channels  = 1
gv current channel    = 0
gv packet delay       = 0 ns
gv packet size        = 8164 bytes
Frame rate = 0 Hz
Frame rate = 0 Hz

EmmanuelP, 2021-04-30:
Please attach the output of `arv-camera-test -d all`.

enarasanz, 2021-04-30:
[logfile.txt](https://github.com/AravisProject/aravis/files/6403908/logfile.txt)
Here is the file.

EmmanuelP, 2021-04-30:
> gv packet size        = 8164 bytes

This is probably the issue. Try to reduce the packet size using; `arv-camera-test -i 1500`.

Which version of Aravis are you using ?

In the latest version, the packet size should be adjusted automatically if the default one does not work.

enarasanz, 2021-04-30:
Thank you, that solved the problem. I can now watch the camera input and save snapshots.
I am using aravis 6.4, I know there were more recent ones.

closed 2021-04-30 by enarasanz