AravisProject/aravis #500: Failed to bootstrap USB device 'Toshiba-Teli:BU505MG:...'

stevenlovegrove, 2021-04-23:
**Describe the bug**
arv-tool-0.8 is able to enumerate my U3V device, but fails when attempting to list features or genicam xml etc. 

**To Reproduce**
[Unplug and replug camera, wait for flashing green ready light]

(base) computer:~/code/aravis/build/src ./arv-tool-0.8 --debug=all features
[21:02:12.615] ❷ interface> Found 1 USB3Vision device (among 13 USB devices)
[21:02:12.615] ❷ interface> [GvDiscoverSocket::new] Add interface 127.0.0.1
[21:02:12.616] ❷ interface> [GvDiscoverSocket::new] Add interface 192.168.0.123
[21:02:12.616] ❶ interface> [ArvGVInterface::send_discover_packet] Error: Error sending message: Can't assign requested address
Toshiba-Teli-0300102 (USB3)
[21:02:13.620] ❷ device> [UvDevice::new] Vendor  = Toshiba-Teli
[21:02:13.620] ❷ device> [UvDevice::new] Product = BU505MG
[21:02:13.620] ❷ device> [UvDevice::new] S/N     = [redacted]
[21:02:13.643] ❷ device> [UvDevice::new] Using control endpoint 1, interface 0
[21:02:13.643] ❷ device> [UvDevice::new] Using data endpoint 4, interface 2
[21:02:13.645] ❷ device> Get genicam
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[21:02:13.645] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND

[... more of the same about 1000 times ...]

[21:02:13.814] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:13.814] ❶ device> [UvDevice::_bootstrap] Error during memory read
[21:02:13.815] ❷ interface> Found 1 USB3Vision device (among 13 USB devices)
[21:02:13.815] ❷ device> [UvDevice::new] Vendor  = Toshiba-Teli
[21:02:13.815] ❷ device> [UvDevice::new] Product = BU505MG
[21:02:13.815] ❷ device> [UvDevice::new] S/N     = 0300102
[21:02:13.838] ❷ device> [UvDevice::new] Using control endpoint 1, interface 0
[21:02:13.838] ❷ device> [UvDevice::new] Using data endpoint 4, interface 2
[21:02:13.839] ❷ device> Get genicam
[21:02:13.874] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:13.908] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:13.944] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:13.977] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:14.011] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:14.011] ❶ device> [UvDevice::_bootstrap] Error during memory read
Failed to open device 'Toshiba-Teli-0300102': Failed to bootstrap USB device 'Toshiba-Teli:BU505MG:0300102'

**[Running again without unplugging, we reach termination much sooner (as below)]**

_________________________________________________

(base) Stevens-iMac-Pro:~/code/aravis/build/src ./arv-tool-0.8 --debug=all features
[21:02:15.468] ❷ interface> Found 1 USB3Vision device (among 13 USB devices)
[21:02:15.469] ❷ interface> [GvDiscoverSocket::new] Add interface 127.0.0.1
[21:02:15.469] ❷ interface> [GvDiscoverSocket::new] Add interface 192.168.0.123
[21:02:15.469] ❶ interface> [ArvGVInterface::send_discover_packet] Error: Error sending message: Can't assign requested address
Toshiba-Teli-0300102 (USB3)
[21:02:16.472] ❷ device> [UvDevice::new] Vendor  = Toshiba-Teli
[21:02:16.472] ❷ device> [UvDevice::new] Product = BU505MG
[21:02:16.472] ❷ device> [UvDevice::new] S/N     = 0300102
[21:02:16.495] ❷ device> [UvDevice::new] Using control endpoint 1, interface 0
[21:02:16.495] ❷ device> [UvDevice::new] Using data endpoint 4, interface 2
[21:02:16.496] ❷ device> Get genicam
[21:02:16.530] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:16.565] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:16.600] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:16.635] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:16.671] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:16.671] ❶ device> [UvDevice::_bootstrap] Error during memory read
[21:02:16.672] ❷ interface> Found 1 USB3Vision device (among 13 USB devices)
[21:02:16.672] ❷ device> [UvDevice::new] Vendor  = Toshiba-Teli
[21:02:16.672] ❷ device> [UvDevice::new] Product = BU505MG
[21:02:16.672] ❷ device> [UvDevice::new] S/N     = 0300102
[21:02:16.695] ❷ device> [UvDevice::new] Using control endpoint 1, interface 0
[21:02:16.695] ❷ device> [UvDevice::new] Using data endpoint 4, interface 2
[21:02:16.696] ❷ device> Get genicam
[21:02:16.729] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:16.765] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:16.800] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:16.836] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:16.872] ❶ device> [UvDevice::read_memory] Command sending error: LIBUSB_ERROR_TIMEOUT
[21:02:16.872] ❶ device> [UvDevice::_bootstrap] Error during memory read
Failed to open device 'Toshiba-Teli-0300102': Failed to bootstrap USB device 'Toshiba-Teli:BU505MG:0300102'

**Camera description:**
 - Manufacturer: Toshiba-Teli
 - Model: BU505MG
 - Interface: USB3

**Platform description:**
 - Aravis version: current master (ff6fb2032b). Before this commit it failed sooner without any command line arguments.
 - OS: macOS 11.2.3
 - Hardware: Intel x86_64

Do you have any tips on how to debug this or where the issue might light?

Thanks! Great project!

EmmanuelP, 2021-04-23:
Hi Steven,

Did you install an USB3Vision driver on your machine ? After a quick search, this issue may happen if a kernel driver has already claim the device.

stevenlovegrove, 2021-04-23:
Thank you for the suggestion. No, I don't think so - Toshiba doesn't provide one for OSX which is why I'm interested in Aravis, and I don't think I've used pleora or similar here.

To make sure I listed loaded modules (which I think is the way a driver would show up?):

(base) Stevens-iMac-Pro:~ kextstat
Executing: /usr/bin/kmutil showloaded
No variant specified, falling back to release
Index Refs Address            Size       Wired      Name (Version) UUID <Linked Against>
    1  141 0                  0          0          com.apple.kpi.bsd (20.3.0) C86236B2-4976-3542-80CA-74A6B8B4BA03 <>
    2   11 0                  0          0          com.apple.kpi.dsep (20.3.0) C86236B2-4976-3542-80CA-74A6B8B4BA03 <>
    3  166 0                  0          0          com.apple.kpi.iokit (20.3.0) C86236B2-4976-3542-80CA-74A6B8B4BA03 <>
    4    0 0                  0          0          com.apple.kpi.kasan (20.3.0) C86236B2-4976-3542-80CA-74A6B8B4BA03 <>
    5  172 0                  0          0          com.apple.kpi.libkern (20.3.0) C86236B2-4976-3542-80CA-74A6B8B4BA03 <>
    6  152 0                  0          0          com.apple.kpi.mach (20.3.0) C86236B2-4976-3542-80CA-74A6B8B4BA03 <>
    7   90 0                  0          0          com.apple.kpi.private (20.3.0) C86236B2-4976-3542-80CA-74A6B8B4BA03 <>
    8  104 0                  0          0          com.apple.kpi.unsupported (20.3.0) C86236B2-4976-3542-80CA-74A6B8B4BA03 <>
    9    2 0xffffff80031e8000 0xe000     0xe000     com.apple.kec.Libm (1) 75E2671F-F050-3765-96B8-BD03232E1A62 <5>
   10   13 0xffffff8003406000 0x91000    0x91000    com.apple.kec.corecrypto (11.1) E009770B-BAFB-3BF9-B176-D6B4CC96DF5F <8 7 6 5 3 1>
   11    0 0xffffff80034a1000 0x8000     0x8000     com.apple.kec.pthread (1) 1E922D2A-9975-3AFB-A009-73EC3EF87A86 <8 7 6 5 3 1>
   12    1 0xffffff8003563000 0x2000     0x2000     com.apple.driver.watchdog (1) 2C1307C5-FD71-3802-8D68-A67DB24BCB4A <8 7 6 5 3 1>
   13   35 0xffffff80026d8000 0x2000     0x2000     com.apple.iokit.IOACPIFamily (1.4) B41AA072-8BB9-3493-9796-6FF849CDABB8 <8 7 5 3>
   14   44 0xffffff8002c1f000 0x28000    0x28000    com.apple.iokit.IOPCIFamily (2.9) BF2C5E86-1E8F-3FD4-9874-7738178FA73B <8 7 6 5 3>
   15   11 0xffffff8001895000 0x19000    0x19000    com.apple.driver.AppleSMC (3.1.9) 1E48B8EB-7BF3-3366-8520-2C4B31A7D3F8 <14 13 12 8 7 6 5 3 1>
   16    2 0xffffff8001292000 0x77000    0x77000    com.apple.driver.AppleACPIPlatform (6.1) 2F2AA264-6BD5-3E4F-829C-8952B3ABB2D5 <15 14 13 8 7 6 5 3 1>
   17    5 0xffffff8002d37000 0x2000     0x2000     com.apple.driver.IOSlaveProcessor (1) FC8BCD16-0792-35FD-B23E-A0720BC40E1A <5 3>
   18    3 0xffffff800187b000 0x10000    0x10000    com.apple.driver.AppleSEPManager (1.0.1) 4FC2AB67-A77F-3028-BFF6-7BE23EEE6773 <17 14 10 8 7 6 5 3 1>
   19    1 0xffffff800158c000 0x4000     0x4000     com.apple.driver.AppleBusPowerController (1.0) 499DB0FD-1FE4-31FC-B0E0-ECE24BF29417 <15 13 7 6 5 3>
   20   11 0xffffff80019e2000 0x4000     0x4000     com.apple.driver.usb.AppleUSBCommon (1.0) 62FE0E66-DCCA-3443-B01E-0F0FAB0A820D <6 5 3 1>
   21    3 0xffffff800304d000 0x1000     0x1000     com.apple.driver.AppleUSBHostMergeProperties (1.2) 90FEE882-3237-3822-91F1-6AD62C8E46AD <5 3 1>
   22   21 0xffffff8002f52000 0x8d000    0x8d000    com.apple.iokit.IOUSBHostFamily (1.2) A84D73D5-4F62-32E2-A1DB-9F021795AE81 <21 20 19 15 13 8 7 6 5 3 1>
   23    3 0xffffff80031d2000 0xf000     0xf000     com.apple.driver.KernelRelayHost (1) 6E0EDF05-F095-3009-AC2B-BE4CEFFF6583 <22 6 5 3>
   24    1 0xffffff8001600000 0x50000    0x50000    com.apple.driver.AppleCredentialManager (1.0) 56AFE766-6003-3409-B493-402505841C1C <23 18 17 10 8 7 6 5 3 1>
   25    3 0xffffff800166d000 0x6000     0x6000     com.apple.driver.AppleEffaceableStorage (1.0) CE527E76-F27F-33BD-933A-ED837F559A6D <8 6 5 3 1>
   26    0 0xffffff8001677000 0x6000     0x6000     com.apple.driver.AppleFDEKeyStore (28.30) FDFE34C9-00DA-3C05-AC10-14CF0EB24902 <25 10 8 7 6 5 3 1>
   27    2 0xffffff8001b7e000 0x7000     0x7000     com.apple.kext.CoreTrust (1) 36944A7C-F451-35A4-82A1-3B6BF8CBAC71 <10 5>
   28    9 0xffffff8001801000 0x15000    0x15000    com.apple.driver.AppleMobileFileIntegrity (1.0.5) 2A454117-CDAA-301F-B609-BA396742C91A <27 10 8 7 6 5 3 2 1>
   29    4 0xffffff8002c5a000 0x14000    0x14000    com.apple.iokit.IOSCSIArchitectureModelFamily (436.40.6) E5CB0567-FBD1-39AE-8847-C559B51918B1 <7 6 5 3 1>
   30    9 0xffffff8002d3f000 0x12000    0x12000    com.apple.iokit.IOStorageFamily (2.1) B5300908-BF34-3D47-8776-FB154A6DEE4C <8 7 6 5 3 1>
   31    2 0xffffff8002c7f000 0xe000     0xe000     com.apple.iokit.IOSCSIBlockCommandsDevice (436.40.6) 17662AB7-6B15-3342-9F23-AB161202B652 <30 29 7 6 5 3 1>
   32    2 0xffffff80031ae000 0x1a000    0x1a000    com.apple.iokit.IOUSBMassStorageDriver (184.40.6) AB4AC255-99A6-38E2-B5FE-22B97EEC79AA <30 29 22 14 7 6 5 3 1>
   33    2 0xffffff80018ef000 0x8000     0x8000     com.apple.driver.AppleUSBTDM (511.60.2) E381525B-81E2-3E87-B197-6355B228751B <32 31 30 29 28 22 8 7 6 5 3 1>
   34    0 0xffffff800175c000 0x80000    0x80000    com.apple.driver.AppleKeyStore (2) EAF92490-F231-30E8-A019-2C01F28E4315 <33 28 27 24 23 18 17 10 8 7 6 5 3 1>
   35    0 0xffffff80018b6000 0x7000     0x7000     com.apple.driver.AppleSSE (1.0) 59F98016-B7A6-3538-8C5E-ABF400FA3EE6 <23 18 17 8 6 5 3 1>
[... cut ...]
  202    0 0xffffff7fa134c000 0x4d000    0x4d000    com.apple.filesystems.afpfs (11.3) 353E923F-991E-3517-9F1F-9E4B385134E0 <201 96 8 7 6 5 3 1>
  273    0 0xffffff7fa0b9f000 0x19000    0x19000    com.apple.driver.usb.AppleUSBUserHCI (1) 6FE7E24C-5793-3AFF-AAFF-9DB4F82066BA <22 20 6 5 3 1>
________________________________________

There was previously one named something like `org.virtualbox.kext.VBoxUSB` in there which I unloaded with

`sudo kextunload -b org.virtualbox.kext.VBoxUSB`

before then unplugging and replugging the camera but got the same behavior with Aravis.

stevenlovegrove, 2021-04-23:
To be sure, I tried inserting the snippet you linked to:

```
static void
arv_uv_device_constructed (GObject *object)
{
	ArvUvDevice *uv_device = ARV_UV_DEVICE (object);
	ArvUvDevicePrivate *priv = arv_uv_device_get_instance_private (uv_device);
	GError *error = NULL;

	arv_debug_device ("[UvDevice::new] Vendor  = %s", priv->vendor);
	arv_debug_device ("[UvDevice::new] Product = %s", priv->product);
	arv_debug_device ("[UvDevice::new] S/N     = %s", priv->serial_number);

	libusb_init (&priv->usb);
	priv->packet_id = 65300; /* Start near the end of the circular counter */
	priv->timeout_ms = 32;

	if (!_open_usb_device (uv_device, &error)) {
		arv_device_take_init_error (ARV_DEVICE (uv_device), error);
		error = NULL;
	}

	arv_debug_device("[UvDevice::new] Using control endpoint %d, interface %d",
			 priv->control_endpoint, priv->control_interface);
	arv_debug_device("[UvDevice::new] Using data endpoint %d, interface %d",
			 priv->data_endpoint, priv->data_interface);

	arv_debug_device("HERE0");
	if (priv->usb_device != NULL) {
		arv_debug_device("HERE1");

		if(libusb_kernel_driver_active(priv->usb_device, 0) == 1)
		{
			arv_debug_device("Kernel Driver Active");
			if(libusb_detach_kernel_driver(priv->usb_device, 0) == 0)
				arv_debug_device("\nKernel Driver Detached!");
			else
			{
				arv_debug_device("Couldn't detach kernel driver!\n");
				// libusb_free_device_list(devs, 1);
				libusb_close(priv->usb_device);
				return;
			}
		}
	}

	if (priv->usb_device == NULL ||
	    libusb_claim_interface (priv->usb_device, priv->control_interface) < 0 ||
	    libusb_claim_interface (priv->usb_device, priv->data_interface) < 0) {
		arv_device_take_init_error (ARV_DEVICE (uv_device), g_error_new (ARV_DEVICE_ERROR, ARV_DEVICE_ERROR_PROTOCOL_ERROR,
										 "Failed to claim USB interface to '%s:%s:%s'",
										 priv->vendor, priv->product, priv->serial_number));
		return;
	}

	if ( !_bootstrap (uv_device)){
		arv_device_take_init_error (ARV_DEVICE (uv_device), g_error_new (ARV_DEVICE_ERROR, ARV_DEVICE_ERROR_PROTOCOL_ERROR,
										 "Failed to bootstrap USB device '%s:%s:%s'",
										 priv->vendor, priv->product, priv->serial_number));
		return;
	}

...
```

but it didn't make any difference and outputs

```
(base) Stevens-iMac-Pro:~/code/aravis/build ./src/arv-tool-0.8 --debug=all features
[12:07:19.269] ❷ interface> Found 1 USB3Vision device (among 13 USB devices)
[12:07:19.270] ❷ interface> [GvDiscoverSocket::new] Add interface 127.0.0.1
[12:07:19.270] ❷ interface> [GvDiscoverSocket::new] Add interface 192.168.0.123
[12:07:19.270] ❶ interface> [ArvGVInterface::send_discover_packet] Error: Error sending message: Can't assign requested address
Toshiba-Teli-[Redacted] (USB3)
[12:07:20.275] ❷ device> [UvDevice::new] Vendor  = Toshiba-Teli
[12:07:20.275] ❷ device> [UvDevice::new] Product = BU505MG
[12:07:20.275] ❷ device> [UvDevice::new] S/N     =[Redacted]
[12:07:20.298] ❷ device> [UvDevice::new] Using control endpoint 1, interface 0
[12:07:20.298] ❷ device> [UvDevice::new] Using data endpoint 4, interface 2
[12:07:20.298] ❷ device> HERE0
[12:07:20.298] ❷ device> HERE1
[12:07:20.300] ❷ device> Get genicam
[12:07:20.300] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[12:07:20.300] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[12:07:20.300] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[12:07:20.300] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[12:07:20.300] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
[12:07:20.300] ❶ device> [UvDevice::read_memory] Ack reception error: LIBUSB_ERROR_NOT_FOUND
```

One deviation from your MacOS instructions is that I used macports to install the dependencies. I guess I could try Brew as you suggest incase there is a difference in libusb version or something.

EmmanuelP, 2026-01-28:
Hi @27Cobalter 

> I would like to submit a Pull Request for this fix. Would that be acceptable?

Yes, please.

closed 2026-01-31 by EmmanuelP