AravisProject/aravis #503: arv_stream_pop_buffer returns NULL

chris80801, 2021-04-28:
**Describe the bug**
An acquisition thread is using arv_stream_pop_buffer to read data sent by a camera through an external framegrabber. I can see with tcpdump that some data is sent to the right interface and port but still arv_stream_pop_buffer returns null.
I suspect that there might be some interference with another thread that uses a different socket to send serial commands to the camera. Before sending a command, each thread writes into the registers "GevMCPHostPortReg" and "GevSCPDirectionReg" the port where a response shall be sent. Still, during the acquisition, the serial command thread does not send any command so in principle the port in these registers should always be the same.
Do you have any idea of what's going on ?

**To Reproduce**
Steps to reproduce the behavior (in a nutshell):
std::string grabberIP = ...;
GInetAddress* host_address = g_inet_address_new_from_string(grabberIP);
GInetAddress* camera_address = g_inet_address_new_from_string(grabberIP);
auto p_device = arv_gv_device_new(host_address, camera_address, NULL);
p_stream = arv_device_create_stream(p_device, NULL, NULL, NULL);
ArvBuffer* buffer = arv_stream_pop_buffer(p_stream);

**Expected behavior**.
arv_stream_pop_buffer should return a non empty buffer.

**Camera description:**
 - Manufacturer: Princeton
 - Model: 1280Scicam
 - Interface [e.g. USB3]: Serial

**Grabber description:**
 - Manufacturer: Pleora technologies
 - Model: iPORT CL-Ten External Frame grabber
 - Interface [e.g. USB3]: GigE Vision

**Platform description:**
 - Aravis version: 0.8
 - OS:Cent OS 7.4.17089
 - Hardware [e.g. x86_64]: x86_64

**Additional context**
Add any other context about the problem here.

EmmanuelP, 2021-04-30:
Hi.

Thanks for the report.

Did you check you are not running a firewall ?

Please run your code with ARV_DEBUG=all and attach the output here.

chris80801, 2021-04-30:
Hi,

I run my program as a normal user and as root as well as I could see in aravis code that different sockets were used according to whether the program has root permission or not.
I put the two outputs in attachment. Some errors are reported about missing variables but this did not stop the program.
When the acquisition start, nothing happens on the application side even if I can see with tcpdump that packets are received on the port of the socket created by aravis.
I am not aware of any firewall running on the pc and there is a point to point connection between the pc and the grabber.

Cheers
Christophe

[GvDevice::new] Interface address = 10.0.0.1
[GvDevice::new] Device address = 10.0.0.5
[GvDevice::load_genicam] xml url = 'Local:iport_cl_ten_full_pt10_cf0ip01_256xg_1_0_0_9E1EB142.zip;3fff8000;9c7f' at 0x200
[GvDevice::load_genicam] Xml address = 0x3fff8000 - size = 0x9c7f - iport_cl_ten_full_pt10_cf0ip01_256xg_1_0_0_9E1EB142.zip
[GvDevice::load_genicam] Zipped xml data
[GcFeature::set_attribute] Unknown attribute 'ToolTip'
[GcFeature::set_attribute] Unknown attribute 'StandardNameSpace'
[GcFeature::set_attribute] Unknown attribute 'ProductGuid'
[GcFeature::set_attribute] Unknown attribute 'VersionGuid'
[GcFeature::set_attribute] Unknown attribute 'xmlns'
[GcFeature::set_attribute] Unknown attribute 'xmlns:xsi'
[GcFeature::set_attribute] Unknown attribute 'xsi:schemaLocation'
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[Genicam::create_element] Unknown tag (Streamable)
[... 276 lines cut ...]
After Exit
-------------
<      0;       0
min     ;     n/a
max     ;     n/a
last max
at:     ;       0
Counter =        0
[GvStream::finalize] n_completed_buffers    = 0
[GvStream::finalize] n_failures             = 0
[GvStream::finalize] n_timeouts             = 0
[GvStream::finalize] n_aborteds             = 0
[GvStream::finalize] n_underruns            = 0
[GvStream::finalize] n_missing_frames       = 0
[GvStream::finalize] n_size_mismatch_errors = 0
[GvStream::finalize] n_received_packets     = 0
[GvStream::finalize] n_missing_packets      = 0
[GvStream::finalize] n_error_packets        = 0
[GvStream::finalize] n_ignored_packets      = 0
[GvStream::finalize] n_resend_requests      = 0
[GvStream::finalize] n_resent_packets       = 0
[GvStream::finalize] n_resend_ratio_reached = 0
[GvStream::finalize] n_duplicated_packets   = 0
[Stream::finalize] Flush 5 buffer[s] in input queue
[Stream::finalize] Flush 0 buffer[s] in output queue

chris80801, 2021-04-30:
Hello,

When looking at tcpdump, I actually saw that the ip address of the receiver was wrong, i.e. 1.0.0.10 instead of 10.0.0.1. It looks like the bytes of the ip address were in the wrong oder.
In line 1202 and 1203 of arvgvstream.c, the address is stored in the feature GevSCDA so that the grabber knows where to send the data.
I do not know whether the problem lies here or anywhere else in my application where a wrong value would be stored in this feature.
Anyway, I added these lines of code in my application to store in GevSCDA  my host ip address after a stream is created and before starting the application:
//    guint32 my_ip_in_bytes;
    std::stringstream ss(hostIp);
    std::vector<std::string> ip_words;
    std::string word;
    // convert host ip to numerical address
    while(std::getline(ss, word, '.'))
    {
        ip_words.push_back(word);
    }
    char* tmp= (char*)&my_ip_in_bytes_;
    for (unsigned int i = 0; i < ip_words.size(); i++)
    {
        *tmp++ = atoi(ip_words[i].c_str());
    }
    // convert  ip from host byte order to network byte order
    my_ip_in_bytes_ = htonl(my_ip_in_bytes_);
    write_register("GevSCDAReg", REGISTER_ADDRESS.at(SCDAREG), my_ip_in_bytes_);

And this time frames were received. Can you think about any case where the ip address could be stored in the wrong order in GevSCDA in arvgvstream.c ?
Moreover, if I execute my application as a normal user, I can see that _loop is called whereas I I launch it as root, _ring_buffer_loop is executed.
What are the differences between  the two ? Which one is it better to used ?

Cheers
Christophe

EmmanuelP, 2021-04-30:
Please attach the genicam data, retrieved using `arv-tool-0.8 genicam`.

> Moreover, if I execute my application as a normal user, I can see that _loop is called whereas I I launch it as root, _ring_buffer_loop is executed.

When you are using executing your application as root, Aravis is able to use a packet socket and the ring buffer capability, for an increased performance. You can disable the use of packet socket using `arv_gv_device_set_stream_options()`.

chris80801, 2021-04-30:
I have the following message:
arv-tool-0.8: symbol lookup error: arv-tool-0.8: undefined symbol: arv_device_set_range_check_policy

Anyway I managed to get the genicam xml file of the external grabber that I put in attachment.

EmmanuelP, 2021-04-30:
> I have the following message:
> arv-tool-0.8: symbol lookup error: arv-tool-0.8: undefined symbol: arv_device_set_range_check_policy

There is something wrong in your installation, you are using arv-tool-0.8 dynamically linked with a wrng version of Aravis library.

> Anyway I managed to get the genicam xml file of the external grabber that I put in attachment.

There is nothing attached here. Please use the Github Web interface, interacting with github by email generates badly formatted comments.

EmmanuelP, 2021-04-30:
After digging in Aravis issues, this issue is the same as issue #179.

I'm closing this one. You will find a workaround in issue #179 comments.

closed 2021-04-30 by EmmanuelP