AravisProject/aravis #504: Accessing buffer variable after pushing it to the output queue

jamesGadoury, 2021-04-28:
**Describe the bug**
In arvgvstream _close_frame, arv_stream_push_output_buffer is called prior to accessing frame->buffer->priv->frame_id. In a multithreaded application, this means that a user can not take ownership of the buffer memory in their defined callback function, or else they might delete the buffer prior to the attempt to access frame->buffer->priv->frame_id after the callback is called.

**To Reproduce**
Create a separate thread in implemented ArvStreamCallback, and try to delete the buffer. Depending on scheduler, a memory error will eventually happen due to deleting the buffer prior to the attempt to access frame->buffer->priv_frame_id after the callback is called.

**Expected behavior**
If a buffer is pushed to the output queue, then I don't think we should be accessing any of the struct;s members after.

**Camera description:**
This issue is present across camera types.,

**Platform description:**
 - Aravis version 0.8.5, but it looks like the relevant code is the same in 0.8.9
 - OS: Ubuntu 18
 - Hardware x86_64

**Additional context**
I think a simple fix would be to copy the frame->buffer->priv->frame_id into a local variable prior to pushing it to the output queue, and then using that variable in the arv_statistic_fill call instead of accessing the struct's member.

EmmanuelP, 2021-04-30:
Thanks for the report.

closed 2021-04-30 by EmmanuelP

jamesGadoury, 2021-04-30:
Thanks for resolving this so quickly!

commit 189db529 (2021-04-30): gvstream: don't use buffer struct after it was given to the consumer

Fixes #504